jgyates/genmon #202: 3 issues - scrolling, incoming imap port, send email if disconnect

wrvolz, 2018-10-30:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs to the developer, if you email is setup and working properly you can issue this command in your browser:

     http://MYIPADDRESS:8000/cmd/sendlogfiles

Replace MYIPADDRESS to the IP address of you pi and port 8000 if you are using a port other than the default.

You can submit your registers from the Monitor page on the web interface.

### Expected Behavior

{Please write here}
three issues:
1) the scroll bars do not work as expected. I have to use the up/down arrow keys to scroll
2) What port is being used for incoming IMAP? My server uses 993 with SSL/TLS and normal password. 
A section of mymail.log:
2018-10-30 15:49:32,173 : Error selecting mail folder! (select)
2018-10-30 15:49:47,224 : Error selecting mail folder! (select)
2018-10-30 15:50:02,226 : Resetting email threadcommand: SELECT => socket error: EOF : mymail.py:216
2018-10-30 15:51:17,791 : Error selecting mail folder! (select)
2018-10-30 15:51:32,841 : Error selecting mail folder! (select)
2018-10-30 15:51:47,899 : Error selecting mail folder! (select)
3) More of a suggestion. Send an email if the monitor loses connection to the generac controller. Unless I check periodically, I won't know it's not connected. A periodic status email might help too.
Thanks

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: {Please write here}
- Genmon Version: {Please write here}

jgyates, 2018-10-30:
Regarding the scrolling, I will need more info. I don't know what browser, operating system or version of the software you are using so I don't really have anything to go on for this topic.

Regarding IMAP, the software uses the default IMAP port of 143. Based on the info you posed about your logs, your problem is not about your port, it is about the folder on your IMAP mail box. In order to enable incoming mail processing you must Have two folders in your IMAP mailbox that match the settings "Incoming Mail Folder" and "Processed Mail Folder". The following error message:  "Error selecting mail folder! (select)" is when the software attempts to select the Incoming Mail Folder. The way I use the mail commands is this:

1) I have a folder named "Generator" (this is "Incoming Mail Folder") and a sub folder "Generator/Processed" (this is "Processed Mail Folder") on my IMAP mail box
2)  I have a mail filter that looks for incoming emails from approved senders, if the email is from an approved sender and has "generator:" in the subject it moves the email to the "Generator" mailbox.

From there the software will:
1) Read all emails in the "Generator" folder and send the subject line as a command to the software. Only commands prefixed by "generator:" will succeed.
2) Once the command is processed the email is removed from the "Generator" folder and moved to the "Generator/Processed" folder 

The filter steps are recommend as it adds an extra layer or security so just anyone can not send your generator commands. There is also the option "Disable Email Write Commands" that is recommended that will disable commands that write to the generator (i.e. only read status) such as the remote start commands.

Let me look into the notification about missing comms. that sounds like a good idea.

wrvolz, 2018-10-30:
Scrolling issue:
I'm on windows 10. I had the same scrolling behavior with Chrome, IE and MS Edge. I also tried an incognito window (which disables everything else) and still same behavior. When I go to say, Settings, I get a scroll bar on the right side that only allows a small amount of page movement. To get to the bottom I need to use the Down arrow key.

Email commands. I have two folders inbox/Generator and inbox/Generator/Processed. I moved a message with the subject line "generator: status maint" to the Generator folder. It hasn't been processed. I don't know why it can't change to the Generator folder. What other reasons would cause that or is there any other diagnostics I can use?

jgyates, 2018-10-30:
I just tested on Windows 10 with Chrome and I can not duplicate this. Unless someone else can duplicate or unless you can provide additional data, I am going to assume this is something specific to your system. I would suggest trying on a second computer to determine if this is specific to this one PC. The web site does not do any specific processing for the scrolling.

If your path to your folder is "inbox/Generator" and "inbox/Generator/Processed", the exact path must match the values on the settings page.  Are you still getting the same errors in your /var/log/mymail.conf file? If you still get the "Error selecting mail folder! (select)" error then the software is not finding your mail folder for the give settings.

wrvolz, 2018-10-30:
Scrolling issue - I tried on a different Windows 10 computer and the scrolling worked as it's supposed to. Not sure why it's not on my laptop. I'll work on it.

The Generator folder is one down from Inbox. I tried Inbox/Generator and it's still not getting the email. I'm back to your defaults. I looked in the various logs and can't find any further info. Unless you have a way to turn on more verbose debug into, I don't know what I'm doing wrong, if anything. It's not that important anyway - i can get to the control page on my smart phone or from a web page and that's enough for what I need. 

On the weather info, have you considered adding Wunderground site? It's similar but has more cities, at least in the US.

jgyates, 2018-10-31:
The reason I am not using Wunderground is that they no longer provide free api keys.

You may want to try creating a folder that is not a subfolder of your inbox, for example gmail has the ability to create a folder at the root. I have not tried this with a subfolder under inbox. The actual inbox name may be different than the displayed name. When you try this are you still getting the "Error selecting mail folder! (select)" error. You never answered that question from my last post. If you don't need email commands you can disable IMAP on the settings page.

Feel free to comment here if you have any questions on this. I will close this thread.

closed 2018-10-31 by jgyates

jgyates, 2018-11-01:
No problem.

I think wunderground used to offer free api keys but they stopped:

https://www.wunderground.com/weather/api/

jgyates, 2018-11-01:
@wrvolz 
I just checked in an update that will send an email if communications are lost for 2 min and another email when comms are restored.

The web interface shows the comm status on the maintenance page and it is updated every 10 seconds.