jgyates/genmon #204: *Not an issue* Questions before I start this project

AMTNate, 2018-11-07:
I have a 2011 residential 17kw generator and I am curious as to whether or not the battery will be sufficient to power the raspberry pi. Has anyone had problems with the battery draining? Or does the pi not use that much power. 

Also, has anyone had any problems with their dealer/maintenance technicians after installing this? I don't know if this type of project is frowned upon by service dealers.

liltux, 2018-11-07:
@anyone, I am a servicing Technician and wouldn't care less.  I would only make sure you make them aware of it so they do carelessly mess with it.  And if they are disconnecting the battery during the service I would (my opinion) go in and shutdown the pi prior and then allow it to restart on its own when the battery is reconnected.

closed 2018-11-10 by jgyates

liltux, 2018-11-10:
@jwletna that is between you and your service provider.  I wouldn't see such a device causing any harm.  But in the end, any outside modifications to Generac equipment could affect warranty.  That would be the the same for any manufacturers equipment.  But what Generac doesn't know... fallback to that's between you and your servicing dealer.