jgyates/genmon #205: Communication Failure happens after some period of time

wotseurba, 2018-11-16:
### Actual Behavior
Genmon stops communicating with controller after a period of time from minutes,  to 6 hrs.

I note packet failure (or slow down) on the monitor page of the web interface.

The Status Page Dashboard appears to be receiving and displaying data during the reported communication failure. (and dashboard graphics of Batt. Voltage, and Utility Voltage display perfectly)
Status :
        Engine :
            Switch State : Auto
            Engine State : Off - Ready
            Battery Voltage : 12.90 V
            RPM : 0
            Frequency : 0.00 Hz
            Output Voltage : 0 V
            Output Current : 0.00 A
            Output Power (Single Phase) : 0.00 kW
            Active Rotor Poles (Calculated) : 0
        Line State :
            Utility Voltage : 241 V
            Utility Voltage Max : 243 V
            Utility Voltage Min : 239 V
            Utility Threshold Voltage : 156 V
        Last Log Entries :
            Logs :
                Alarm Log : 10/30/18 14:38:23 Undervoltage (Prolonged) : Alarm Code: 1900
                Service Log :
                Run Log : 11/10/18 12:05:18 Stopped - Auto
        Time :
            Monitor Time : Friday November 16, 2018 09:57:02
            Generator Time : Friday November 16, 2018 09:45

Monitor Page when working:
Monitor :
        Generator Monitor Stats :
            Monitor Health : OK
            Controller : Evolution, Air Cooled
            Run time : Generator Monitor running for 0:37:44.
            Generator Monitor Version : V1.11.22
        Serial Stats :
            Packet Count : M: 72005, S: 72005
            CRC Errors : 0
            CRC Percent Errors : 0.00%
            Packet Timeouts : 0
            Packet Timeouts Percent Errors : 0.00%
            Modbus Exceptions : 0
            Validation Errors : 0
            Invalid Data : 0
            Discarded Bytes : 0
            Comm Restarts : 0
            Packets Per Second : 63.59
            Average Transaction Time : 0.0309 sec
        Platform Stats :
            CPU Temperature : 97.70 F
            Pi Model : Raspberry Pi 3 Model B Rev 1.2
            Pi CPU Frequency Throttling : OK
            Pi ARM Frequency Cap : OK
            Pi Undervoltage : OK
            CPU Utilization : 7.31%
            OS Name : Raspbian GNU/Linux
            OS Version : 9 (stretch)
            System Uptime : 0:38:21
            Network Interface Used : wlan0
            WLAN Signal Level : -59 dBm
            WLAN Signal Quality : 51/70
            WLAN Signal Noise : -256 dBm

Monitor Page just after notification of communication failure:
Monitor :
        Generator Monitor Stats :
            Monitor Health : Not receiving data.
            Controller : Evolution, Air Cooled
            Run time : Generator Monitor running for 0:49:51.
            Generator Monitor Version : V1.11.22
        Serial Stats :
            Packet Count : M: 85324, S: 85220
            CRC Errors : 0
            CRC Percent Errors : 0.00%
            Packet Timeouts : 103
            Packet Timeouts Percent Errors : 0.12%
            Modbus Exceptions : 0
            Validation Errors : 0
            Invalid Data : 0
            Discarded Bytes : 0
            Comm Restarts : 0
            Packets Per Second : 57.01
            Average Transaction Time : 0.0309 sec
        Platform Stats :
            CPU Temperature : 97.70 F
            Pi Model : Raspberry Pi 3 Model B Rev 1.2
            Pi CPU Frequency Throttling : OK
            Pi ARM Frequency Cap : OK
            Pi Undervoltage : OK
            CPU Utilization : 6.95%
            OS Name : Raspbian GNU/Linux
            OS Version : 9 (stretch)
            System Uptime : 0:50:27
            Network Interface Used : wlan0
            WLAN Signal Level : -60 dBm
            WLAN Signal Quality : 50/70
            WLAN Signal Noise : -256 dBm

### Your Environment
- Generator Model: 7063R1 Air Cooled 20kW
- Genmon Version: V1.11.22

### Steps to Troubleshoot
Fresh install of OS and Genmon onto a Class 10 SD card
Tried 2 different Raspberry Pi's (same generation)
Tried two different TTL-Serial converter (Max3232 units you specify)
Loopback serial tests always passes in 3 modes: GPIO pins, DB-9 Pins, and to Molex pins
Serial tests after Genmon goes into communication failure always work perfectly 
Performed all serial troubleshooting steps with results fine
Re-booting Genmon does not clear the communication failure
Re-booting Raspberry Pi does not clear the comminication failure
Shut Down and Restart of Raspberry Pi DOES rectify the communication failure, but after some random time goes into communication failure.

Kinda stumped on what to do next

wotseurba, 2018-11-16:
Many thanks for the reply.

I did previously perform all of the serial troubleshooting detailed on the wiki.

Per your recommendation, I again ran serialconfig.py -c  with the following results:
pi@raspberrypi:~ $ sudo python /home/pi/genmon/OtherApps/serialconfig.py -c

Checking : BT service disabled: OK
Checking : Serial console command line removed: OK
Checking : Is enable UART in boot config: OK
Checking : Is BT overlay disabled: OK
Checking : Serial console service disabled: OK

Serial port settings are OK. A reboot is needed if changes were made.

What's curious to me is that a shutdown and restart of the Pi resolves the communication issue (for a while).  I did build the Molex from the power cable you recommended, so there's are no faulty crimps.  I also soldered all connections.  Have re-seated both the Molex and the DB-9 many times with no effect.

Caveat:
My Pi is indoors, with serial cable (22 ft.) extending to the generator, which is well within RS232 maximum of 50 ft.

I used Cat6 Ethernet cable, with Rx & Tx on a twisted pair.  Of course, no shielding which I would fault  if the communications were to break down while the generator is running due to EMI.   Again, I am stumped why the communications run for a while, then packets slow down to cause the warning from Genmon.

If you can't think of anything else, I will change out the CAT6 cable for heavier gauge shielded cable to see if that helps.  Won't get to that until next week.

wotseurba, 2018-11-17:
myserial.log is empty

mymodbus.log is loaded with: (just a few lines copied here)

2018-11-16 10:36:41,677 : Error: timeout receiving slave packet for register 05f6 Buffer:0
2018-11-16 10:36:44,708 : Error: timeout receiving slave packet for register 05f5 Buffer:0
2018-11-16 10:36:47,734 : Error: timeout receiving slave packet for register 05f4 Buffer:0
2018-11-16 10:36:50,759 : Error: timeout receiving slave packet for register 05f3 Buffer:0
2018-11-16 10:36:53,795 : Error: timeout receiving slave packet for register 05f2 Buffer:0
2018-11-16 10:36:56,820 : Error: timeout receiving slave packet for register 023e Buffer:0
2018-11-16 10:36:59,847 : Error: timeout receiving slave packet for register 023d Buffer:0
2018-11-16 10:37:02,878 : Error: timeout receiving slave packet for register 05ed Buffer:0

jgyates, 2018-11-17:
This indicates that either the controller is not receiving the request for data or the pi is not receiving the response. When you performed a loopback, did you try the loopback through the 22ft cable ? Also, are you running a ground wire in your 22ft cable?

wotseurba, 2018-11-17:
Ground wire = Yes

Loopback tests (all 3 modes, GPIO pins, DB-9 at serial converter, and to end of cable to Molex) all pass on initial power up of pi.

After some time period (hours - days) the communication eventually fails.  I find this odd.

**Loopback tests while in failure mode:**

**This is test to end of Molex:**
Loopback testing for serial port /dev/serial0...
Serial port opened
write data: sent test string
waiting to received data....
FAILED: Sent data does not match receive. Received 0 bytes

**This is test at DB-9:**
Loopback testing for serial port /dev/serial0...
Serial port opened
write data: sent test string
waiting to received data....
FAILED: Sent data does not match receive. Received 0 bytes

**This is test at GPIO pins:**
Loopback testing for serial port /dev/serial0...
Serial port opened
write data: sent test string
waiting to received data....
PASSED! Loopback successful

From this, I would condemn the TTL-Serial converter.  
**BUT, I tried a second identical (new) one with same results.**

BTW, I've run the TTL-Serial converter on both 5v and 3.3v no difference.

I am using this one:
https://www.amazon.com/dp/B072KJSS5C/ref=sspa_dk_detail_3?psc=1&pd_rd_i=B072KJSS5C&pd_rd_wg=Nd6FS&pd_rd_r=VZ5RZZSGJ0JP5JMS1AYK&pd_rd_w=UqGO1

Perhaps I should try the other one you recommend??

Again, it takes a full shutdown and removal of power to the pi and new startup to clear the fault.

liltux, 2018-11-17:
I have ran into a batch of poor ttl converters.  They would work sometimes and sometimes not.

wotseurba, 2018-11-17:
I suppose both of the converters I purchased are defective.  Can't expect 6 Sigma quality from Asian sweatshop electronics factories.  

I just ordered: 
https://www.amazon.com/gp/product/B00EJ9NAKA/ref=oh_aui_search_detailpage?ie=UTF8&psc=1

Will try it as soon as it arrives.  Fingers crossed.
Thank you !!

wotseurba, 2018-11-24:
Resolution:
Installed this serial converter: 
https://www.amazon.com/gp/product/B00EJ9NAKA/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1&pldnSite=1

It has solved the issues I was experiencing.  My system has now been running with uptime of 3 days with no communication failure.

Note to others, this serial converter is junk:
 https://www.amazon.com/dp/B072KJSS5C/ref=sspa_dk_detail_3?psc=1&pd_rd_i=B072KJSS5C&pd_rd_wg=Nd6FS&pd_rd_r=VZ5RZZSGJ0JP5JMS1AYK&pd_rd_w=UqGO1

I purchased two of them 5 weeks apart from Amzon, the second one for troubleshooting purposes, and BOTH are defective.  Please consider removing it s a recommended piece of hardware.

Many thanks for the assistance !!!

jgyates, 2018-11-24:
Will do.

closed 2018-11-24 by jgyates

rsockanc, 2025-05-17:
I will share my story for those who think they will be brave with a long copper run outside to the generator.  I ran a serial over Cat6 to my generator as my Pi was inside the house for a few years until a lightning storm that had a few strikes within a mile, yet not on property, fried my serial to USB converter AND my Evolution controller, and had to replace it for a wonderful $495.....   So I now run  PUSR USR-W610 serial over Wifi that is powered directly from the battery, and my Pi is inside the house as it runs other apps, and I didn't want it right next to the generator.

8 Bit Guy was struck as well and moved to fiber: https://youtu.be/Ev0PL892zSE?t=407

Other options: https://blog.networkprofile.org/monitoring-generac-generator-with-raspberry-pi-and-om3-fiber/