jgyates/genmon #207: Non-Issue / Question: View web Interface remotely?

nah2323, 2018-11-17:
So without opening a port on a router, has anyone achieved remote viewing of the web interface?  I’ve seen hints to VPN, but Wouldn’t that just allow me to get into the Pi?  Does GenMon show up on the Pi directly?  I’ve only seen it on the “iPaddress:8000”   Any help would be awesome.  I have a few generators on different ip’s I’d like to see remotely.

*Note- I’m super new to RPi, so a good portion of Linux is lost on me, but this wiki has been helpful and I’ll get it if explained clearly.

jgyates, 2018-11-17:
You have a few options, however the easiest method, in my opinion is to use the VPN functionality built into most home routers. Many routers have VPN functionality built in so once you setup and enable VPN on the router you can access your entire network remotely, not just the pi. Another option is to have a VPN server running on a remote system and install a VPN client on the pi however this requires a lot more setup on both ends in my opinion and there are some pitfalls (like you need to make sure the pi is still accessible to the local network and the internet). A third method is to open ports on a router / firewall.

If you go with method one you can access my pi from my phone or computer remotely. You just have to have a VPN client on each remote device and your router must support a VPN server functionality.

liltux, 2018-11-17:
I agree with @jgyates.  But, I have used VPN access to view GenMon on a generator that backs up our shop/business as well as use the same VPN to access a webserver on my security camera at my house.  I use DigitalOcean (5$ month for a droplet) and setup a VPN Server on a Ubuntu virtual machine.  Digital Ocean has great tuturials that walk you through the whole process.  I just create ovpn configuration files for each device I want connected to the VPN.  Then anywhere I have internet access I can access any device on my VPN.  And the Openvpn on the pi is a breeze to setup the client configuration.  Thats just what I do... best way to learn is by doing (and Google 'ing).

closed 2018-11-18 by jgyates