jgyates/genmon #208: Display of SN missing with upgrade to V1.11.24

hokie21, 2018-11-17:
### Expected Behavior
The Generator Serial Number on the maintenance page should be set by the serialnumberifmissing parameter from genmon.conf.

### Actual Behavior
With an upgrade to version V1.11.24 (and with a generator that has had its controller board previously replaced) the serialnumberifmissing parameter from genmon.conf is no longer working to display Generator Serial Number on the maintenance page.

jgyates, 2018-11-17:
I think I know what the issue is regarding the serialnumberifmissing parameter. The initial version of this feature had and entry in /etc/genmon.conf named serailnumberifmissing  (note the misspelled word 'serail' instead of 'serial'). This type was changed a while back and the software supported either spelling (incorrect and correct spelling). Around version 1.11.9 some changes were made to support the advanced page. On the advanced page the serial number could be entered from the web interface (click the gear in the top right twice to get to the advanced page). 

To resolve this you can do one of two things. Either:
1) change the spelling of the entry in /etc/genmon.conf from serailnumberifmissing to serialnumberifmissing, save the file and restart the software.
2) Go to the advanced page and enter your serial number. This will result in two entries (one misspelled and one correct) in /etc/genmon.conf. The misspelled one will be ignored.

The reason that I took out the support for the misspelled entry is that the advanced page can only support one entry per value displayed with the way the code is structured so this was the more maintainable thing to do.

Let me know if this does not solve your problem.

hokie21, 2018-11-17:
I did number 1 and that fixed it.  I just found the advanced page. I didn't know that existed.
Thanks again for your help!

closed 2018-11-17 by hokie21