jgyates/genmon #209: Question From Control Engineer

trimixdiver1, 2018-11-20:
Hello

Im an automation engineer, I would like to bring this into my PLC, I understand the serial pins in the connector, but can you tell me what the registers are? Im used to seeing 40100 type registers for MODBUS. Also the machine ID?

Thanks so much!

trimixdiver1, 2018-11-20:
Do you have a list of registers and what they display? I have a 9kw 2017 model
So everything is hex?

Thanks

jgyates, 2018-11-20:
For evolution / nexus the list of registers read is here:

https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L94

I am assuming you have an evolution controller based on a 9kw 2017 model.

In the code I typically use hex, you could use decimal in the code, however what is read over serial is binary so hex is just the representation I used in the code as it lends better to bit manipulation.

trimixdiver1, 2018-11-20:
Got it, so as long as I connect my PLC to it as Modbus RTU, your 0009 should translate to 40009 for me?

jgyates, 2018-11-20:
I am closing this issue for now. Feel free to respond to this thread if you have any related questions.

closed 2018-11-20 by jgyates

trimixdiver1, 2018-11-20:
What is the difference between Evo LC and LQ?

jgyates, 2018-11-20:
LQ is a typo in the comments, think some of the comments use AQ instead of AC and LQ instead of LC

Evolution Air Cooled (Evo AC)
Evolution Liquid Cooled (Evo LC)

trimixdiver1, 2018-12-23:
Hello

So Im starting to fool with the input registers. Are there registers to change the state of the Evo? OFF, Manual and Auto?

Thanks

jgyates, 2018-12-23:
You can do this on the liquid cooled units. I am not aware of how to do this on Air cooled units. It requires writing two registers (0003 and 0004) so for the 9kw model that you have, I do not know a way to do this.

trimixdiver1, 2018-12-23:
Those registers are zero on mine, what do you write to them?

jgyates, 2018-12-23:
According to you earlier post, you have an air cooled unit, correct (9kw)? To my knowledge, you can not change the state of the switch via modbus on an air cooled unit.

trimixdiver1, 2018-12-23:
Yes I have an AC 9kw.  Ok, just checking. mine are zeros, so LC would have different numbers based on status?

jgyates, 2018-12-23:
No these registers (0003 and 0004) are write only registers, I don't think reading them will produce any meaningful data. They appear to be an indexed set of registers (i.e. write the index number to register 0003 and the value to register 0004).

trimixdiver1, 2018-12-23:
Ok, so I shouldnt even try? Your pretty certain then?

jgyates, 2021-10-17:
That is great. I am glad it is working for you. I hope the job works out too.

stellagits, 2021-10-18:
Thanks I'll let you know how it goes.

In the menatime I have another question :(

I've been able to establish a connection to the generator controller and even bring a couple of points into the SCADA. The problem is I am really unfamiliar with the MODBUS addressing in the EVO controller and I am having trouble trying to figure out from the GENMON ource code. I'm also not very good ant C++. I basicaly got lucky with the three points I've brought in so far.

1. As I understand it, all of the registers that GENMON interfaces with (and that Generac used) are Holding Registers. I don't see any coils or discrete outputs being used. Is this correct?

2. It appears that some of the registers in the controller are 16bit and some are 32bit. I say this because I was able to bring in the Frequency amd Line Voltage values by addressing them as 32bit Holding Registers but the Battery Voltage value was in a 16bit register:

[image]

Is there anyway to tell from the source code which registers are 16 and which are 32?
Is there any instance in the source code where a value is using two 16bit registers?

3. Referencing the above screen cap and based on the GENMON source code, the Battery Voltage should be in register 40010, but I found it in register 40011. I'm not sure why this is, any ideas?

4. I am still trying to figure out how to handle addressing the high and low bytes in the SCADA. I want to make sure I undersand the concept of high and low bytes correctly first though. From what I can gather, the first byte (high) and second byte (low) of a register are used to store seperate values within the same register, is this correct? I believe I can bring in the full register as a raw value tag and then create a second expression tag in the SCADA that just reads the high or low bytes, deriving the stored value that way. I think that's what you did in the source code but I'm not sure.

stellagits, 2021-10-18:
"I hope this helps."

Um, yeah. It helps. A lot.  I'm bringing points in left and right now! Understanding how to differentiate the differnece between the 16 and 32 bit registers was big but figuring out the non-indexed addressing in the SCADA clinched it. Thank you so much.

Using your example for engine run time, I can perform a 32bit read and get both the high an low bytes (which I did in SCADA). This works because the total run time is stored in the 4 bytes. What I need to figure out is how to handle something like:

            "000e" : [2, 0],     # Read / Write: Generator Time Hi byte = hours, Lo byte = min

In this case we have a 16bit register where the high byte is hours and the low byte is minutes. If I just do a 16bit read of that address I get "3367" (0000 1101 0010 0111 in binary). So I need to figure out in SCADA how to write an expression that breaks out the high (0000 1101) and low (0010 0111) bytes in order to get the hours and minutes values. I'm posting that question to the forum for the SCADA vendor and see what they come back with. I see posts there from people asking about flipped registers and expressions on how to deal with them so my hunch is this is possible.

Seriously? How did you figure out the MODBUS mapping?