jgyates/genmon #210: Older Generac 2008 Controller

AquaStuff, 2018-11-21:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs to the developer, if you email is setup and working properly you can issue this command in your browser:

     http://MYIPADDRESS:8000/cmd/sendlogfiles

Replace MYIPADDRESS to the IP address of you pi and port 8000 if you are using a port other than the default.

You can submit your registers from the Monitor page on the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generac 20kw Air Cooled 2008 model
- Generator Registers: {Please write here}
- Genmon Version: V1.11.23

Wondering if anyone has been able to get Genmon to work on the older Generac 2008 Controller 

I receive the following in the MyModbus.log file
2018-11-21 10:26:38,087 : Discarding byte slave: 41
2018-11-21 10:26:38,088 : Error Receiving slave packet for register 000c
2018-11-21 10:26:38,357 : Discarding byte slave: 41
2018-11-21 10:26:38,358 : Error Receiving slave packet for register 0258
2018-11-21 10:26:38,624 : Discarding byte slave: 41
2018-11-21 10:26:38,625 : Error Receiving slave packet for register 0242
2018-11-21 10:26:38,890 : Discarding byte slave: 41
2018-11-21 10:26:38,891 : Error Receiving slave packet for register 0053
2018-11-21 10:26:39,156 : Discarding byte slave: 41
2018-11-21 10:26:39,157 : Error Receiving slave packet for register 0052
2018-11-21 10:26:39,422 : Discarding byte slave: 41
2018-11-21 10:26:39,424 : Error Receiving slave packet for register 0001
2018-11-21 10:26:39,689 : Discarding byte slave: 41
2018-11-21 10:26:39,691 : Error Receiving slave packet for register 05f1
2018-11-21 10:26:39,705 : Error in SendPacketAsMaster: argument must be an int, or have a fileno() method. : mymodbus.py:474
2018-11-21 10:26:39,706 : Packet: [157, 3, 0, 9, 0, 1, 73, 148]

liltux, 2018-11-21:
Anything pre Nexus controller was not setup to work with MobilLink.  So it's a good chance it will not work.  What controller are you trying to communicate with?  Example, printed circuit board or is it a small grey square controller with two arrow buttons, enter and esc?

jgyates, 2018-11-21:
Unfortunately no. I don't have any data to work with so this remains unsupported.

AquaStuff, 2018-11-21:
It's the small grey square controller with two arrow buttons, enter and esc.

Anyone know if you can upgrade the older 2008 controller to the newer Nexus controller?

liltux, 2018-11-21:
There is no upgrade available from Generac, the harness and alternator are different.  Though if you really wanted to you could Frankenstein it and make it work, but nothing official is offered.

closed 2018-11-22 by jgyates

liltux, 2019-03-25:
@jgyates , I had someone get a preassembled unit for their generator.  According to this person their generator was running with MobilLink, but their controller is the small square controller as mentioned above.  Being that it was working with MobilLink, would it be possible to get some good data points to integrate into the project?  If so what would you need.  I have a controller I am going to take home and hook up to a the Genmon software and see what is displayed.   I feel the biggest issue with these older controllers is there is very minimal information to retrieve.  There is no internal voltage regulator, so all output voltage and power calculations are not there.  You should have though utility sense, start/stop, excercise, and shutdowns.  Let me know what I can do for you if your interested.

jgyates, 2019-03-25:
@liltux 

Sure, we can try it. The problem I ran into before was that the controller did not appear to respond to modbus, but it is possible that this is a controller that looks the same but actually does support modbus communications. 

The first thing I would try is to just run genmon as you normally would. If this does not work we will see errors in the /var/log/mymodbus.log file. 

The second thing to try is to then stop genmon from running by running this command in the genmon directory:

     ./startgenmon.sh stop 

Then in the genmon/OtherApps directory there is a program called modbusdump.py. This is a python script that will read modbus registers over the serial link. The program has this syntax:

python mobusdump.py -r <Baud Rate> -p <serial port> -a <modbus address to query> -s <start modbus register>  -e <end modbus register>

   Example: python mobusdump.py -r 9600 -p /dev/serial0 -a 9d -s 5 -e 100 

      -r  Baud rate of serial port (9600, 115300, etc)
      -p  Operating System device name of the serail port (/dev/serial0)
      -a  Modbus address to query in hexidecimal, 0 - ff. (e.g. 9d, 10, ff)
      -s  Starting modbus register to read (decimal number)
      -e  Ending modbus register to read (decimal number, must be greater than start register)
      -b  Stop bits. If omitted 1 stop bit, if present use 1.5 stop bits
      -x  Omit for no parity. -x 1 for odd parity, -x 2 for even parity

For example this command:

      python modbusdump.py -r 9600 -p /dev/serial0 -a 9d -s 5 -e 7

Will assume port /dev/serial0 on the pi at 9600,n,8,1 with a modbus address oef 9d (the default for Nexus and Evolution) and it will attempt to read modbus registers 5 - 7 (which are valid on both Nexus and Evolution). The program will dump all the registers to the screen so you can redirect this output to a file

       python modbusdump.py -r 9600 -p /dev/serial0 -a 9d -s 5 -e 7 > output.txt

Let me know if you have any questions.

liltux, 2019-03-26:
@jgyates 
I didnt have much time last night to tinker around a whole lot, but when I ran your modbusdump.py all that was displayed was my settings and then the program finished.  No errors just no data?  Do I need to run the file in a different directory?  I have asked the person that is interested to send me pictures of their installation, curious if the 2008 controller used a different pinout.  So far I am not finding any information in the manuals on the correct pinout.  I would assume it should be the same as well as the communication settings.  I will try it out again tonight and post some screen shots.

jgyates, 2019-03-26:
@liltux 

It will store log files in the current directory. I would change directories to the OtherApps directory and run it from there. You will likely see a mymodbus.log file in the directory in which you ran the test. If it did not print anything out then I am guessing the log file will show timeout errors.

jgyates, 2019-03-29:
I have checked an update (V1.12.31) that should allow support for 2008 models. You will need to update to the latest software, then set the "Modbus slave transmit address" to 41 on the advanced page in the web interface and restart. [This page](https://github.com/jgyates/genmon/wiki/Appendix-4-Known-Issues) as a little more info.

jgyates, 2024-10-04:
@AquaStuff 

do you see anything like the behavior described in this thread: https://github.com/jgyates/genmon/issues/1191