jgyates/genmon #211: Fule Monitor

diverbill, 2018-11-22:
You can submit your registers from the Monitor page on the web interface.

### Expected Behavior

Fuel gage would show estimated fuel in the tank, The latest update with the fuel pressure sensor works well.
### Actual Behavior

The gage shows as always Empty and get the e-mail that the tank is below 10% (Fuel Level Sensor : 88 %)

### Steps to Reproduce (including precondition)

unsure - more of a suggestion have a switch on the settings page to use the level sensor for both the fuel gage and alerts. then have your current method as the other option.

### Your Environment

- Generator Model: RD02023ADAE
- Generator Registers: Sent
- Genmon Version: V1.11.25

jgyates, 2018-11-22:
@diverbill 

thanks for the feedback. I thought about doing this but wanted to get some feedback before proceeding since I have a limited number of folks using Liquid Cooled Evolution models.

To be clear, the fuel level sensor only appears to be valid for diesel units and I have coded it so this only shows up for diesel LC Evolution units.

Currently the fuel gauge works differently as there is not sensor reading involved (if you tell it how much fuel you have and how big your tank is it will subtract fuel based on you kw output and the fuel consumption algorithm based on the model of generator taken from the data sheet)

I will add an option to use the sensor for the gauge. Once I do let me know how it works as time progresses. The sensor appears to be a resistor that changes based on a float in the tank. Based on the data read from the controller it looks like they are normalizing the voltage read to a percentage. My tank gauge reads around 74% but my sensor reads 80%, which is close but keep in mind the sensor may not be a linear as the physical gauge on the tank.

I will likely get to this some time this weekend.

jgyates, 2018-11-22:
FYI, here is the info on the current implementation of the fuel tank estimate:

https://github.com/jgyates/genmon/wiki/Appendix-8-Monitoring-Fuel-and-Power-Usage

diverbill, 2018-11-22:
Thank you, I have been reading that I have a 20Kw Diesel Generator with a 32 Gallon tank. I'll be happy to test for you. Have a Happy Thanksgiving!

jgyates, 2018-11-22:
I checked in the update. The default  case for EvoLC Diesel is to use the sensor for the fuel gauge. 

If I disconnect the sensor on my unit the reading goes to zero. The thing we need to look for is does the value read from the sensor track with the level in the tank. Based on my data and your data it appears to do this however we should watch the values as the tank approaches empty. An additional calculation may be needed to make the value more linear.

jgyates, 2018-11-22:
There is a setting on the settings page that allows you to switch between the two methods for the fuel gauge.

jgyates, 2018-11-24:
Sure. I have a way to import data from an external app now that is documented here:

https://github.com/jgyates/genmon/wiki/Appendix-3--Interfacing-Generator-Monitor-to-External-Applications

See the section that talks about "Displaying External Information in the Web UI". This could be used to import the data from a external fuel sensor. If the data were formatted in a way that we could abstract the gauge hardware then we could just make it an add on program and have the gauge software import a reading of 0  to 100 based on a percentage of the tank reading. This way of someone else wanted to use a different sensor they could.

diverbill, 2018-11-25:
Hello, I have installed the new version and have been running it for a few days. Seems to be running well showing about the correct level I have in my tank, the next scheduled run is next weekend. I did notice that the 10% warning still is tripped when the pi boots. I'm not sure why but the calculation method has not worked on my install since I set it up. 

A few more suggestions/feedback
- Allow us to set the low level alarm (10%, 15%, 20%, or 25%)
- Utilize the same setting as the fuel gage
- Fuel Consumption in last 30 days be used on a calculation of fuel level

jgyates, 2018-11-25:
Hi @diverbill 

I checked in an update that should address some of these issues.

I added the fuel consumption curves for RD015, RD020, RD030 and RD050. RD048 was the only one supported before. This means that if your power reading (kw output) is correct then the fuel estimates should now work properly.

I also changed the notification behavior: if you have "Use Fuel Sensor for Gauge" enabled on the settings page then the email notices will check the fuel sensor instead of the fuel estimate calculation.

The way the fuel calculation works is that the power log is examined and the total power logged for a given period is used to determine the percentage load of the generator. This load is then fed into the power curves (a polynomial that is derived from the power consumption data in the generac data sheet). This yields an volume of fuel consumed and this is subtracted from the tank size (more details are [here](https://github.com/jgyates/genmon/wiki/Appendix-8-Monitoring-Fuel-and-Power-Usage)). The fuel consumption for the last 30 days is calculated this way. The software does not log the fuel sensor, only the power log so this calculation has not changed, it is still based on the estimate derived from the power log.

Regarding the low fuel level alarms, the software currently sends alarms at 10% and 20%. I am guessing you were getting a false notice of a zero reading of fuel since I did not have the fuel consumption curves loaded for your model. If you still get the fuel level email let me know and we can figure that out. I will think about adding the ability to change the notice level. I have not witnessed this, but  I think on the diesel units the controller will generator an warning when the fuel goes below 20% so you should get a notices on that even like you normally would when an alarm occurs.

Let me know if you have any questions or issues.  I am closing this issue for now, but feel free to post your feedback / questions to this thread.

closed 2018-11-25 by jgyates

fbobro, 2019-02-27:
I have had an extended run of my Generac 20KW diesel generator. The fuel gauge appears to be tracking well - except when the tank is visibly full, the program says 93%. 

Suggestion - 
Perhaps a user defined setting to calibrate the program when the tank is full. 
A second for empty. 
Perhaps a third for 1/2 as indicated on the mechanical gauge. 

I say user setting to allow for differences in the mechanical gauge. It could be a button on the set up page that calibrates the program to the generator tank. 

I did press the submit registers button tonight at 21:15 ET 2/26/19

Model Number: RD02023ADAE
Controller: Evolution, Liquid Cooled
Version:  V1.12.18

One off topic item, if I may,

The program takes a really long time display the first time it is opened. I can then easily move between the pages on the left. Tonight it took 30 minutes. 
I thought that I had done something wrong, so I set up a second pi. It was quick on my test bench without the serial connection to the generator plugged in to it. 

When I plugged in the serial cable to PI #2 it slowed down.
Pi #1 (without a connected serial cable) now loaded fast. 

Any thoughts on what I can check.

Love the program. Thank you!!!!!

jgyates, 2019-02-27:
@fbobro 

I did not receive  your registers. Make sure your outgoing mail is configured and working. You can test your outgoing mail on the settings page. 

Regarding the fuel gauge. The fuel sensor that is available for diesel liquid cooled units **appears** to function as a percentage of the tank level. The fuel sensor is attached to a float in the tank so it's accuracy may vary based on the size of the tank. When researching this it appears that the same sensor is used for all size tanks. I stress the word appears since this is just empirical data. I have a liquid cooled diesel unit as well.  When testing if I disconnect the sensor the reading goes to zero and I noticed that it does track the fuel level fairly accurately. If you disable the setting "Use Sensor for Fuel Gauge" on the settings page the fuel gauge in the web interface will use the calculated fuel usage for the gauge instead of the sensor.

A long delay for displaying is not normal. I would first look at the speed rating of the SD card in your pi. If it is slow this could cause problems. A Class 10 SD card is recommended. Speeds below a class 10 will affect performance in file system reads and writes and you may experience timeouts on the web interface with a slower card. Another thing to look at is your serial stats on the monitor page. If your cable is intermittent it can cause things to be slower, but still, 30 min is unusually long delay.

As a point of reference here are my serial stats after running for 3 hours (i recently restarted due to a software update):

Serial Stats :
        Packet Count : M: 390792, S: 390791
        CRC Errors : 0 
        CRC Percent Errors : 0.00%
        Packet Timeouts : 0
        Packet Timeouts Percent Errors : 0.00%
        Modbus Exceptions : 0
        Validation Errors : 0
        Invalid Data : 0
        Discarded Bytes : 0
        Comm Restarts : 0
        Packets Per Second : 64.14
        Average Transaction Time : 0.0306 sec

I would try to resend your logs as they may shed some light on the cause of the problem.

jgyates, 2019-02-27:
@fbobro 

I received your logs. Two things that are noteworthy:

1) There appears to be an error in your mail setup, up to the last reload. If you changed anything about the our email settings, then see if you still see the delay. the failure was retrying the email in the queue many times so if this fixes the problem I can look at adjusting the logic in the retry.
2) You are several errors related to the openweather api settings. Try disabling that and see if it loads as expected. If it does then we can look at why you seeing the problem with the openweather api.

jgyates, 2019-02-27:
Of course the SD card may still be a problem too...

one other thing to try: restart genmon, let it load then immediately (once the web UI is up) send the log files. this will give me time stamps for the startup events / errors that could be helpful. You genmon.log file is rolling over so I can can't see the error output for genmon on initial startup.

fbobro, 2019-02-27:
I did an outgoing mail test to my address and it was received by me. The from address is not the one I am using here but another one that starts as n3zk.paqp@ I did resend the "submit registers" and the "submit logs" using the buttons on the about page. 

I made a change that caused a program reset 26 minutes before the data below - I turned on the "Disable Power / Current Display" as it was not working and I do not have the current sensors on my generator. I would like to know if I can add them and what pins they would connect to on the controller.  

When I change the setting of "Use Sensor for Fuel Gauge" to off, my fuel gauge drops to 0%. 

    Serial Stats :
        Packet Count : M: 57818, S: 57817
        CRC Errors : 0
        CRC Percent Errors : 0.00%
        Packet Timeouts : 0
        Packet Timeouts Percent Errors : 0.00%
        Modbus Exceptions : 0
        Validation Errors : 0
        Invalid Data : 0
        Discarded Bytes : 0
        Comm Restarts : 0
        Packets Per Second : 63.52
        Average Transaction Time : 0.0309 sec

I understand that the gauge may be different in how it reads for the different generators. Let me try to say it a different way - can I define the top of the gauge's resistance and the bottom of the resistance so the program can calculate 100 and 0 percent from those two readings?  

I am pleased that the gauge moves and I have an idea of what it is without having to go outside and look in the cold. 

I just did the "upgrade to latest version" when it restarts I will disable the weather app. When it allows me to I will resend the logs. 

What I do find interesting is that the "bench" pi web display loads really quick - the only thing missing from that one is the real connection to the generator. 

Again, I really appreciate the effort you have invested into this program and also the quick reply. Thank You!

fbobro, 2019-02-27:
Sent logs after restart.
Then turned off weather in settings. 

I have not changed any of the email settings. The Firefox has still not loaded - I sent the logs from the i-phone which loaded the page quickly.

jgyates, 2019-02-27:
These logs looks a lot cleaner. Nothing of note in the logs. 

If the iphone loads quick and firefox does not then I would suggest looking to see if you have any firefox extension that are causing the problem. Try loading with Chrome, Edge or Safari and see if you have the same problem. Also see if you can disable any Firefox add ons or extensions and see if this helps the delay you are seeing.

In firefox you can load the builtin in inspetor to see what web call is failing. In Firefox look under Tools->Web Developer - > Inspector. Then a window will appear at the bottom of the browser window. in this window click on the "Network" menu item. the load the web page for genmon.

It should look something like this:

[image]

You may see items that are failing (like status_json, gui_status_json, star_info_json). these are calls that the web page is making into genmon. if they fail for some reason the the web page will not load. There may be some problem with an extension that is causing problems with these functions. This will let us know which calls.

jmcjclark, 2019-05-24:
I would like to add this fuel level sensor to my propane tank and raspi running the gen monitor program. https://rochestergauges.com/product/ds-1828/
Has the functionality been added to support this? If so, which pins of the pi do I connect the sensor?

jgyates, 2019-05-24:
The software does not support any third party fuel sensors. The fuel monitoring that is supported is:

1) For Evolution Controllers - Fuel consumption is derived by current flow from built in current transformers (CTs) or Hall Effect sensors.
2) for Evolution Liquid Cooled Diesel units the generac tank has a fuel sensor that is read for thoes models
3) You can have a third party app export the fuel sensor data and write it to ./genmon/userdefined.json and it will be displayed on the monitor page:  https://github.com/jgyates/genmon/wiki/Appendix-C--Interfacing-Generator-Monitor-to-External-Applications#displaying-external-information-in-the-web-ui