jgyates/genmon #213: Thanks for the help

trimixdiver1, 2018-11-30:
So I used the information here and got the Evo to communicate with my PLC via MODBUS. I looked at the screen capture of the Genmon and was able to get several registers up and running. I noticed my EVO also gives the internal Temp, but the register in the code is for LC models only and I tried it too. Are there any updated lists of registers?

Thanks Again

jgyates, 2018-11-30:
Unfortunately no. All of the registers in the code were determined from observation / testing. Many of the registers are specific to the generator type. In general the following are the types of generators:

Nexus Air Cooled 
Nexus Liquid Cooled 
Evolution Air Cooled (1.0)
Evolution Liquid Cooled 
Evolution 2.0 Air Cooled 
Evolution PowerPact 

As I understand it each of these has separate firmware.

The ambient temperature sensor is exported to modbus for Evo 2.0 Air Cooled. The Temp sensor for Evo liquid Cooled is not exactly accurate as it appears to be designed simply to modify the battery charging algorithm. I have not been able to find the ambient temp sensor for Evo (1.0) Air cooled in the exportred modbus registers.

trimixdiver1, 2018-11-30:
Thanks

One more question

The status register values, mine says 9 right now and Im unable to find it in the code. Do you have a list of status and alarm codes?

jgyates, 2018-12-01:
No problem.

I am not exactly sure what you mean by "status" register since there are several status registers (engine / switch status, input status, output status). I am assuming you mean register 0001 which is the engine / switch state. In the code, this is treated as a single 64 bit value that combines reading registers 0001 and 0002 in one modbus read. The code that deals with this is in the functions GetSwitchState, GetEngineState and GetAlarmState. Based on the info you gave, I would assume the 9 means that the generator is stopped or an unknown alarm is present depending on if you are referint to register 0001 or 0002.

Alarm codes are represented in two different ways
1) there is an Error Code (E-Code) that is documented by Generac. The E-Code is written to the alarm log and the last / current E-Code is read from register 05f1. The E-coder is a human readable decimal value. The software looks up this value in /genmon/data/alarms.txt to interpret the E-code.
2) The engine / switch status uses the high bytes for the engine status (See GetSwitchState)(register 0001) and the low bytes (register 0002) for the switch state (see GetSwitchState) and and alarm code (See GetAlarmState) .

trimixdiver1, 2018-12-01:
I thought if I look at 0001 and 0002, the value would relate to the status and errors listed in the code you linked the screen shot of for me. 
Im not a programmer, just PLC stuff. So im lost on some of this.

gm58, 2018-12-01:
@trimixdiver1 

I'm also a PLC guy so I'm hoping you will share the logic to capture these registers.  Thanks in advance.

trimixdiver1, 2018-12-01:
Thanks, thats just what I needed. On Gen time (hrs,min) Im getting numbers like 2061, what does this relate to? There is no 2061hrs.
Thanks

jgyates, 2018-12-01:
I am assuming you are reading register 000e? My guess is that you are either reading the wrong register (time is 000e) or 2061 is a decimal value. The decimal value 2061 is 0x080d in hex. The high byte (0x08 is the hour (0 - 0x17 or 0 - 23) and the low byte 0x0d is the minute (0x00 to 0x3b or 00 - 59).

trimixdiver1, 2018-12-03:
Looking good!
One more question, how is 001b Service A Date, read? Its a 3 digit number.
Thanks

jgyates, 2018-12-03:
Take the register value and multiply by the number of seconds in a day, then add the number of seconds in a day (add one more day). This will give you a Unix / POSIX time stamp (https://en.wikipedia.org/wiki/Unix_time).

jgyates, 2018-12-03:
Closing this issue, you can still post questions to this thread.

closed 2018-12-03 by jgyates

jgyates, 2018-12-03:
Probably, the unix time stamp is the number of seconds since 1/1/1970.

trimixdiver1, 2018-12-03:
Have you ever got that?

jgyates, 2018-12-03:
No, but I have never ran across a generator that has not run yet. You can easily validate this by looking at the info on your generator. It should be under Maint -> Scheduled and it should have a date for the next scheduled maintenance. Note the generator will only show the date for maintenance A or B depending on which one comes first.