jgyates/genmon #214: Feature request- full backup button

jcherney, 2018-12-02:
Feature request

Hi,

Would it be possible to include an option for a full and complete backup of the software? I have a backup of my existing system, but changes are made often (thank you!) . It would be nice to click a link and backup the latest and greatest. Props if it can be done as an ISO image to immediately download to a new card.....

jgyates, 2018-12-03:
Hi @jcherney 

Good idea. An ISO is doable, but only really feasible if you save it to a USB drive since there may not be enough spare space to write the ISO image. Saving the config and data files and downloading them from the gui is relatively easy. Let me think on the best way to do this.

liltux, 2018-12-03:
@jgyates maybe instead of a backup button, but a way to export the configuration files to a download prompt.  Easy way to get the files from the pi to the pc.  I use winscp, which works great, especially with putty.  Just an idea.  I don’t think creating an image of the pi from the pi would work the best, resource issues.  But the config files backed up should make for seamless rebuild.  Maybe even an option to load said config files from WebUI.

jgyates, 2018-12-05:
I checked in an update adds a Backup button to the about page. this will add all of the configuration files, the power log and outage log to a compressed archive (tar.gz) and download the file from the browser.

Using this method you can restore a given installation. The /etc/genmon.conf file contains the version of the software you are current using so you can always download that from github.

For a full system backup I would suggest making a clone of your SD card by powering down your pi, unplugging the card and put it into a PC. There are many sites that have directions (depending on your OS) on how to make an ISO from an SD card.

I will close this issue but feel free to comment here if you have issues or questions.

closed 2018-12-05 by jgyates

liltux, 2018-12-08:
@jgyates used the backup button, works great!

rob7419, 2020-11-24:
What is the process to restore from a backup file? I've had to start my pi with a fresh install. Can't find any mention of this in the WiKi and apologize if I've missed it.

jgyates, 2020-11-24:
Good question. I don't think that is documented anywhere. I will add this to a wiki page (probably the tips and tricks page) soon. The following assumes you are using the default settings (i.e. your config files are stored in the /etc/genmon folder on the pi:

First take the file that you downloaded from the web interface and transfer it to your pi. You can do this in a few different steps but I like to use the scp program. This commaond will copy the file genmon_backup.tar.gz to the /home/pi folder on a pi at IP address 192.168.1.10

     scp genmon_backup.tar.gz pi@192.168.1.10:./

Then log into the pi with ssh. You should log into the /home/pi directory if you log as  the pi user. Once you are logged in type:

      tar -xzf genmon_backup.tar.gz

This will unpack the archive file into a folder named genmon_backup. The contents of this folder should be copied to the /etc/genmon folder. You will need root privileges (sudo) since you are copying to the /etc/ folder.

     sudo cp ./genmon_backup/*.* /etc/genmon

Then restart genmon:

       cd genmon
       ./startgenmon.sh restart

or reboot

       sudo reboot

The backup includes the power log, outage log, service journal, and all config files. If you only want the config files you can on copy the *.conf files instead of *.*.

jgyates, 2020-11-24:
https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon