jgyates/genmon #215: Scaling CT Evo LC

n4ziq, 2018-12-07:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs to the developer, if you email is setup and working properly you can issue this command in your browser:

     http://MYIPADDRESS:8000/cmd/sendlogfiles

Replace MYIPADDRESS to the IP address of you pi and port 8000 if you are using a port other than the default.

You can submit your registers from the Monitor page on the web interface.

### Expected Behavior
Should get 0 KW when no load.
current divisor override value should correct scaling of current.  

### Actual Behavior

Get 18.8KW no load. Scaling is incorrect when it is running. 
Tried inputting current divisor but it does not override. Calculated should be 0,516. 

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: RG02724ANAX
- Generator Registers: 0058
- Genmon Version: V1.11.30
Hardware Version V1.04
Firmware Version V1.36

jgyates, 2018-12-07:
@n4ziq 

You should get a kw reading when the generator transfer switch is not engaged. In practice it ends up being about 20% of the rated load of the generator. This thread should explain this behavior: https://github.com/jgyates/genmon/issues/112

In your comments you said "Tried inputting current divisor but it does not override. Calculated should be 0,516." To help I would need to know which value you are using for each parameter, not just the values. Also, how are you calculating these values and what is the reading you are taking for current?

I am assuming you have read #5 on this page: https://github.com/jgyates/genmon/wiki/Appendix-4-Known-Issues

I will be happy to help, but I will need more info.

liltux, 2018-12-07:
@jgyates , alot of these CT, current, and KW problems I feel may be due to the units not having CT's.  Not all units have Current transformers installed.  This is not identified by the unit Model number or owners manuals.  I am looking at an Evo LC Generator right now, and it does NOT have CT's.  So before you can calibrate or offset CT's you need to make sure that the unit in question has CT's installed.  The Evolution Controller monitors current on its AC Excitation winding, wires 2 and 6, to monitor overload conditions (Hall Effect Module in Controller).  If your unit has CT's, they would be located on wires 11 and 44 (if single phase located on the Main Line Circuit Breaker of the generator) or L1, L2, L3 (if 3 phase).  They would be small donut shaped objects with these wires passing through them.  What I am afraid of is this register is a place hold for units with CTs installed and units without CTs  you are monitoring an analog signal on an open circuit.  There by it is erratic.

n4ziq, 2018-12-07:
I assumed there were CT’s… But your statement seems to clarify why the CT register had a reading under no load since it is coming from the excitation. With no load the web display from genmon had ~20KW and the register (58) had a decimal value of 182. When I put load on the generator the display dropped to ~16kw with a real load of 20 amps and register value of 177. When I measured 16 amps the register value was 175.

 

I took the default current offset and calculated a new current divider at 0.516. That divider seemed to work for 16 amps and 20 amps.

jgyates, 2018-12-07:
@n4ziq 

You may not have CTs in your generator, however, empirical evidence shows you will still see an output of the CT (assuming you have them) even with no load. In the link I posed above there is a diagram that I am reposting here:

[image]

This demonstrates the minimum load on the generator while under load. The minimum appears to be present with no load. I admit that this is observational so there may be some info I am missing, but there appears to be a dummy load on the generator when running with no load or minimum load. I have seen some discussion about how it is not healthy for a generator to run with no load and this may be why this dummy load is present. 

Another thing to keep in mind is even if you do have CTs the firmware in the generator appears to only read the CT values when the engine is running, but you can always read the modbus register. If you note the value of register 0058, it retains the last value when the generator was running. As a result the genmon software will ignore the register and return zero for current unless the generator is running.

jgyates, 2018-12-07:
Another thought is that this register may not represent the CT output, but rather the Hall Sensor reading taken by the firmware. From genmon's perspective this is just a value that appears to have a linear relationship with the current measured under load. When creating the current output functionality (power is derived from this calculation), for the EvoLC, the output of register 0058 was compared to the measured current on E1 and E2 while under load and a linear equation was created to match the observation. In short, I don't know if register 0058 maps to CTs or a Hall Effect Sensor. The relationship between  0058 and the current output does appear to vary between models, however, based on the data I have it does appear to be linear.

Now that I know where the CTs are (thanks @liltux ) I will see if my EvoLC has CTs. I will let you know once I validate this.

jgyates, 2018-12-07:
The EvoAC current is read from registers 003a and 003b making one long value. The EvoLC reads from 0058.  So what *could* be the case is that EvoAC reads CTs from registers 003a and 003b and EvoLC reads a Hall Effect from 0058.

liltux, 2018-12-07:
I would agree that is the more likely event.

jgyates, 2018-12-07:
I do not have CTs on my system so I am assuming that 0058 is the Hall Effect Sensor. I don't see that this changes the code at this point, but I will update the documentation to reflect this. Since the Hall sensor is related to the rotor turning this is still a reasonable way to estimate fuel usage and this also explains the reading when no load is present. I have a diesel unit so my fuel calculation (based on the Hall Sensor for current / power) is very close to the actual fuel in the tank.

What is still odd to me is that the data from EvoAC users also show a current output when not load is connected. The diagram I posted above is form an EvoAC.

liltux, 2018-12-07:
I would say the Hall Effect is a good assumption and should be linear with load and usage.  I just wanted to clarify it is not exactly load but is affected by the load.  I also believe the problem with no load feedback on the CT's, the actual physical current transformers, is possible just EMF noise from the generator.  Once there is enough current flowing through the wire, 11 or 44, it produces enough of its field to overcome the noise.  Again assuming, I have not tried measuring this, but locating the wires from the CT's (399A and 399B) and measuring voltage would prove this.  Under no load, no current flow (i.e. MLCB off)  there should be theoretically no voltage induced on the CT's.  But if a minimum amount of voltage is present, this would indicate the induction of other EMF in the generator onto the CT's.

jgyates, 2018-12-07:
That sounds very reasonable to me. Thanks @liltux . As usual you are a wealth of info.

n4ziq, 2018-12-08:
> Since I don't know your background, be advised that messing around in your transfer switch without knowing what you are doing can literally kill you. Having said that, what I did to calibrate the software was:
>
> 1.	take a current reading on E1 and E2 of your transfer switch with the generator under load. Add the two values to get a total actual current. I used a clamp on style multi-meter.
> 2.	then take the decimal value of register 0058 with the load remaining the same as in step one
> 3.	Then use the above data points to determine values of currentdivder and currentoffset:
>
> Current Output = (Modbus Register Values / currentdivider) + currentoffset)
>
> You can also repeat steps 1 and 2 at two different loads and it may be easier to determine the values using the formula for a straight line https://www.mathsisfun.com/straight-line-graph-calculate.html except use the inverse to divide instead of multiply.

I calculated an offset of 334 and divisor of 0.5.

 

16 total amps had a register 58 value of 177

20 total amps had a register 58 value of 175

Using the simultaneous equation gave me a slope of 2 and intercept of 334. 

 

The genmon config file states the offset and divisor are for EVO air cooled units. Is there another location I should be entering these numbers? Do I need to hard code the config file from EVO LC to EVO AC to have it read the override values? 

 

Thanks,

 

Joe

jgyates, 2018-12-08:
That comment in the conf file needs to updated. AC and LC use the same entries. You can also enter them from the advanced page (click the gear icon in the upper right  corner of the web interface twice).

If you are good with these values I can make these the default for a liquid cooled 27kw model.

n4ziq, 2018-12-08:
I must be doing something wrong. The offset value changes the power output values, but the divisor value does almost nothing. I see register 58 changing, but it does not show up on the power output or the calculated amps. I should be using 20amps or 4800 watts based on register 58 value of 177, but it is showing 14.12KW instead. I used offset -334 and put 0.5 in for the divisor. 

 

    Engine : 

        Switch State : Auto

        Engine State : Running

        Active Relays : Transfer Switch Activated, Fuel Relay On, Bosch Governor On

        Active Sensors : 

        Battery Voltage : 13.40 V

        Battery Status : Not Charging

        RPM : 1806 

        Frequency : 60.00 Hz

        Output Voltage : 239 V

        Output Current : 59.07 A

        Output Power (Single Phase) : 14.12 kW

        Active Rotor Poles (Calculated) : 4 

 

    Line State : 

        Transfer Switch State : Generator

        Utility Voltage : 246 V

        Utility Voltage Max : 244 V

        Utility Voltage Min : 244 V

        Utility Threshold Voltage : 144 V

        Utility Pickup Voltage : 192 V

        Set Output Voltage : 240 V

jgyates, 2018-12-08:
For liquid cooled controllers, the default values of current divider is 0.44 and the default value of current offset is -323.16. 

I am a little confused as to what you are describing,.. are you saying that you see a value of 177 for register 5b but the current output is not match the calculation? or are you saying that the current output stays the same value and does not change? or are you saying both are happening.
 
Assuming you are saying the value of 177 doers not match the displayed value of 58A:
If register 005b is changing the output current will change also (and power). Keep in mind that the web interface is taking a snapshot every 3-5 seconds and the register page takes a snapshot every 3-5 seconds. The actual register values change change between the 3-5 seconds. If you see 005b at 177 for a moment but then it changes and you then look at the current output it will change also so there is no direct correlation between the register page and the status page (other than things will likely be in the same ball park). 

the way the software is designed is that a thread will read all the relevant registers and cache them. As request are made by the web interface or other software, calculations are made on the cached values. In the web interface we have one timer that updates the register page and another that displays text info like current and voltage and a third that updates the graphs / gauges. All of these values are not synchronized with a single register read so seeing differences is somewhat expected. As long as the range of data that you are seeing is within the bounds: You said that reg 5b is change, what is the range and if you do the calculation of a given sample, is the calculated value in the range of the values displayed?

Let me know if this does not make sense or if I am not understanding your question.

n4ziq, 2018-12-09:
It could be that I am misunderstanding how the application works. I apologize for all the questions. I was under the impression that CT Output (EvoLC) 0058 is the register used by the software for amperage and that the offset and divisor were applied to this value? Then I thought the software multiplied the calculated value by the voltage to produce the KW value as displayed on the web interface? Register 0058 is changing as the load changes. The problem I have is that I can’t get the correct KW value displayed even when I put in what should be the correct offset and divisor via the web interface . Since I couldn’t seem to get the correct KW value to display on the webpage, I tried different numbers for offset and divisor to see what impact that would have on the displayed KW value on the webpage. I noticed that my divisor values seemed to have little or no impact on what is displayed on the webpage for KW. I wonder if my config file has something wrong since my fuel gauge is 0 and won’t reset and that me be a symptom of something related? I think I manually put in EvoLC rather than let it read the controller and autoconfig?

jgyates, 2018-12-09:
No problem on the questions. I am actually glad I am getting another Liquid Cooled data point. 
You can submit your log files to me on the about page. This will let me know if something is wrong with the setup. I would recommend not overriding the auto-detect.

From you sentence above : "The problem I have is that I can’t get the correct KW value displayed even when I put in what should be the correct offset and divisor via the web interface . "

How are you determining what the correct value is? By reading the current with a meter or by reading register 005b and doing the calculation?

The way you are describing the software working is correct. Based on the input @liltux added, we think the LC version have a Hall Effect Sensor instead of a Current Transformer (CT), so in in theory, we should be getting the same readings since the Hall Effect Sensor is in the controller and our generators use the same controller (I have a LC 48kw diesel).

n4ziq, 2018-12-09:
I put a clamp on ammeter onto the 2/0 copper cables going to the transfer switch from the generator while under load. I then copied down the values I was getting from each 2/0 leg. I summed the current measurements from the two cables together to get the total current. I then copied down the register 58 values next to the measured current values. Therefore, I had total current at 16 amps which was about 8 amps per leg and a register 58 decimal value of 175. When I hit 20 amps, the register 58 decimal value was 177. Based on those values, I came up with an offset of -334 and a divisor of 0.5.

jgyates, 2018-12-09:
I got your logs. I don't think this is relevant to your kw problem, but it could have do with your fuel reporting zero. It looks like you have an invalid setting for the "subtractfuel" config entry. This entry needs to be a floating point number, but you appear to have some text in the entry base on the error log:

     Error in MyConfig:ReadValue: subtractfuel: could not convert string to float

You can change this value from the advanced page (click the gear icon in the upper right twice). If you use the web interface it will validate the values before they are written.

Let me think on the issue you are seeing with current...

jgyates, 2018-12-09:
@n4ziq 

I think one if the issues you are having is your expectation of the values being read is not inline with the actual behavior. For example, when I read register 0058 from the command line I get the following:

    python ClientInterface.py 192.168.11.14
    OK : Auto, Running
    >generator: readregvalue=0058
    00ae
    >generator: readregvalue=0058
    00cb
    >generator: readregvalue=0058
    00c9
    >generator: readregvalue=0058
    00ca
    >generator: readregvalue=0058
    00ca
    >generator: readregvalue=0058
    00c9
    >generator: readregvalue=0058
    00c9
    >generator: readregvalue=0058
    00ca
    >generator: readregvalue=0058
    00be
    >generator: readregvalue=0058
    00bc
    >generator: readregvalue=0058
    00bc

These reads all took place with the course of about 2 seconds and the values range from 0x00ae to 0x00ca (174-202 decimal). I am currently running on generator power. Unless you have a fixed load, the values will vary within a range. This is likely due to a combination of the sensor type and the actual load fluctuations.

n4ziq, 2018-12-10:
First let me apologize for my persistent questions. I do appreciate all the work you put into the genmon software. I feel badly about pestering you. If I am way off on my thought process, please let me know. 

 

I uninstalled genmon and reinstalled it to ensure I had no corrupt files, but my results have not changed. I understand the fluctuation you are stating under load. Although, I decided to try and understand the software behavior I am experiencing. So, I ran the generator without load and it seems I get a constant decimal value of 183 from register 58. (Please verify that it is stable under no load since I did it via the webpage and not using realtime serial values) Since I had a constant state, I tried playing with the offset and divisor. No matter what value I put in for divisor, it stayed at 0.44. I tried entering new values via the webpage and directly into the config file located in the etc directory. Each time I restarted the raspberry pi to ensure the config loaded. The offset value does affect the amperage calculation, but the divisor had no effect. I am entering 0.5 and tried entering .5 into the divisor field, but neither worked. Am I not entering the value in the correct format?

 

The fuel gauge still shows 0 too… Strange. 

 

I will send my log file over too. Maybe you can see something I have done wrong. 

 

 

Thanks,

 

Joe

jgyates, 2018-12-10:
@n4ziq 

No worries about the questions. As a matter of fact your persistence has helped find a bug in the code. I checked in an update, try this and it should now respond to the divisor. I had a parameter check that was suppose to disallow divide by zero, but it was disallowing dividing by anything less than one. Error had no effect on Air Cooled units as the divisor there is typically greater than one, but on Liquid Cooled units it made it always use the default divisor. Sorry about that but again, thanks for helping.

I had an outage today and took some readings of my register 0058 and also took measurements for a pair of CTs that I have installed on an energy monitoring device from openenergymonitor.org. I will be analyzing this data next.

Let me know how this update works for you and also send your logs as your fuel should not be zero. What is the value of your Tank size parameter on the settings page?

jgyates, 2018-12-10:
For Liquid Cooled, the new default values for current divider is 0.425 and the default value for current offset is -323.31.

jgyates, 2018-12-10:
Logs look OK. 

If your fuel is still showing zero  and you have a valid entry for Tank Size let me know and we can look into it.

jgyates, 2018-12-10:
Question: What is the value of register 05ed and what is the current temperature at your location? This is not related to your issue but I wanted a data point on what I think is the ambient temp sensor for the LC units.

n4ziq, 2018-12-10:
5ed shows decimal 60, and temp is 57.5F or 14.2C

jgyates, 2018-12-10:
For tank size, just put 400, leave off the "Gal". This field is a number. If you have the setting  "Use Metric" disabled then the software assumes gallons, otherwise it is liters. the settings page should validate these values if you enter them from the web interface. If you edited the file directly then you get no validation.

n4ziq, 2018-12-10:
Tanksize variable shows a value of 400 in the genmon.conf file in the etc directory. 

 

Does usesensorforfuelgauge need to be False? The config file shows true.

 

(optional) - Use fuel level sensor for fuel gauge. This option if for

# Evolution Liquid Cooled Diesel Units only. If enabled this option will use

# the fuel level sensor for the web interface fuel gauge.

usesensorforfuelgauge = True

jgyates, 2018-12-10:
that option should not show up in the UI at all unless you have a Diesel unit. The setting should ve ignored unless you have a diesel unit. According to your logs you have propane, correct?  I you edited the /etc/genmon.conf file manually,  you may want to consider  starting with a fresh copy of the genmon.conf file and update it only from the web interface. there are no settings in conf file that can not be set in the web interface (with the exception of debugging specific settings). Also the web UI will validate the settings and not allow invalid parameters to be saved.

jgyates, 2018-12-10:
Does your generator use propane?

jgyates, 2018-12-10:
I checked in an update that added the propane fuel consumption polynomial for models  RG022, RG027, RG032, RG038, RG048

jgyates, 2018-12-14:
I am going to mark this issue closed, feel free to post to this thread if you have any questions on this issue.

closed 2018-12-14 by jgyates