jgyates/genmon #216: Non Issue - Program suggestion.

liltux, 2018-12-10:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs to the developer, if you email is setup and working properly you can issue this command in your browser:

     http://MYIPADDRESS:8000/cmd/sendlogfiles

Replace MYIPADDRESS to the IP address of you pi and port 8000 if you are using a port other than the default.

You can submit your registers from the Monitor page on the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: {Please write here}
- Genmon Version: {Please write here}

liltux, 2018-12-10:
@jgyates I am going to send you the logfiles.  something does not look right to me.  It was working saturday, but now is not.  Not sure, but would like a second opinion.  Thanks.

jgyates, 2018-12-10:
Sounds good.

There is a chicken and egg problem with the test email. If the user has not saved their settings then the backend would need to get the settings from the web interface, which is doable, just not the way the code is currently structured. Right now we just save settings to a file and restart to make changes. I will likely make this change at some point, it is just more time consuming than it appears to implement it correctly.

liltux, 2018-12-10:
It was just a thought, I have seen a few different user interfaces that have a similar feature.  By the way have you had a chance to look at the logfiles I sent?  I looked into it a little and it says it has to do with the emial formating but I did not change anything on the webUI between Saturday and Today.  I even went as far to run the "genmonmaint.sh install" to see if something had changed.  Nope, still getting syntax errors.  I just dont get it.  I refers to line 323 in mymail.py, which is around were the email is sent.

jgyates, 2018-12-10:
The last logs I got from your were for you 50kw unit yesterday. I have not received any today. I looked at line 323 and there is a bug there. I checked in an update. It looks like you are getting an  error fetching the email from the server and the error handler had a function that was removed a while back.

liltux, 2018-12-10:
That took care of it, thanks.  Was that problem related to me or do to one of the email recipients had an auto response turned on and the code didn't like it?

jgyates, 2018-12-10:
When trying to read IMAP email, an error was generated by the IMAP library (probably due to the auto response I assume, not exactly sure at this point what causes the initial error condition). The way I was handling that error was just bad code that should not be there.

jgyates, 2018-12-10:
If you have any other errors  they will be reported in /vat/log/mymail.log

liltux, 2018-12-10:
Sounds good, thanks.

closed 2018-12-10 by liltux