jgyates/genmon #217: Did I just break my controller?

AMTNate, 2018-12-14:
I made my cable yesterday and decided to hook everything up for the first time. I ended up having wifi problems so I disconnected everything and brought it all back inside. This morning, I wake up to my generator constantly trying to crank for a what I can only assume had already been a few minutes. I realized that when I had soldered the connections to the max3232 board, I reversed the RS232 and TTL sides of the board. Would doing this have damaged my controller in any way?

liltux, 2018-12-14:
@AMTNate I would doubt that damaged your board.   What kind of generator do you have?  Controller type?  Do you have ant problems, I say this because I have seen ants get into the control boards and cause similar issues.  If your wiring damaged the board I would imagine the board would have acted up yesterday.  You may have loosened one of the other connecters and it is not seated completely?  Just some thoughts.

AMTNate, 2018-12-15:
As a side note, we did recently have a problem where it didn't start during an actual power outage. the company that maintains it said that the starter overspeed sensor had failed, and they replaced that. I know it was working because it ran its weekly schedule Sunday afternoons like it is supposed to.

AMTNate, 2018-12-16:
[image]

Update: I opened up the controller, and I don't really know what I'm looking at, but it doesn't look right. thats for sure. Going to call my service dealer tomorrow.

jgyates, 2018-12-16:
thanks for the feedback. I will close this thread however feel free to post here if you have any questions related to this topic.

closed 2018-12-16 by jgyates