jgyates/genmon #219: kW graph, fuel usage/hour

wrvolz, 2018-12-22:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs to the developer, if you email is setup and working properly you can issue this command in your browser:

     http://MYIPADDRESS:8000/cmd/sendlogfiles

Replace MYIPADDRESS to the IP address of you pi and port 8000 if you are using a port other than the default.

You can submit your registers from the Monitor page on the web interface.

### Expected Behavior

3 requests:
1) on the Status->kW output graph, allow scrolling back in time, How far? Don't know, configurable perhaps? Would like to look at use from past outages
2) On Maintenance page, under maintenance, add run time for the last 30 days. This allows calculation of fuel usage/hour
3) On Outage page, under Outage, add fuel usage for the outage. again fuel usage/hour under load.

Note that my generator runs on propane and the tank also supplies my house which this monitor does not account for. Fuel usage/hour gives me an indication of estimated run time based on my current tank level. Thanks, great tool 

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: G0062500
- Generator Registers: {Please write here}
- Genmon Version: V1.12.1

jgyates, 2018-12-26:
The Maintenance page now has the run hours in the last 30 days.

By the way, the program [kwlog2csv.py](https://github.com/jgyates/genmon/wiki/1----Software-Overview#otherappskwlog2csvpy-optional) will allow you to export the kw log to a CSV file, which you can import into Excel or any other spreadsheet application.

I don't think I will add item one in your list due to the complexity vs benefit tradeoff. 

I have the code in place to handle number 3, I just need to settle on how to handle the previous format of the outage log. This also makes the outage log format generator dependant since not all units support the power log. In shot, I am still thinking about the best way to format the log for this change.

wrvolz, 2018-12-27:
> AVG logo <https://www.avg.com/internet-security>
> This email has been checked for viruses by AVG antivirus software.
> www.avg.com <https://www.avg.com/internet-security>

The scrolling KW graph would be nice, but not a necessity, especially 
since I can export the data. Thanks for the 30 day run time/fuel usage. 
It will be useful. The fuel usage on the outage log would be nice.

That said, the 30 day fuel usage would also include the fuel used during 
weekly system exercising. Since the generator doesn't run at full speed 
at this time, the fuel usage/hour from the 30 day would be a little low. 
Hence the need for fuel usage during an outage. Right now it appears 
that my generator eats about 2 gal/hour of propane.

jgyates, 2019-01-10:
@wrvolz 

I just checked in version V1.12.10 which adds the fuel consumption to the outage log. Let me know if you have any issues. Works as expected in my testing.

I am going to close this thread for now. Feel free to comment here if you have any other input on this subject. 

Thanks for the input!

Thanks
Jason

closed 2019-01-10 by jgyates