jgyates/genmon #220: Z-Wave interface for integration to home hub

proppilot, 2018-12-25:
Been thinking about different solutions to provide a z-wave interface to the RPi used in Genmon and finally came up with a (relatively) cheap solution.  I purchased a Z-Uno Arduino based interface board for $ 69 USD.  There were enough examples of code on their website and user forum to be able to relatively easily figure out how to program a sketch (C code) that would interface with the GPIO pins on the RPi and be driven by the gengpio.py and gengpioin.py programs on this project.

https://z-uno.z-wave.me/technical/

I am using a VERA PLUS hub and it easily created a device group that can accept the outputs from the PI and send inputs to the PI via the VERA hub.

[image]

**Z-UNO (sketch) code that drives the inputs and outputs**
____________________________________
/* 
This sketch is used to interface the Genmon Raspberry Pi microprocessor to the Z-Uno Z-wave Arduino processor to provide a 
Z-wave interface between the generator and the VERA Plus home controller

                  PI     Z-Uno  
STATUS_READY      16      17  
STATUS_ALARM      18      18  
STATUS_SERVICE    22      19
STATUS_RUNNING    26      20  
STATUS_EXERCISING 24      21  
STATUS_OFF        21      22  
          
INPUT_STOP        11      9 
INPUT_START       13      10  
INPUT_TRANSFER    15      11  

Off - LOW
On - HIGH
*/

// Pin definitions

#define BINARY_PIN1 17
#define BINARY_PIN2 18
#define BINARY_PIN3 19
#define BINARY_PIN4 20
#define BINARY_PIN5 21 
#define BINARY_PIN6 22
#define LedPin1 9
#define LedPin2 10
#define LedPin3 11

// Global variables to store data reported via getters// Global variables to store data reported via getters

byte lastBinaryValue1;
byte lastBinaryValue2;
byte lastBinaryValue3;
byte lastBinaryValue4;
byte lastBinaryValue5;
byte lastBinaryValue6;
byte switchValue1 = 1;
byte switchValue2 = 1;
byte switchValue3 = 1;

ZUNO_SETUP_SLEEPING_MODE(ZUNO_SLEEPING_MODE_ALWAYS_AWAKE);

// Send Basic Set to association group 

// ZUNO_SETUP_ASSOCIATIONS(ZUNO_ASSOCIATION_GROUP_SET_VALUE); 

// Set up 3 output channels and 5 input channels

ZUNO_SETUP_CHANNELS(
  ZUNO_SENSOR_BINARY(ZUNO_SENSOR_BINARY_TYPE_MOTION, getterBinary1),
  ZUNO_SENSOR_BINARY(ZUNO_SENSOR_BINARY_TYPE_MOTION, getterBinary2),
  ZUNO_SENSOR_BINARY(ZUNO_SENSOR_BINARY_TYPE_MOTION, getterBinary3),
  ZUNO_SENSOR_BINARY(ZUNO_SENSOR_BINARY_TYPE_MOTION, getterBinary4),
  ZUNO_SENSOR_BINARY(ZUNO_SENSOR_BINARY_TYPE_MOTION, getterBinary5),
  ZUNO_SENSOR_BINARY(ZUNO_SENSOR_BINARY_TYPE_MOTION, getterBinary6),  
  ZUNO_SWITCH_BINARY(getterSwitch1, setterSwitch1),
  ZUNO_SWITCH_BINARY(getterSwitch2, setterSwitch2),
  ZUNO_SWITCH_BINARY(getterSwitch3, setterSwitch3)
);

// set up I/O pins. 

void setup() {
  pinMode(BINARY_PIN1, INPUT_PULLUP);
  pinMode(BINARY_PIN2, INPUT_PULLUP);
  pinMode(BINARY_PIN3, INPUT_PULLUP);
  pinMode(BINARY_PIN4, INPUT_PULLUP);
  pinMode(BINARY_PIN5, INPUT_PULLUP);
  pinMode(BINARY_PIN6, INPUT_PULLUP);
  pinMode(LedPin1, OUTPUT);
  pinMode(LedPin2, OUTPUT);
  pinMode(LedPin3, OUTPUT);
}

void loop() {
  byte currentBinaryValue1 = digitalRead(BINARY_PIN1);
  byte currentBinaryValue2 = digitalRead(BINARY_PIN2);
  byte currentBinaryValue3 = digitalRead(BINARY_PIN3);
  byte currentBinaryValue4 = digitalRead(BINARY_PIN4);
  byte currentBinaryValue5 = digitalRead(BINARY_PIN5);
  byte currentBinaryValue6 = digitalRead(BINARY_PIN6);

  if (currentBinaryValue1 != lastBinaryValue1) {
    lastBinaryValue1 = currentBinaryValue1;
    zunoSendReport(1);  
  }
  if (currentBinaryValue2 != lastBinaryValue2) {
    lastBinaryValue2 = currentBinaryValue2;
    zunoSendReport(2);  
  } 
  if (currentBinaryValue3 != lastBinaryValue3) {
    lastBinaryValue3 = currentBinaryValue3;
    zunoSendReport(3);  
  }
  if (currentBinaryValue4 != lastBinaryValue4) {
    lastBinaryValue4 = currentBinaryValue4;
    zunoSendReport(4);  
  }
  if (currentBinaryValue5 != lastBinaryValue5) {
    lastBinaryValue5 = currentBinaryValue5;
    zunoSendReport(5);  
  }
  if (currentBinaryValue6 != lastBinaryValue6) {
    lastBinaryValue6 = currentBinaryValue6;
    zunoSendReport(6);  
  }
}

// Getters

byte getterBinary1() {
    return (lastBinaryValue1 == 0) ? 0 : 255;
}
byte getterBinary2() {
    return (lastBinaryValue2 == 0) ? 0 : 255;
}
byte getterBinary3() {
    return (lastBinaryValue3 == 0) ? 0 : 255;
}
byte getterBinary4() {
    return (lastBinaryValue4 == 0) ? 0 : 255;
}
byte getterBinary5() {
    return (lastBinaryValue5 == 0) ? 0 : 255;
}
byte getterBinary6() {
    return (lastBinaryValue6 == 0) ? 0 : 255;
}

// Setters

void setterSwitch1(byte value) {
  digitalWrite(LedPin1, (value > 0) ? HIGH : LOW);
  switchValue1 = value;
}
byte getterSwitch1(){
  return switchValue1;
}
 void setterSwitch2(byte value) {
  digitalWrite(LedPin2, (value > 0) ? HIGH : LOW);
  switchValue2 = value;
}
byte getterSwitch2(){
  return switchValue2;
}
void setterSwitch3(byte value) {
  digitalWrite(LedPin3, (value > 0) ? HIGH : LOW);
  switchValue3 = value;
}
byte getterSwitch3(){
  return switchValue3;
}

jgyates, 2018-12-25:
@proppilot , Very nice!!

proppilot, 2018-12-28:
Here is the pinout between the Zuno Z-wave board and the RPi

[Z wave generator interface pinout.xlsx](https://github.com/jgyates/genmon/files/2713913/Z.wave.generator.interface.pinout.xlsx)

Here is the multi channel device in my VERA Plus.  The INPUTS from Genmon look like motion detector inputs in the Vera until I figure out how to do something more eloquent.  

[image]

Now if the generator starts, I can send notifications via the z-wave hub and even better can trigger some scenes and/or activities

Not a major practical use but sure fun to do !

proppilot, 2018-12-28:
[image]

proppilot, 2018-12-28:
Hi Jason,  I have a question.  I can see both gengpioin and gengpio are running in htop.  I am reading the outputs from the GPIO to the Z-wave board in the home controller properly however sending commands from the home controller to START STOP or START/TRANSFER are having no effect.

The outputs from the Z-wave board are through a 120k resistor to ground (pulled down) and the commands pull them high.  I wrote some test python code yesterday that triggered a GPIO output based on receiving a GPIO input from the Z-wave module and it worked 100% so I figured it is electrically working.

I tried to find more detail around the gengpioin program but was not successful.

Question is : when I pull GPIO27 (start) high should the generator just start or am I missing something ?

Thanks, Tony

jgyates, 2018-12-28:
Hi Tony,

The gengpioin.py program sets up the inputs to triggered on a falling edge (high to low transition) to keep the command from being sent the entire time the signal is high or low. Also the program sets the internal pull up resistor in the pi active for each gpio input. In the code for gengpioin will see three sections like this that set both the pull up resistor and the falling edge detection:

     # input, set internal pull up resistor#
     GPIO.setup(INPUT_START, GPIO.IN, pull_up_down=GPIO.PUD_UP)             
     GPIO.add_event_detect(INPUT_START, GPIO.FALLING)                        # detect falling edge
     GPIO.add_event_callback(INPUT_START, callback = StartCallBack)

Let me know if you have any questions.

jgyates, 2018-12-28:
Changing the code is simple, however the problem is that if I change the default behavior then it will effect others who are using it. I could make some options that are configurable via a conf file that would allow you to override the default.

proppilot, 2018-12-28:
If you could point me to where the code would change / changes I could do a local version.  

Just need to understand what the differences would be vs you having to add features for one user !

Regards,

Tony Varga

jgyates, 2018-12-28:
I checked in an update that will give you the options to change the GPIO input trigger and pull up/down config. Let me know if you have any issues.

proppilot, 2018-12-28:
Outstanding !

Regards,

Tony Varga

proppilot, 2018-12-28:
> <reply+02039b7a673a0efd930f8257dc8c9120d943d6bbb97eac9592cf00000001183e36159
> 2a169ce177b537d@reply.github.com>
> <state_change@noreply.github.com>
> (#220)

Jason.  WOW.  Donation made.  Thanks very much !

1TzNLKwks5u9muVgaJpZM4ZgzJT> .
b","name":"GitHub"},"entity":{"external_key":"github/jgyates/genmon","title"
:"jgyates/genmon","subtitle":"GitHub
repository","main_image_url":"https://github.githubassets.com/images/email/m
essage_cards/header.png","avatar_image_url":"https://github.githubassets.com
/images/email/message_cards/avatar.png","action":{"name":"Open in
GitHub","url":"https://github.com/jgyates/genmon"}},"updates":{"snippets":[{
"icon":"PERSON","message":"@jgyates in #220: I checked in an update that
config. Let me know if you have any issues."}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/220#issuecomment-4504
"https://github.com/jgyates/genmon/issues/220#issuecomment-450411025",
"url": 
"https://github.com/jgyates/genmon/issues/220#issuecomment-450411025",
"name": "View Issue" }, "description": "View this Issue on GitHub",
"publisher": { "@type": "Organization", "name": "GitHub", "url":
"https://github.com" } } ]

proppilot, 2018-12-28:
Hi Jason, even with the change I cannot trigger a START via the GPIO.  With my meter I am able to see GPIO17, 22 and 27 at zero volts.  When I toggle the Z-wave board to pull GPIO27 high, I read 3.0 volts but nothing happens.  Cleanly moving from zero to 3 with the toggle.  Set up the new options to disable the pull-up resistor and trigger on the rising edge.  

Any thoughts ?

jgyates, 2018-12-29:
I found a bug that would keep the config file from being read, I checked that in. I also added some logging that may be helpful to /var/log/gengpioin.log. It will log the settings used and when an input event is detected.

Once you try this out look to see if there are any errors logged to /var/log/gengpioin.log? You can send me your log files and I can take a look.

Let me know how it goes..

proppilot, 2018-12-29:
Still no joy.  I sent you my logs for a peek.  I can see that it is logging the change of configuration but is not picking up any trigger.  I assume that a positive trigger should show in the log ?  It is cold and dark now so I will pull the unit into the house tomorrow and play with manually triggering the GPIO with jumper leads to see if/what shows in the logs.  If I see something in the logs when done manually, it may mean that the 3v output of the Zuno board is not enough (which I think is unlikely) but I am not sure.  

Let me know if you see anything in the log and confirm that the log will show a trigger and then I can do some more hardware debugging after sunrise

jgyates, 2018-12-29:
you should be able to type:

    cat /sys/class/gpio/gpio27/value

and it report the value of GPIO27 

I never received your log files.

jgyates, 2018-12-29:
This assumes that you have already ran the gengpioin.py program and setup GPIO27 as an input

proppilot, 2018-12-29:
> <reply+02039b7a39ea27a456b92d0dbba5801782e5e11c891efff592cf00000001183ebf9d9

Hi Jason, check for logs now.  I had a password update to my smtp account
that I had not updated in Genmon.

Thanks for the other replies.  I will do some testing.

Date:  Friday, December 28, 2018 at 11:52 PM

QquMKQSks5u9vUdgaJpZM4ZgzJT> .
"icon":"PERSON","message":"@jgyates in #220: This assumes that you have
input"}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/220#issuecomment-4504
"https://github.com/jgyates/genmon/issues/220#issuecomment-450465667",
"https://github.com/jgyates/genmon/issues/220#issuecomment-450465667",
"https://github.com" } } ]

jgyates, 2018-12-29:
No logs received. You can test the outbound email on the settings page as of V1.12.2

proppilot, 2018-12-29:
Strange.  I did do that test and received an email.

I then went to the page in Genmon and clicked ‘send logs’

I did not get any confirmation etc when I did that.

Regards,

Tony Varga

proppilot, 2018-12-29:
> <reply+02039b7a2ac5e69e93131d8ffec98a8d884d2aaa7c863f2e92cf00000001183f563f9

My bad.  I was changing but not saving at the bottom.  You should have the
logs now ?

Date:  Saturday, December 29, 2018 at 10:35 AM

u9HnnjVks5u94u_gaJpZM4ZgzJT> .
"icon":"PERSON","message":"@jgyates in #220: No logs received. You can test
"https://github.com/jgyates/genmon/issues/220#issuecomment-450500727",
"https://github.com/jgyates/genmon/issues/220#issuecomment-450500727",
"https://github.com" } } ]

jgyates, 2018-12-29:
Nothing is jumping out as wrong concerning gengpioin.py. In your logs you sent you don't appear to be receiving data (assuming you have it disconnected for GPIO testing).

proppilot, 2018-12-29:
Yes, the box is now sitting beside me in the house so I can troubleshoot logic levels

Not related to this but somewhat, I did get the mail function working again however looking in mymail.log this message is repeating.  Is it a setup issue or normal ?

[image]

proppilot, 2018-12-29:
So I was able to confirm that toggling the bit in my home controller is being properly read by the GPIO

[image]

proppilot, 2018-12-29:
All three inputs are zero and I confirmed that as I trigger each one, they go to 1 in the GPIO

[image]

jgyates, 2018-12-29:
This error is due to the incoming (IMAP) mail setting. The Incoming Mail Folder from the settings page is not found. There needs to be a folder on your mailbox with the same name as the Incoming Mail Folder setting.

Not sure what is going on with the gpio, let me look into it...

proppilot, 2019-01-03:
Hi Jason, did you get a chance to look at the GPIO issue at all ?

jgyates, 2019-01-03:
HI Tony,

I tried it out this morning and it is not working for me either. I am testing with just a jumper connected to 3.3V. I will let you know once I am able to determine what the problem is. I am testing with a Pi3B+

jgyates, 2019-01-03:
Tony,

I checked in an update that works for me. I also added a new parameter to allow you to set the software debounce time (the time in milliseconds that you will not recieve back to back notices of a GPIO change for a given input). Let me know how it goes for you.

proppilot, 2019-01-04:
Hi Jason, just deployed the change and it works perfectly.  Curious what you found ?  I really appreciate you resolving this !

jgyates, 2019-01-04:
Hi Tony, glad to hear that it is working.

I had originally written both of the GPIO programs (in and out) for another user. I was mistakenly under the impression that he had tested both programs, but it appears that gengpioin.py likely never worked as expected to start with. The call back function needed an input parameter of the channel to be called correctly. The error checking did not pick up this error because it was in a thread within the GPIO library. It was good that we went through this exercise as now the gpioin support has a lot more configurable options.

I will close this thread for now. Feel free to comment if you have any more input on this topic.

closed 2019-01-04 by jgyates

proppilot, 2019-01-04:
> <reply+02039b7a40baf62b3290b014affcf7e9bf0eb45804a5ed7992cf00000001184761469

Thanks Jason.  We are pushing into some of the more obscure features of
GenMon (I was early in and amazed at how far it has come) so good to knock
them out.  

Date:  Friday, January 4, 2019 at 1:00 PM

b1k5GbRks5u_5bGgaJpZM4ZgzJT> .
"icon":"PERSON","message":"@jgyates in #220: Hi Tony, glad to hear that it
is working.\r\n\r\nI had originally written both of the GPIO programs (in
configurable options.\r\n\r\nI will close this thread for now. Feel free to
comment if you have any more input on this topic."}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/220#issuecomment-4515
"https://github.com/jgyates/genmon/issues/220#issuecomment-451520132",
"https://github.com/jgyates/genmon/issues/220#issuecomment-451520132",
"https://github.com" } } ]