jgyates/genmon #221: Open Weathermap SSL warning

zekyl314, 2018-12-27:
### Expected Behavior
When using Chrome and SSL certificate is installed and enabled, I'd expect that every page I visit show the secure lock in Chrome.

### Actual Behavior
When I click on the Monitor menu, it changes from secure to (info) icon in Chrome.

### Steps to Reproduce (including precondition)
- Enable SSL within genmon
- Make sure openweather API is configured and display minimum info is off
- Go to Monitor tab
- You will see an icon near the weather info that is not found, and links to (http://openweathermap.org/img/w/undefined.png)  
-- This http call is causing the info warning in Chrome and probably other browsers I assume.

### Screenshot or Pictures relating to the problem (if possible)
[image]

### Your Environment
- Generator Model: N/A
- Generator Registers: N/A
- Genmon Version: 1.12.2

jgyates, 2018-12-27:
@zekyl314 

I checked in a change that hopefully will fix this (V1.12.3). The url used to download the image for the weather conditions from openweathermap was using http. I changed it to use https. 

Let me know if this works for you. I tested with a self signed cert so Chrome always complains about that.

zekyl314, 2018-12-27:
That did solve the SSL warning, but still a broken image link.
https://openweathermap.org/img/w/undefined.png

I am using lets encrypt, which is free, plus a few scripts I found on-line to automatically update the SSL cert.

jgyates, 2018-12-27:
If you type this command in your browser:
https://YOURIPADDRESS/cmd/gui_status_json

it will return some data, in this data you should find a filed that looks like this:
 
{"icon": "10d"} 

where 10d is an icon code. Do you see the icon field in the data returned?

The web page takes the icon code and uses it as the file name of the weather icon. For example and icon code of "10d" will get the following icon:

https://openweathermap.org/img/w/10d.png

My guess is that you may have something wrong with the icon code returned from the gui_status_json call. If you don't have the "icon" field in the data returned then we should look at your /var/log/genmon.log file for any errors.

zekyl314, 2018-12-27:
Right now, it is working with 50n.  I'll keep an eye on it, and if shows broken image, then I'll see what it is and maybe a problem over at weathermap.  Thanks for all the help and wonderful project.

swvalenti194, 2018-12-28:
I never had an issue with the icon being displayed, after most recent update I now have a broken image file like you posted.

jgyates, 2018-12-28:
A fixed has been checked in. This ended up being related to a change in formatting to the JSON data, however the problem was only present under specific weather conditions. Let me know if this does not fix the problem for you and thanks to everyone for pointing this out.

closed 2018-12-28 by jgyates