jgyates/genmon #222: fail to set exercise time

steve-bailey1, 2018-12-31:
### Expected Behavior

When in the Maint UI I changed the day of week for the exercise from Monday to Tuesday.  I get the confirm propmt and confirm.  Refresh the screen and everything flips back to Monday.

### Actual Behavior

No change

### Steps to Reproduce (including precondition)

Change the day of week for the exercise and confirm

### Your Environment

- Generator Model: G0062440
- Generator Registers: Submit via the About page
- Genmon Version: V1.12.5

jgyates, 2018-12-31:
@steve-bailey1 

Can you submit your logs on the about page? I can not duplicate this.

jgyates, 2018-12-31:
You may have something wrong with your system. I have not received your logs. 

First I would use the test email button on the Settings page to validate your email is working.

You can also ssh into to the system and look at /var/log/genmon.log

steve-bailey1, 2018-12-31:
I have email coming out of that IP restricted to my domains so that explains that.  The contents of the log file show this consistently and even when I try to make the change.

2018-12-31 11:59:57,741 : [Errno 2] No such file or directory
2018-12-31 12:00:00,371 : [Errno 2] No such file or directory
2018-12-31 12:00:03,382 : [Errno 2] No such file or directory
2018-12-31 12:00:06,396 : [Errno 2] No such file or directory
2018-12-31 12:00:09,381 : [Errno 2] No such file or directory
2018-12-31 12:00:12,371 : [Errno 2] No such file or directory
2018-12-31 12:00:15,376 : [Errno 2] No such file or directory
2018-12-31 12:00:18,371 : [Errno 2] No such file or directory
2018-12-31 12:00:21,374 : [Errno 2] No such file or directory
2018-12-31 12:00:24,371 : [Errno 2] No such file or directory
2018-12-31 12:00:27,371 : [Errno 2] No such file or directory
2018-12-31 12:00:30,377 : [Errno 2] No such file or directory
2018-12-31 12:00:33,374 : [Errno 2] No such file or directory
2018-12-31 12:00:36,374 : [Errno 2] No such file or directory
2018-12-31 12:00:39,367 : [Errno 2] No such file or directory
2018-12-31 12:00:42,369 : [Errno 2] No such file or directory
2018-12-31 12:00:45,371 : [Errno 2] No such file or directory
2018-12-31 12:00:48,377 : [Errno 2] No such file or directory
2018-12-31 12:00:51,369 : [Errno 2] No such file or directory
2018-12-31 12:00:54,371 : [Errno 2] No such file or directory
2018-12-31 12:00:57,369 : [Errno 2] No such file or directory
2018-12-31 12:01:00,376 : [Errno 2] No such file or directory
2018-12-31 12:01:03,381 : [Errno 2] No such file or directory
2018-12-31 12:01:06,433 : [Errno 2] No such file or directory
2018-12-31 12:01:09,374 : [Errno 2] No such file or directory
2018-12-31 12:01:12,380 : [Errno 2] No such file or directory
2018-12-31 12:01:15,368 : [Errno 2] No such file or directory
2018-12-31 12:01:18,373 : [Errno 2] No such file or directory
2018-12-31 12:13:30,331 : [Errno 2] No such file or directory
2018-12-31 12:13:33,387 : [Errno 2] No such file or directory
2018-12-31 12:13:36,381 : [Errno 2] No such file or directory
2018-12-31 12:13:52,922 : [Errno 2] No such file or directory
2018-12-31 12:13:56,855 : [Errno 2] No such file or directory
2018-12-31 12:13:58,983 : [Errno 2] No such file or directory
2018-12-31 12:14:01,989 : [Errno 2] No such file or directory
2018-12-31 12:14:04,980 : [Errno 2] No such file or directory
2018-12-31 12:14:07,982 : [Errno 2] No such file or directory
2018-12-31 12:14:10,989 : [Errno 2] No such file or directory
2018-12-31 12:14:13,982 : [Errno 2] No such file or directory
2018-12-31 12:14:16,992 : [Errno 2] No such file or directory
2018-12-31 12:14:19,982 : [Errno 2] No such file or directory
2018-12-31 12:14:22,988 : [Errno 2] No such file or directory
2018-12-31 12:14:26,959 : [Errno 2] No such file or directory
2018-12-31 12:14:29,958 : [Errno 2] No such file or directory
2018-12-31 12:14:32,967 : [Errno 2] No such file or directory
2018-12-31 12:14:35,958 : [Errno 2] No such file or directory
2018-12-31 12:14:37,989 : [Errno 2] No such file or directory
2018-12-31 12:23:05,897 : [Errno 2] No such file or directory
2018-12-31 12:23:08,953 : [Errno 2] No such file or directory
2018-12-31 12:23:11,863 : [Errno 2] No such file or directory
2018-12-31 12:23:12,097 : [Errno 2] No such file or directory
2018-12-31 12:23:12,336 : [Errno 2] No such file or directory
2018-12-31 12:23:14,957 : [Errno 2] No such file or directory
2018-12-31 12:23:17,959 : [Errno 2] No such file or directory
2018-12-31 12:23:20,959 : [Errno 2] No such file or directory
2018-12-31 12:23:23,957 : [Errno 2] No such file or directory
2018-12-31 12:23:26,953 : [Errno 2] No such file or directory

jgyates, 2018-12-31:
Is there anything unique about your setup that differs from the installation described in the docs? i.e. are you running Raspbian on a pi? Or have any files recently been deleted?

I checked in an update that may give us a hint with an additional debug print. The log above shows prints every 3 seconds, which is the frequency  the web app updates so these are likely caused by the call from the web app.

I would do this:

1) update the the version I just checked in, this will restart the system (assuming your software is functional enough to update form the web ui, if not you can type "sudo bash ./genmonmaint.sh update" from the genmon directory, then type "bash ./startgenmon.sh restart")
2) check the /var/log/genmon.log file and see if it is different

jgyates, 2018-12-31:
OK. This is what I thought was the problem with the No such file error. The file /opt/vc/bin/vcgencmd is not present in CentOS so the software will fall back and read /sys/class/thermal/thermal_zone0/temp to get the internal CPU temp. I checked in an update that removes that print to the log file to keep your SD card from the writes every 3 seconds when the web page is up. I did not bump the version number but if you update it should pull the new file.

Back to your original problem of the exercise day not changing. once you update to the version without the log, do you see any errors in your log when you try to change the exercise day? check /var/log/genmon.log, /var/log/genserv.log and /var/log/mymodbus.log.

Also, can you perform any other commands on the maintenance page, like start and stop the generator?

steve-bailey1, 2019-01-01:
Just updated.  The log files are now clean.  The generator starts and stops on command via the Maint page but the changing of the exercise time still doesnt change.  There are no log entries during the attempt to change the exercise time.

jgyates, 2019-01-01:
If anyone else with a Evolution Air Cooled unit can try to duplicate this it would help.
My guess at this point is that this may be a setting on your system, but it is hard to know for sure at this point. The Evolution Liquid Cooled and Air Cooled units both use the same technique to set the exercise time. The Nexus models are different.

Let's check out a couple of things:
1) Look at the web interface Maintenance and Monitor page. They both list the Controller. Are these entries the same. Based on your model number, both pages should show "Evolution, Air Cooled". If either does not then that could be a cause of the problem (assuming this is unique to your setup)
2) On the setting page, is the setting for "Enhanced Exercise Time" enabled or disabled? This enables the exercise option for weekly/ bi-weekly / monthly instead of the standard weekly only. This features is available on most Evolution units, depending on the firmware version. You can check register 002d on the registers page (icon in top right), if this value is set to 0001 or 0002 then this is likely your problem and you should enable this option on the settings page. It would be nice if the software could autodetect this but I don't know the firmware revision that this feature was added and if register 002d is 0000 (weekly) I have no way of knowing if the generator supports this option, hence the option in the settings page. You generator may be set for monthly exercise and then the day of the week option is not valid.
3) In the file /etc/genmon.conf is uselegacysetexercise = False? This option forces the use of the Nexus exercise method if true.

steve-bailey1, 2019-01-01:
1 - both pages list Evolution Air Cooled.
2 - Enhanced exercise is off. 002d is set to 0000.  Firmware version is V1.07 (would love to upgrade this but dont feel like calling a dealer)
3 - uselegacysetexercise is set to false.

jgyates, 2019-01-01:
OK. I know that is older firmware since I have seen Air cooled Evolution firmware in the 1.4x range. Not that you need to upgrade if it is working for you. 

Has this ever worked for you are you just now noticing this. The reason I am asking is that the default method I am using to set the exercise for Evolution could possibly not be supported with your firmware version. It is worth trying the uselegacysetexercise = true option in the /etc/genmon.conf. If this works then note number 2 on [this page](https://github.com/jgyates/genmon/wiki/Appendix-4-Known-Issues) relating to the legacy exercise method. Let me know how this goes.

steve-bailey1, 2019-01-01:
Setting that to True worked as far as keeping the setting on the Maint page after reloading.  I will let you know tomorrow at noon if it runs :)

jgyates, 2019-01-01:
Thanks Steve, Happy New Year! I am going to close this issue, feel free to comment here if you have any other questions on this or if your generator does not exercise.

closed 2019-01-01 by jgyates