jgyates/genmon #224: Cat6 for cable?

rastam4n, 2019-01-05:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

rastam4n, 2019-01-05:
Hi All
I am making my cable and I wonder if Cat6 could be used for the two wires between the generator and the db9 connector. Also is there any reason not to locate the pi3 away from the generator... for example in the house approximately 20’ of cable.
Thanks 
Peter

trimixdiver1, 2019-01-05:
I used outdoor grade cat5e and my run is 50ft, no problems

krisasmith, 2019-01-05:
I put my pi3 in a electrical box next to the transfer switch in my garage about 20ft away from the generator. I power the pi with PoE (using an adapter from amazon) and use a shielded 4 conductor security wire from home depot for the data (send/receive) an a common ground. The wire is much smaller in diameter then cat6 so its easier to run though the conduit with the rest of the wires for the generator.

closed 2019-01-05 by jgyates