jgyates/genmon #225: pyowm need to be pinned to < 2.10

sdadh01, 2019-01-06:
Just did a new install and pyowm>=2.10 just went Python 3 only. For a Python 2 install requirements.txt needs to have 'pyowm<2.10' to pin to the LTS 2.9 version (or upgrade everything else so it works under Python 3 ;-)).  I can create a PR if needed but running out of time today.

Thanks for all the work on this so far solid software...

jgyates, 2019-01-06:
I checked in an update that addresses this. Thanks for the heads up.

FYI, python 3 is working, but needs more testing. I have not put this in the documentation yet, but if you add a "3" to the install script it will install the version 3 libraries (i.e. sudo bash ./genmonmaint.sh install 3 ). There is still work that needs to be done on modifying the cron startup script and I need to figure out the best way to upgrade everyone with no issues. I have performed testing with most of the the code on 3.x, but I have not tested 100% at this point. To date I don't know of any specific python 3 incompatibilities.

jgyates, 2019-01-07:
Just thought of one more issue that needs to be addressed with the switch to python 3, genmon loader needs to be updated to use python 3 to load the other modules. That is an easy fix..

jgyates, 2019-01-07:
I checked in a few more updates for python3 support. Now to run python3 you just have to:

     bash ./genmonmaint.sh install 3

instead of this for python 2.7:

     bash ./genmonmaint.sh install

Also for python3 you will need to edit the startgenmon.sh script. Change this line:

     pycmd="python"

to this:

    pycmd="python3"

closed 2019-01-07 by jgyates

sdadh01, 2019-01-07:
I'll do a full install from scratch as a test for Python 3 this evening... I see those brackets being added ;-).  I'd already set up python in update-alternatives as follows : 

sudo update-alternatives --install /usr/bin/python python /usr/bin/python2.7 1  # add python 2 
sudo update-alternatives --install /usr/bin/python python /usr/bin/python3.4 2  # add python 3 (as default)
sudo update-alternatives --config python # to choose which version to use (currently at 1 for me)

Now I can change python versions for the host without bothering with code changes.  I guess some virtualenv incantation would do the same trick too but that would need another pip command or requirements.txt entry.

jgyates, 2019-01-07:
one caveat is that if you modify startgenmon.sh as noted above and then do an update to the latest version it will get overwritten. You can avoid this by copying startgenmon.sh to another location (outside of the genmon directory) and updating  crontab to use the copied version.

sdadh01, 2019-01-07:
Starting from a reinstall over lunch and got install errors : 

```
bash ./genmonmaint.sh install 3
.
.
Traceback (most recent call last):
  File "/usr/bin/pip3", line 9, in <module>
    load_entry_point('pip==1.5.6', 'console_scripts', 'pip3')()
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 356, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2476, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2190, in load
    ['__name__'])
  File "/usr/lib/python3/dist-packages/pip/__init__.py", line 74, in <module>
    from pip.vcs import git, mercurial, subversion, bazaar  # noqa
  File "/usr/lib/python3/dist-packages/pip/vcs/mercurial.py", line 9, in <module>
    from pip.download import path_to_url
  File "/usr/lib/python3/dist-packages/pip/download.py", line 25, in <module>
    from requests.compat import IncompleteRead
ImportError: cannot import name 'IncompleteRead'
.
.
.
```

Repeats each time pip install is called in 'installgenmon()'. I've seen this before - it's an old version of pip that has been installed for pip3 clashing with other installed modules (see [.](https://www.raspberrypi.org/forums/viewtopic.php?t=126338)) The fix is to use easy_install3 to update to the latest pip :

```
sudo easy_install3 -U pip
```

Maybe adding that to genmonmaint.sh is an answer but it's not really a nice thing (TM). Raspbian is a pain sometimes because it seems to lag.

Anyway, after doing the easy_install3 all seems to works as expected. Back to the day job...