jgyates/genmon #227: Setting Up Static IP on Ethernet Port

youngdy19, 2019-01-17:
I just received my preasembled unit and i was wondering if there is a way to set up a static IP address on the Raspberry Ethernet interface.  I do not want to run WiFi for it and need to set a static for remoting in.  Also is anyone aware if this software is SNMP Compatible for monitoring.

Thank You

jgyates, 2019-01-17:
The following link describes how to setup a static IP address. This applies to both wired and wireless interfaces.

https://raspberrypi.stackexchange.com/questions/37920/how-do-i-set-up-networking-wifi-static-ip-address/74428#74428

Another option (assuming your router supports this) is to setup a static IP assignment in your router. You give your router the MAC address and when ever that interface requests an IP address via DHCP then you will always get the same IP address.

The genmon software does not support SNMP, however there is an add one for Nagios (network monitoring tool) that will allow the state of the generator and the software to be monitored.

Also, you can add SNMP support for the raspberry pi. This will monitor the raspberry pi, but not the generator controller software.

https://bigdanzblog.wordpress.com/2015/01/03/installing-snmp-onto-a-raspberry-pi/

liltux, 2019-01-17:
@youngdy19 , to add to what @jgyates replied with, I personally prefer setting up static IP at router.  That way it always assigns the Pi with its MAC address the same IP.  I had an issue with one I programmed for someone that I setup a static IP on the raspberry pi.  At my wifi it worked fine but at his wifi the router would never update routing information to the PI.

jgyates, 2019-01-17:
@youngdy19 

One other note on SNMP: you could use the nagios plugin with SNMP linux support. 

This is the Nagios plugin:

https://github.com/jgyates/genmon/wiki/1----Software-Overview#otherappscheck_monitor_systempy

It will return a one line status of the generator on the command line.

Using the instructions in this link:

https://runops.wordpress.com/2015/04/04/create-custom-snmp-oid-in-linux/

you could create a custom SNMP OID.

If you run this command:

     python /home/pi/genmon/OtherApps/check_monitor_system.py -H IPAddressOfYourPI

It will return a one line status of the generator and monitor software. I use this to monitor my generator using Nagios, but it should work to extend the linux SNMP functionality.

closed 2019-01-21 by jgyates