jgyates/genmon #229: Fuel usage after outage

wrvolz, 2019-01-20:
On the new fuel usage on the outage page, where the fuel  usage is listed, it's showing the total fuel used since it was reset, not the amount of fuel used during the outage. 
Thanks,

Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2019-01-20:
@wrvolz 

Can you provide more details. After looking at the code related to this I do not see how this would be possible. I will need your logs (submit form the about page) and the outage log details as well as the fuel usage data from the maintenance page.

Thanks
Jason

jgyates, 2019-01-21:
I received your logs, nothing is noteworthy here. I will still need your fuel usage info form the outage page and maintenance page to take this further.

wrvolz, 2019-01-23:
Here's the outage page:

Outage : 
    Status : Last outage occurred at 2019-01-20 16:49:59 and lasted 0:00:00.
    System In Outage : No
    Utility Voltage : 247 V
    Utility Voltage Minimum : 191 V 
    Utility Voltage Maximum : 255 V 
    Utility Threshold Voltage : 156 V
    Startup Delay : 10 s

    Outage Log : 
        2019-01-20 16:49:59, Duration: 0:00:00, Estimated Fuel: 11.67 gal
        2019-01-20 16:49:59, Duration: 0:00:00, Estimated Fuel: 11.67 gal
        2019-01-05 08:31:18, Duration: 0:00:11
        2019-01-05 06:41:22, Duration: 0:00:11
        2019-01-05 06:41:21, Duration: 0:00:00
        2019-01-05 06:20:15, Duration: 0:00:00
        2018-12-22 01:19:00, Duration: 3:44:01
        2018-11-24 04:54:48, Duration: 0:00:11
        2018-10-24 15:31:44, Duration: 0:00:03
        2018-10-24 11:00:26, Duration: 0:00:25

Here's the maintenance page:

Maintenance :
    Model : 
G0062500

    Generator Serial Number :  0008370131

    Controller :  Evolution, Air Cooled

    Nominal RPM :  3600

    Rated kW :  20

    Nominal Frequency :  60

    Fuel Type :  Propane

    Engine Displacement :  999 cc

    kW Hours in last 30 days :  33.56

    Fuel Consumption in last 30 days :  9.33 gal

    Run Hours in last 30 days :  4.52

    Controller Settings :
        Calibrate Current 1 :  1470

        Calibrate Current 2 :  1453

        Calibrate Volts :  1010

    Exercise :
        Exercise Time :  Weekly Wednesday 13:00 Quiet Mode On

    Service :
        Service A Due :  175 hrs or 08/19/2020 

        Service B Due :  375 hrs or 08/19/2022 

        Total Run Hours :  215

        Hardware Version :  V1.05

        Firmware Version :  V1.07

jgyates, 2019-01-23:
@wrvolz 

This makes sense now. You data was helpful. I was not checking for an outage of zero seconds and the fuel usage / power log functions are setup to return queries with no duration (zero seconds) are for the entire power log.  I checked in V1.12.14 which should fix the problem. Let me know if you have any issues. I will close this issue for now but feel free to post to this thread of you have any issues.

Thanks
Jason

closed 2019-01-23 by jgyates