jgyates/genmon #230: HTTPS and Secure Web App

liltux, 2019-01-21:
When using the Secure Web App, is there a built in redirect?  That way http request are redirected to https?  Or you are forced to use https?

jgyates, 2019-01-21:
No, there is not built in redirect. The Flask library only responds on one port.

liltux, 2019-01-21:
Ok, I was just playing around with that feature.  I noticed the login screen when adding login credentials, the box identifiers lay over the user input and don't go away when user types their respective user name or password.  Everything works, it just looks weird.

jgyates, 2019-01-21:
if you can send me a screen capture of what looks weird.

liltux, 2019-01-21:
[image]

liltux, 2019-01-21:
@swvalenti194 yes.  By turning on Webserver Security it will "turn off" the standard protocol "http" and operate on the secure protocol "https".  It will use the default certificates in Flask or you can create your own.  These will be self-signed certificates, unless you get signed certificates (example letsencrypt.org), and you will get an "are you sure" page from your browser.  Just click on the advanced tab and you should see a link "proceed anyways" or something similar.

swvalenti194, 2019-01-21:
Ok, so I added username and password and went to ip using https:// and now it won't display at all.  Refused to connect.  Any ideas?

jgyates, 2019-01-21:
The web address changes when using the secure web server . 

Non-secure example:

    http://192.168.1.10:8000

Note http, not https prefix.

Secure example:

     https://192.168.1.10

Notice the 8000 is not present with secure (https).

Let me know if this does not  address you problem.

swvalenti194, 2019-01-21:
Correct using the secure example and doesn't work, I just edited the genmon.conf and started it again using port method.  Saved my settings again use https and refused to connect.  Oddly if i use http and port it brings me to the login page.

liltux, 2019-01-21:
@swvalenti194 You should not need a key and cert file as it will use the default in Flask.  What changes did you make to the genmon.conf file?  I was just testing mine and I did not have to change anything manually in the genmon.conf file.  I only had to type:
https://ipaddressofmypi

swvalenti194, 2019-01-21:
I couldn't get to the page at all using the port or https so I just made the https setting etc false in the config to get me back to genmon software page.

jgyates, 2019-01-21:
I would guess that you are using the wrong URL or you have an issue with your libraries installed. If you see errors in /var/log/genserv.log relating to SSL then I would run this command at the console to correct any problems with you libraries:

    bash ./genmonmaint.sh install 

This will download and install all of the required libraries needed for the software.

Also, it would be helpful to know the address format your are typing in the the address box of your browser.

swvalenti194, 2019-01-21:
https://192.168.0.21

here is an error I see in log

2019-01-21 13:48:41,045 : Error in app.run: [SSL: CA_MD_TOO_WEAK] ca md too weak (_ssl.c:2699) : genserv.py:1473

jgyates, 2019-01-21:
ok. This error has to do with the order of the library . install and the fact that the default encryption of the SSL library that Flask uses changed. Running the command at the console should fix this problem:

    bash ./genmonmaint.sh install

swvalenti194, 2019-01-21:
Ok, should I say yes or no?

Installing rpirtscts...
~/genmon ~/genmon
fatal: destination path 'rpirtscts' already exists and is not an empty directory.
make: 'rpirtscts' is up to date.
Finished installing rpirtscts.
~/genmon
Copy configuration files to /etc? (y/n)?

jgyates, 2019-01-21:
No, you already have copied the config files

swvalenti194, 2019-01-21:
Still doesn't work, loads the login page using port and not https

swvalenti194, 2019-01-21:
disregard you made a typo in the update command

swvalenti194, 2019-01-21:
updated and still same issue

liltux, 2019-01-21:
Have you tried copying over the original conf file and starting over?

sudo cp /home/pi/genmon/conf/genmon.conf /etc/genmon.conf

liltux, 2019-01-21:
@jgyates Thanks for the update I will check it out here soon.

swvalenti194, 2019-01-21:
oh well it was worth shot but still doesn't work.  And no don't want to start from scratch again

jgyates, 2019-01-21:
Another thing you could try is to update the system software:

      sudo apt-get update
      sudo apt-get dist-upgrade

During the "sudo apt-get dist-upgrade" you will be prompted (Yes or No) to update packages that are out of date, select yes.

Once this is complete, reboot your system with this command:

    sudo reboot

jgyates, 2019-01-21:
Some background: I have seen this before, but it usually is corrected with the first two commands we ran. You can search the issues with the "SSL / HTTP" tag for people who have had a similar issue.

swvalenti194, 2019-01-21:
I got the same version you do...honestly if it's a pain for you don't worry about it.

jgyates, 2019-01-21:
You best bet outside of reinstall is the apt-get commands. If that does not fix it then I would recommend a reinstall if you want this feature. For reference I included the version info for the apt based ssl libraries and pip based ssl libraries installed on my system:

If i type this:

      apt list --installed | grep -i ssl

It will give me this:

libssl-dev/stable,now 1.1.0j-1~deb9u1 armhf [installed]
libssl-doc/stable,now 1.1.0j-1~deb9u1 all [installed,automatic]
libssl1.0.2/stable,now 1.0.2q-1~deb9u1 armhf [installed]
libssl1.1/stable,now 1.1.0j-1~deb9u1 armhf [installed]
openssl/stable,now 1.1.0j-1~deb9u1 armhf [installed,automatic]

and this:

     pip list | grep -i ssl

will return this:

pyOpenSSL (18.0.0)

swvalenti194, 2019-01-21:
libssl-dev/stable,now 1.1.0j-1~deb9u1 armhf [installed]
libssl-doc/stable,now 1.1.0j-1~deb9u1 all [installed,automatic]
libssl1.0.2/stable,now 1.0.2q-1~deb9u1 armhf [installed]
libssl1.1/stable,now 1.1.0j-1~deb9u1 armhf [installed]
openssl/stable,now 1.1.0j-1~deb9u1 armhf [installed,automatic]
python-openssl/stable,now 16.2.0-1 all [installed,automatic]
python3-openssl/stable,now 16.2.0-1 all [installed,automatic]

pyOpenSSL (19.0.0)

and i did the apt-get commands prior

liltux, 2019-01-21:
@jgyates 
Looks Great!! Thanks.

jgyates, 2019-01-22:
@swvalenti194 

Based on the data it looks like you have pyopenssl installed via apt-get and pip. This may be due to the fact that once the install procedure used apt-get and it later changed to pip (due to the SSL problems).  If you are not using the pi for anything except genmon then I would recommend this:

    sudo apt-get purge python-openssl

this will remove the version of openssl installed via apt-get.

Then type this:

    sudo pip install pyopenssl

If anything fails during the above step you can always just run the install script again:
     
     bash ./genmonmaint.sh install

Select no for copying the files as you did before.

Then restart genmon by typing:

     bash ./startgenmon.sh restart

swvalenti194, 2019-01-22:
Still doesn't work after actions performed

jgyates, 2019-01-22:
Well, I hate to say it but I am out of ideas. I will think on this and see if I can come up with anything else to try...

swvalenti194, 2019-01-22:
All good my friend, no worries.

jgyates, 2019-01-22:
@swvalenti194 

@buschauer suggested a workaround that could solve this issue. I checked in V1.12.13. This version uses a genmon created cert for the secure webserver option. For compatibility with older version, if for some reason the genmon cert fails then if falls back to using the Flask self signed cert (which is not working for you). If you get a chance, try this out and let me know if it works.

swvalenti194, 2019-01-22:
@jgyates  updated and now works.  As always thanks for all for time and patience.

jgyates, 2019-01-22:
Thanks for the feedback!

closed 2019-01-22 by jgyates