jgyates/genmon #231: Outage Reporting (logs) Incorrect/not consistent

USAFPride, 2019-01-24:
There is some odd reporting in the program when outages occur with my 14kw air cooled Nexus controller  

On the Logs page (under Run Log) it shows and outage on the 23rd:
```01/23/19 05:53:23 Stopped - Auto 
01/23/19 03:22:18 Running - Utility Loss 
01/20/19 08:42:18 Stopped - Auto 
01/20/19 08:30:10 Running - Exercise
```
Software was running during the time of the outage. From the Monitor Page:
```Monitor Health : OK
Controller : Nexus, Air Cooled
Run time : Generator Monitor running for 3 days, 0:28:19.
Generator Monitor Version : V1.12.11
```
On the Outage Page it doesn't show the outage in the log
```Outage : 
    Status : No outage has occurred since program launched.
    System In Outage : No
    Utility Voltage : 245 V
    Utility Voltage Minimum : 0 V 
    Utility Voltage Maximum : 0 V 
    Utility Threshold Voltage : 156 V

    Outage Log : 
        2018-05-04 10:24:00, Duration: 0:00:00
        2017-12-16 11:42:26, Duration: 0:01:32
```

On the Status page, the Last Log Entries match the Logs page so that is correct.

jgyates, 2019-01-24:
FIrst, lets check to see that outage checking has not be inadvertently disabled. On the settings page, validate that "Do Not Check for Outages" is not enabled. If it is enabled, the this would  cause this behavior.

If this is not disabled then we can start looking at other things. There are two internal flags that can be set that would disable the outage checking. 1) the setting mentioned above and the failure to set a flag that signals the system initialization is complete. The fact that your utility voltage min / max are both zero indicates that one of these two conditions is likely. Can you submit your log files on the about page?

USAFPride, 2019-01-24:
"do not check for outages" was enabled.

I've submitted the logs for you.

jgyates, 2019-01-24:
You logs look good. If you disable "do not check for outages" it should work. This setting is for installations that use a "smart" transfer switch. Smart switches monitor the utility voltage for an outage and initiate the start process. Most switches that are used with Evo/Nexus controllers are "dumb" switches since the Evo/Nexus controller has the ability to monitor the utility line. In an installation with a smart switch the utility line is typically not connected to the controller so it can not sense when an outage occurs. The tool tip description of this setting describes this. Let me know if you have any other questions or problems.

closed 2019-01-24 by jgyates

USAFPride, 2019-01-24:
I'm thinking the tool tip and wiki may need a tweak to be a bit clearer. The first sentence implies something different than the second in the [wiki](https://github.com/jgyates/genmon/wiki/1----Software-Overview). 

Keep in mind, I am an electrician, so I read your response above as follows:
Smart switches monitor the utility
Dumb switches monitors the utility
Smart switches are not connected to the utility line

and that is where I get confused. If a smart switch monitors the utility, but is not connected ...

Per the wiki
`If you have a Transfer Switch that monitors the utility line and starts your generator you may need to set this to true.`

Whereas the second sentence:
`Most transfer switches requires the generator controller to monitor the utility line voltage and notify the switch to activate so in most cases this will not be needed.`

The third sentence is also a bit odd as my utility voltage was reading Zero, I was getting the reading, but no false emails.
`If you software is reading zero volts for the utility voltage and you are getting false emails regarding an outage then enable this setting.`

Maybe changing the tip a bit would help as well.

> Disable the monitoring of the utility line voltage for detecting power outages. If you have a 'smart' transfer switch, this may need to be enabled.

to

> Disable the monitoring of the utility line voltage for detecting power outages. If you have a 'smart' transfer switch, this may need to be enabled. Evo/Nexus controllers are typically set to disabled.

or give an example of a 'smart' switch in the tip as well

jgyates, 2019-01-24:
I will update the wiki to try to make this clearer.

USAFPride, 2019-01-24:
It's the nuance between controller and transfer switch.  I get it as the controller is not necessarily the transfer switch, but could be.

jgyates, 2019-01-24:
I updated the wiki and tool tips (you will need to update the software for the tools tips to show up. I did not update the version as this update was just for the documentation and the code did not change. Let me know if this is better.