jgyates/genmon #232: Enhancement request: Remove TLS from mail config

ToltingColtAcres, 2019-01-25:
mymail.py contains the following code:

        try:
            if self.SSLEnabled:
                 session = smtplib.SMTP_SSL(self.SMTPServer, self.SMTPPort)
                 session.ehlo()
            else:
                 session = smtplib.SMTP(self.SMTPServer, self.SMTPPort)
                 session.starttls()
                 session.ehlo
                 # this allows support for simple TLS
        except Exception as e1:
            self.LogErrorLine("Error SMTP Init : SSL:<" + str(self.SSLEnabled)  + ">: " + str(e1))
            return False

thus, it assumes if you are not running ssl on your mail server, you're running TLS.

Some of us run neither (internal mail server w/o external communication capabilities) so tls and ssl really are not necessary. However, genmon will not allow this configuration. The workaround at the moment is to simply comment out the starttls and ehelo lines, but that's sort of a pain as it requires shelving changes each time you want to pull updates from the repo.

Please consider adding a

disabletls = true/false

option to /etc/mymail.conf

and changing genmonlib/mymail.py to wrap the tls code in an if statement.

jgyates, 2019-01-25:
@ToltingColtAcres 

It did not occur to me that this could be an issue. I will make this update. I should have it completed later today or tomorrow. I will let you know when it is complete.

closed 2019-01-25 by jgyates

hokie21, 2019-02-17:
While it still works if an IP address is entered for the SMTP server, the GUI makes it red which would lead you to believe that only FQDN is supported.
[image]