jgyates/genmon #233: ERR_CONNECTION_RESET after latest update

SethL, 2019-01-30:
### Expected Behavior
Web console up 100% of the time

### Actual Behavior
After the most recent update, genmon will fail to load the webpage after a period of time with this error:
ERR_CONNECTION_RESET

### Steps to Reproduce (including precondition)
Reboot pi, everything works great.  eventually (minutes hours etc) the page will not load.  I submitted my logs but I had to reboot before submission because the webconsole wouldn't load until I did

### Your Environment

- Generator Model: G0070421
- Generator Registers: Submitted
- Genmon Version: 1.12.15

SethL, 2019-01-31:
> Hi @SethL
>
> I received your logs. This helps a lot. According to your logs you update to version 1.12.15 on the 27th and upon reboot the openweathermap support reported the following error:
>
> 2019-01-27 16:41:25,460 : New version detected: Old = V1.12.14, New = V1.12.15
> 2019-01-27 16:41:26,985 : GenMon Loaded for site: 833 Majestic Ct
> 2019-01-27 17:12:24,443 : Error in GetObservation: Exception in calling OWM web API.
> Reason: <urlopen error [Errno -3] Temporary failure in name resolution>
>
> this error repeated several times (name resolution).
>
> The software was restarted twice today:
>
> 2019-01-30 16:15:51,277 : Generator Monitor Shutdown
> 2019-01-30 16:16:27,889 : GenMon Loaded for site: 833 Majestic Ct
> 2019-01-30 16:16:57,015 : Validation Error: Invalid register length (Start) 01ac 00000000
> 2019-01-30 18:29:40,880 : Generator Monitor Shutdown
> 2019-01-30 18:30:39,446 : Validation Error: Invalid register length (Nexus Alarm) 0064 0009
> 2019-01-30 18:30:41,118 : GenMon Loaded for site: 833 Majestic Ct
> 2019-01-30 18:31:12,862 : Unable to convert type <type 'NoneType'> in GetDispatchItem
> 2019-01-30 18:31:12,863 : Item: None
>
> The errors above (Validation Errors) are due to missed packets.The log file /var/log/mymodbus.log shows the missed packets here:
>
> This caused the Validation errors at 1/30/2019 at 16:16
> [... 32 lines cut ...]
> Average Transaction Time : 0.0306 sec
>
> On my system, I sometimes drop a packet or 2 when the generator (48kw diesel) starts it's exercise cycle, probably due to emi or vibrations.
>
> Let me know if you have any additional data or have any other questions.

Thanks so much for looking at this.  It has been really cold here, almost below freezing now.  Maybe that has something to do with it.  I don’t really use the weather part so I just disabled that.  I’ll open up the gen cabinet tomorrow and check all of the connections, maybe something is not making good contact.

 

Thanks again!

SethL, 2019-01-31:
Could the serial packet drops cause the fuel counters to reset to 0?  This has happened 2 times in the past couple of weeks?  I had 425 gal to start with and it was down to 422 when it reset to zero.  I set it at 422 and then it slowly went down every 2 weeks when it exercised.  Now it is back to 422 again and it says I have consumed 0 gal.

 

1.	Software request modbus register A
2.	The controller does not respond before the timeout period
3.	The software request modbus register B
4.	The software responds with the register A value (not the register B value as expected)
5.	the software will invalidate (if possible) the value returned and log an error

1.	Validate or replace your cable or check your connections on your RS-232 converter board, they may be loose and a change temperature is causing the connection to be marginal.
2.	You could try extending the modbus timeout. This will make the system wait longer for response from the controller, however it could have the side effect of making the system less responsive if packets are dropped since you are waiting longer for data that may not come if the packet was lost due to a marginal cable. To extend the modbus timeout, click the icon in the top right of the web interface twice to get to the advance settings, then add a value greater than zero to the field labeled "Additional Modbus Timeout (sec)", for example putting 1.5 in this box will add 1.5 seconds to the modbus timeout. I would start with a value like 1.0 and see if that helps or hurts. If you still see the same issue try . increasing the timeout to 5 seconds and see if that helps or hurts. In theory if you serial communications is working as expected you should never reach the timeout since every request to the controller would be answered.

jgyates, 2019-01-31:
@SethL 

I don't know of any correlation in the code that would cause that to happen.  Also, the logs do not report any errors except the ones I mentioned earlier. Resetting the power / fuel monitoring logs will just delete the file genmon/kwlog.txt so I don't think that would cause the problem.

SethL, 2019-01-31:
The first time it happened I got an email saying that I had low fuel on the generator and it said I had 0gal remaining.  I about had a heart attack until I reminded myself I didn’t have a fuel sensor lol.  I always take the updates so maybe something got screwy during one of them or something.

jgyates, 2019-01-31:
The fuel email notice is not from a fuel sensor, but from the estimated fuel calculation. It calculates the load (0-100%) of the generator and uses the fuel consumption numbers provided by generac to calculate how much fuel is being used, then it subtracts this from the tank size.

More details are here:

https://github.com/jgyates/genmon/wiki/Appendix-8-Monitoring-Fuel-and-Power-Usage

jgyates, 2019-02-04:
I am closing this issue for now. Feel free to comment here if you have any other data or questions.

closed 2019-02-04 by jgyates