jgyates/genmon #234: Question Evolution 2.0 wiring

jonathanpisarczyk, 2019-02-04:
Quick question regarding evolution 2.0 wiring.

I read its possible to get genmon and mobilelink running at the same time.  on issue: https://github.com/jgyates/genmon/issues/183

Is it possible to power the PI with PIN1 (5V) & PIN2 (ground) and power the Evolution 2.0 mobilelink at the same time? 

I assume I need to splice wires for PIN1 & PIN2.  so both PI & mobilelink can be powered at the same time.  I was thinking of plugging current mobilelink molex into a female molex and then wire in pi then plug a new male molex into generac. 

Does the PI need to use PIN4 ground instead of PIN2?

PIN outs from wiki:
Evolution 1.0 and Nexus Aux Port Pinout:
Pin 8  - Transmit from Controller, connect to Receive on the Pi
Pin 7  - Received to the Controller, connect to Transmit on the Pi
Pin 2  - Ground. 
Pin 1  - 5V. This is an optional method for powering your Pi

The Evolution 2.0 Controller Aux Port Pinout adds these to the above pinout:
Pin 3 - +12V Accessory Power
Pin 4 - RS-232 port ground
Pin 5 - RS-485 Data (-), Used for built in WiFi support
Pin 6 - RS-485 Data (+), Used for built in WiFi support

jgyates, 2019-02-04:
Maybe @jimland58 or @beschaaflh can weigh in on how they powered their Evo2.0 based pi?

Without this info from them, I would recommend powering from the battery as I don't have a valid data point on power the pi with Evo2.0 and the Generac Wifi module both connected.

beschaaflh, 2019-02-05:
Yes, I power from the battery.  I've lost communication between Evo2.0 and Raspberry pi twice since November.  Communication was reestablished by disconnecting serial cable at serial adapter (rebooting didn't help).  I'm thinking maybe there's a problem with the adapter or cable so I've ordered more parts and will replace if I lose commiuncation again.  I initially had the mobilelink working but disconnected because I didn't want to risk having an Evo firmware update somehow disable genmon.  From my perspective, GenMon seems a lot more useful then mobilelink.  Thanks @jgyates for the great software and support!

jimland58, 2019-02-05:
Same here, power from the battery using this adapter: https://www.amazon.com/dp/B01DYE54LI/ref=cm_sw_em_r_mt_dp_U_dNowCbGGJ3M36

Might do the same with disconnecting the MobileLink to prevent updates.  Was talking to the guys that installed my generator last weekend and according to them they were told by Genrac that they were going to stop 3rd party devices from getting generator data and make it where you could only get data via MobileLink.  Anyone heard anything about this?

liltux, 2019-02-05:
I haven't heard anything about that.  Would it surprise me, no.  But there are other venders out there that  have pay for devices such as Omnemetrix, I don't see generac locking them out.  Push come to shove, get an RS485 adapter and piggy back off of the same serial connection as their wifi device.

jgyates, 2019-02-05:
I have not heard this this yet. If they do we could possibly communicate with the controller via RS-485 like the wifi modules does. I am guessing this also uses modbus so it may be as simple as using a RS-485 converter instead of a RS-232 converter. The genmon software would need set and reset RTS on the serial port to enable / disable the RS-485 transceivers but that should be achievable. 

If anyone want to test this let me know and I can recommend a Rs-485 adapter for the pi and make the necessary changes in the code.

jgyates, 2019-02-05:
sounds like @liltux and I had the same idea

jgyates, 2019-02-05:
@ts4s , yes, the newer model has a wifi module (Evolution 2,0).  Generac has very little detail  in the default info reported over wifi. To get the more details you have to pay a recurring fee.

jgyates, 2019-02-11:
Closing this issue for now. Feel free to post on this thread for any updates to this issue.

closed 2019-02-11 by jgyates

jonathanpisarczyk, 2019-02-16:
Hey Guys,

I thought I would share my new setup for anyone in the future.  I haven't tested it yet.  I hope to test next week.  The Startech EPS8EXT  makes it plug and play inside generator without having to modify wiring inside the generac.  

https://www.startech.com/ca/Cables/Computer-Power/Internal/8-EPS-8-Pin-Power-Extension-Cable~EPS8EXT

[image]

[image]

[image]

jgyates, 2019-02-16:
@jonathanpisarczyk , thanks!

jonathanpisarczyk, 2019-03-10:
Reporting back.  Thanks for all the help.  The startech cable worked perfect.  I have both genmon & Generac monitoring running at the same time. 

[image]

jgyates, 2019-03-11:
Thanks for the info. Glad to hear it is working.

pheenawm, 2019-05-16:
> Hey Guys,
>
> I thought I would share my new setup for anyone in the future. I haven't tested it yet. I hope to test next week. The Startech EPS8EXT makes it plug and play inside generator without [...]

Hi @jonathanpisarczyk ,

I am almost done with my setup, also to an Evolution 2.0 .  These photos are extremely helpful, but I had two questions for you:
1) are you using a desiccant packet long-term just in case you get moisture in the box?
2) is there a purpose for the red and black wires that are in the strand with the yellow and white coming out of the DB-9 breakout?

zekyl314, 2019-05-16:
Can you share a picture of it mounted and a link for the BOX?  I really like that setup.  My pi is just inside the generator sitting on it using double sticky tape.  Sort of thinking I should upgrade to a better mount method.  I have a really nice external antenna due to distance from AP.

pheenawm, 2019-05-28:
> Can you share a picture of it mounted and a link for the BOX? I really like that setup. My pi is just inside the generator sitting on it using double sticky tape. Sort of thinking I should upgrade to a better mount method. I have a really nice external antenna due to distance from AP.

Here is my final setup, which is based on the box mentioned in the wiki.  I needed stronger magnets than I'd anticipated.

Were I to do it again, I would have placed the disconnect for the power outside of the box; now to remove the box, I have to disconnect from the battery terminals.

[image]
[image]
[image]
[image]
[image]
[image]

half-vast, 2020-01-20:
> I have not heard this this yet. If they do we could possibly communicate with the controller via RS-485 like the wifi modules does. I am guessing this also uses modbus so it may be as simple as [...]

@jgyates  - not sure if anyone volunteered to try the RS-485 method.  If not, I'll gladly jump in and help.  Used to do remote monitoring/management stuff for Verizon Wireless before I retired.

jgyates, 2020-01-20:
My recommendation for RS-485 would be to use something like [this](https://www.amazon.com/RS485-CAN-HAT-Long-distance-Communication/dp/B07DNPFMRW). This TTL to RS485 converter, like may others on the market, has automatic enabling and disabling of the transceivers so you do not have to worry about the timing pitfalls of the old RTS enable / disable method. The newer methods detects the start and stop bits and enables / disables the transmitter without software intervention making this a drop in replacement. 

The main issue with the RTS method is knowing exactly when to lower RTS, if you wait too late you will corrupt the response packet, if you do it too early you will corrupt the packet you just sent. this was a bit more feasible with a microcontroller and assembly, but with a high level language like python you don't have direct access to the hardware and the data is buffered by the OS, so you are left with a timing scheme that you hope the far end will respect.