jgyates/genmon #235: Enhancement Request - Maintenance Repair Log

alphaa211, 2019-02-06:
Is it possible to add a user editable log for entering maintenance / repairs / checks that have occurred?  A lot of generator end users perform maintenance and keep that in a paper record but it would be nice to see the basics in Genmon.  Possibly a simple log that records Date / Engine Hours / Check-Maintenance-Repair / Short Description.

Example:

01/15/2019 - 69.8 - Check - Oil, 2 hash marks down, white condensation present
01/15/2019 - 69.8 - Check - Air filter, clean, no oil or water in air box
01/29/2019 - 70.5 - Maintenance - Oil & Filter Change, Synthetic 5w-30 
02/05/2019 - 71.1 - Maintenance - Adjust Valves, Rear right out of spec @ .06
02/05/2019 - 71.2 - Repair - Leaking bottom oil cooler clamp, replaced

jgyates, 2019-02-06:
@alphaa211, that sounds like a good idea. Let me think about the best way to implement this.

dennis48755, 2019-03-10:
I'm loving this feature addition, thanks so much.

I have a couple additional requests for it..,
- can you please save/show the decimal digit for Service Hours
- is it possible to add the ability to edit an entry? I'm notorious for typo's.

Thanks
Dennis

jgyates, 2019-03-10:
@dennis48755 

Thanks for the feedback. I will add these to the list. In the meantime you could edit the Service Journal by logging into your pi an edit the file maintlog.json in the genmon directory. Be case not to change the brackets or commas as they provide JSON formatting .

dennis48755, 2019-03-10:
Thanks Jason,

Please consider adding the decimal digit wherever the Run Hours are displayed (Maintenance & Service Journal pages, I believe).

jgyates, 2019-03-11:
I checked in V1.12.25 which allows decimal values to be saved. If the run hours has a decimal value it will populate the default of the service log.

Note that Nexus Air Cooled models do not support a decimal value for run hours so on these models it will be displayed as a whole number. Also, if you run hours are an whole number (i.e. 123.0) it will populate as 123 due to the javascript run time.

Regardless of your model you can change the run hours before you save to any number.

Let me know if you have any problems.

dennis48755, 2019-03-11:
Looks good Jason.

Thanks Again
Dennis

dennis48755, 2019-09-25:
Jason,

Thank you for adding support of decimal values for Service Hours in the journal.
I would ask for you to take it a couple steps further.
- Could you include the decimal value when the "+Add" function is used in the Journal and the Service Hours are retrieved/prefilled.
- Also could you please add the decimal value in the Service Value displayed on the Maintenance page.
Basically everyplace the Service Hours are used/displayed.

Thanks Again,
Dennis

jgyates, 2019-09-25:
Hi Dennis,

I am clear  on what you are are asking for.  Presently, if you  click "+Add" it will  create a new entry  and populate the run hours with the current run hours. If you controller supports a decimal value  of run hours it will use  it the decimal value, if your controller  does not  support a decimal value for run hours, it will not be displayed. Is there something else that I am missing?

jgyates, 2019-09-25:
Sorry, meant to say "I am not clear on what you are asking"

dennis48755, 2019-09-25:
Jason,

Your understanding of my request is correct. I guess my Evolution (1st generation) controller must not support it as I don't see the decimal value.

Interesting that I do see the tenths value on the Evolution control panel.

Thanks for all you are doing.

jgyates, 2019-09-25:
Do you see  a decimal value of run hours on the Maintenance page? This should display the decimal values that your controller outputs.

dennis48755, 2019-09-25:
No. That was the other part of my request, for you to add it. Not knowing it should be there.

dennis48755, 2019-09-25:
Just to be clear, by decimal value, I am referring to a value following a decimal point, tenths of an hour.

If it helps, my register 000b & 000c values are 0000 0088

jgyates, 2019-09-25:
I assume this value of 0088 is hex so that would be 136 hours

Liquid cooled evolution is the only evolution model that supports a tenths place. the run hours with a whole number value are  read from 000b and 000c on all Nexus and Evo models. On Evo Liquid Cooled you can read them from 000b and 000c to get the whole number run hours and from 005f and 005e to get the run hours to the tenths place.

dennis48755, 2019-09-26:
Yes 136 is correct for my run hours. What a shame Generac has the tenths displayed on the control panel but not via registers.

Thanks again Jason.