jgyates/genmon #236: Gen Fuel log keeps resetting to zero

SethL, 2019-02-06:
### Expected Behavior

Generator Fuel log calculating fuel usage correctly by subtracting it each time the gen runs

### Actual Behavior

Every time my generator runs I get a message that the fuel is at 10% or lower
"Time: 12:06:01 PM
Date: Wed 06-Feb-2019
Warning: The estimated fuel in the tank is at or below 10%"
When I go to the genmon webpage it tells me the fuel level is 0 when there is ~422Gal in the tank

### Steps to Reproduce (including precondition)
This used to work fine.   A few weeks ago I had a small propane leak.  The gas company came out and repaired it.  Before the tech left, I went on genmon and started the gen to make sure everything worked ok.  (BTW this impressed the heck out of him and I had to show him how it worked etc :-))
When I went back inside, I had the email saying I was out of gas.  I reset the fuel usage and then just set my lank level to 422 (which is what I remember it was before it went to zero in error).  
Thinking it was a fluke, I just went on my merry way.

Today the gen exercised and the same problem happened.  It seems that if the gen runs (manual start or exercise) the fuel is reported right away as zero.  I dont know if an outage would do it as I haven't had one since I got the gen.  (the logs do show one but only for a few seconds, I dont remember it though)

### Your Environment

- Model : G0070421
- Generator Registers: Submitted as well as logs
- Genmon Version: Version V1.12.16

jgyates, 2019-02-06:
@SethL 

A couple of questions:

1) If you view the power log graph on the main web page and select the month view, do you see any activity on the graph?
2) If you do see any activity, go to the maintenance page (at the bottom of the page) and click the button labeled "Reset Power Log & Fuel Estimate". At this point you should not see any activity on the power log graph.

The way the software works is the fuel starts with the size of the tank, in your case 422. fuel is subtracted from this in one of two ways:

1) If "Subtract Fuel" on the advanced page is used it will subtract this amount from the tank size. This allows you to use the fuel monitoring with less than a full tank
2) As entries are added to the power log when the generator runs, the fuel consumption is calculated and subtracted from the total in the tank.

The amount of fuel used over time is not stored, it is calculated, only the power log entries are stored and the fuel consumption is derived from the power log.

The likely situation is that your power log has not been reset. If you try to reset it and it does not remove the log entries from the graph let me know.

SethL, 2019-02-06:
Ok this is what I had for the month view (graphic below):

.

I reset the power and fuel log, then set my tank to 425 gal which is what it was full (6 months ago)

Month graph is now clear

Then I did the offset for the “subtract fuel” and set it to 4 Gal (to account for the small usage)

 

Started the gen

 

Now it looks like it is calculating the fuel 😊

 

I just stopped the gen and it held and didn’t reset to zero.

 

Thanks!

closed 2019-02-06 by jgyates