jgyates/genmon #237: SMTP server red for ip address

hokie21, 2019-02-17:
While it still works if an IP address is entered for the SMTP server, the GUI makes it red which would lead you to believe that only entry by FQDN is supported. Folks using a local mail server may not have DNS set up for the local mail server and may be using an SMTP server by IP address. 

This is a low priority, cosmetic issue you may want to consider bundling with other updates. 

[image]

jgyates, 2019-02-17:
ok. let me take a look at this. thanks for pointing this out.

hokie21, 2019-02-22:
Here's another thought while you're in this part of the code. There is currently configuration of an email sender address, but not an email sender display name. So my name may be "John Smith," but my email is johnsmith5421@gmail.com. Today you would see johnsmith5421@gmail.com as the sender in your email inbox. With a display name you would see email from "John Smith" in your inbox. Again, really a cosmetic change, but something to consider.

jgyates, 2019-02-22:
The SMTP fix you first mentioned should be updated in about a week, the javascript still needs to updated to support this. The 2nd request was checked in today as V1.12.19

hokie21, 2019-02-22:
Looks good! Thanks!

jgyates, 2019-02-25:
@hokie21 ,

This has been fixed with V1.12.21. Let me know if you have any issues. Thanks for the help!

closed 2019-02-25 by jgyates