jgyates/genmon #238: Enhancement suggestions -  Extend exercise duration and exercise under load

UPS99, 2019-02-22:
Many owners and servicers in very cold areas have found that the engines do not adequately warm up during the 5 minute fixed exercise time period (originally 12 minutes in earlier firmware versions).  That can be seen in many posts on generator forums.  

While that time period is built into the firmware, it should be possible to code a separate routine to essentially "press" the start and stop buttons.  This could be an independent routine which optionally would trigger upon sensing of the standard built in exercise period, but run the system longer before stopping.  

Another useful feature could be a less frequently scheduled routine that would actually run the generator under load, by initiating start, followed by "start and transfer" approximately 30 seconds later, to avoid an actual outage.

jgyates, 2019-02-22:
Let me thing about this. This sounds like a good feature, however there may be some caveats. For example the commands to "start" and "start and transfer" are internal to the firmware so we may not be able to issue the start command followed by start and transfer unless we first issue a stop.

Also, there is another command that is available, but not in the web interface that will start the exercise cycle in quiet mode (quite mode runs at a lower output voltage, frequency and RPMs). This could be used to make an exercise cycle of a duration that is not offered by the firmware, however we would need to test to make sure that the exercise cycle,  when manually executed, does not interfere with the automatic generator functions when an outage occurs.

UPS99, 2019-02-22:
The Start and Transfer command will definitely transfer if the generator is already running from the Start command.  I have done that.  

 

I don’t know what will happen if the generator is already running in exercise mode and Start, or Start and Transfer, is selected.  However, per the manuals, if is running in exercise mode, and a real power failure occurs – it will simply transfer after 10 seconds, increasing speed if in quiet mode, without shutting down first.  

 

Probably it would also keep running if the Start and Transfer command was issued while running in exercise – since it would likely need more code in the firmware to react differently to start due to power failure vs manual start, with little to be gained.

 

Of course, if it is not workable to link the new added routine to trigger after sensing the built-in exercising had started – another way would to set the built-in exercising to run once per month only and use the added routine as the primary scheduler.  

Importance: High

liltux, 2019-02-22:
@jgyates, are there cli commands for starting, stopping, and starting with transfer?  I could see cronjob being a great way to handle this new suggested feature. Since you will not be able to disable the current excerise or built-in excerise of 5 min.  That way if someone wished to have an increased excerise run time or even a loaded excerise, using the cli commands and crontab I think would be an easy solution and shouldn't brick anything.  Just a thought.  
And I would advise a use at own risk, do to the fact I'm not sure what would happen if an outage occurs during a "start with transfer" and the stop command was issued during outage.  If the command is ignored that would be great, else the generator could shutdown under a load and could cause equipment damage in the home and or to the generator.  Just my thoughts.

UPS99, 2019-02-22:
“Start”, “Start and Transfer”, and “Stop” were user available options on the relatively rare, discontinued, Generac “Advanced Wireless Monitor”.  The instructions stated basically that in the event of a power failure, the automatic functions would override manual operation, and that, for example, the manual  Stop would not stop the generator.

jgyates, 2019-02-25:
@UPS99 

I checked in V1.12.21 which has a new Add On program that supports enhanced exercise functionality you suggested. Let me know if you have any questions or feedback.

Thanks
Jason

closed 2019-02-25 by jgyates

UPS99, 2019-02-26:
I have updated to the latest version (looks good) and have scheduled a 12 minute run on my unit for Thursday morning.  (The earliest I have time during the day, preferring not to do at night possibly bothering neighbors.)

 

One other option may be a consideration – the ability to schedule 2 different exercise processes separately.  For example – weekly or bi-weekly Normal run, and a less frequent Transfer run, once per month or even less frequently than that.

UPS99, 2019-02-28:
The enhanced 10 minutes exercise I had scheduled ran fine.  I had scheduled it to overlap the standard scheduled exercise time to see if there was any effect, such as extra starts or stops, but there was no problem.   The only difference I noted is that the Run Log shows (correctly): "Running - Remote Start", instead of "Running - Exercise".

jgyates, 2019-02-28:
@UPS99 

thanks for the feedback. Regarding the option to set an additional exercise, i need to make some changes to the way add-on handle settings parameters to make this happen, but i will put it on the list of things to do..