jgyates/genmon #239: [Non Issue] Translation to other languages.

JMVS, 2019-02-22:
Hello @jgyates.

First and foremost, thank you very much for such a great & useful program. I've assembled my build using a Raspberry Pi Zero W (getting power from the controller) and I've been running it for 4 days now.

I would like to know if you would be open to receive translations to other languages. As of now it would seem like a complicated task to mantain as the strings are hardcoded into each file. If you ever implement a translation scheme, please count me in for translating into Spanish (LatAm).

Thanks again.

jgyates, 2019-02-22:
Hi @JMVS

This is a good idea. It will take some time to do this properly, but I will start this process. the first step is issolating all of the translatable strings. Once I get it to that point I will let you know.

Thanks
Jason

JMVS, 2019-02-22:
That's great @jgyates! I've been using (what I think it is) a free platform for translation collaboration: oneskyapp.com. There are a few projects that I collaborate with that rely on that platform. Let me know if I can be of any help.
Jose.

jgyates, 2020-09-29:
I am tracking the feature request list here: https://github.com/jgyates/genmon/blob/master/feature_requests.md

I will be closing this thread for now but the request is still on the master request list.

closed 2020-09-29 by jgyates

i3laze, 2026-09-03:
Picking this up six years later: I've done the string inventory you described as the first step, and opened a discussion with a proposed architecture and a finished Russian translation — https://github.com/jgyates/genmon/discussions/1531. @JMVS, your OneSky suggestion fits the proposed key format directly, so your offer to help may finally be actionable.