jgyates/genmon #240: [Feature Request] Shutdown RPi gracefully from web interface

JMVS, 2019-02-24:
Hello @jgyates.

It would be great to have an option to shutdown the Pi (sudo shutdown -h now) directly from the web interface and not via ssh. This is specially useful when you're tinkering with the configuration & assembly. 

In my case, the generator is on the roof (not easy access) and there were times I just forgot to shut the Pi down (or worst, not having ssh enabled) and I just risked to unplug directly from power.

Thanks.
José.

jgyates, 2019-02-24:
You can do this now by entering the following URL in your browser:

http://piIPAddress:8000/cmd/shutdown

closed 2019-02-24 by jgyates

JMVS, 2019-03-18:
Hi @jgyates. Thank you for your response. I've been without a computer and unable to respond sooner. That's great! Is there a list of all the commands available?

Thanks again 
Jose

jgyates, 2019-03-18:
All of these commands are used by the web server to communicate with the javascript code. The commands listed on line 157 are passed from the web server (genserv.py) to genmon.py for processing. The commands at line 216 and below are processed within the web server (genserv.py). The only two commands that are not utilized by the web interface are "shutdown" and "stop". "shutdown" will shut down the pi an "stop" will stop the web server but leave genmon.py running.

Since these commands are used for communications between the web app and genmon I have decided to let the source be the documentation. The commands that are passed on to genmon.py are also callable from the command line using ClientInterface.py. If using ClientInterface.py you can use the help command to get a list of commands that genmon.py will accept by typing:

     generator: help 

To give you a little more info as an example, the command  http://piIPAddress:8000/cmd/getreglabels on line 269 will return the names of the modbus registers, which are displayed in in the web interface on the registers page.