jgyates/genmon #241: [Feature Request] Status LED for genmon

JMVS, 2019-02-24:
Hello @jgyates.

Would it be possible to implement a genmon status LED?
Using a single LED with diffeHello @jgyates.

Would it be possible to implement a genmon status LED?
Using a single LED with different blinking/patterns to indicate a different status.
For example:
Solid = Genmon up and running (serial OK, wifi/eth OK)
Blinking pattern 1 = Serial error
Blinking pattern 2 = Wifi/eth error / not connected
Blinking pattern 3 = Updating
and so on...

Thanks
José

jgyates, 2019-02-24:
@JMVS 

It is possible, but I try to keep the base genmon.py program independent of raspberry pi hardware dependencies and try to put things like this in add-on programs. We have a program now that will display status of the generator on the pi's gpio outputs:

https://github.com/jgyates/genmon/wiki/1----Software-Overview#gengpiopy-optional

It currently does not support the items you listed above, but it does display the generator status on different gpio outputs.

I can look into extending this or adding a new add on program.

jgyates, 2019-02-25:
@JMVS 

I updated gengpio.py to include GPIO outputs for monitor health (which includes communication errors) and internet connectivity check. The link above for gengpio.py has been updated to show the corresponding GPIO pins for these new events. I opted not to make the GPIO toggle (blinking LEDs) but rather have a separate LED for each event.

Let me know if you have any questions.

closed 2019-02-25 by jgyates