jgyates/genmon #242: Add On Evolution Enhanced Exercise - Feature Request / Change

alphaa211, 2019-03-05:
Can the add-on Evolution Enhanced Exercise bi-weekly setting be changed to activate on the soonest / next available future selected day and time to start the bi-weekly cycle versus starting the following week?

When I tested the settings, the standard Maintenance tab, Generator Exercise Time, bi-weekly option will select the soonest available future date and time but the add-on enhanced rolls to the next week.

I'd like to be able to sync both to the same week and day if possible like you can with the weekly option.

The add-on enhanced exercise option is awesome just like the rest of the project.

Thanks.

jgyates, 2019-03-05:
Hi @alphaa211 

The way it works now is that whenever there is an exercise cycle generated by the add on, the date is logged to the configuration file for the add on. When bi-weekly is selected it will use this date to determine if it is time for the biweekly test (i.e. has 14 days expired). A quick way to sync these is edit the file /etc/genercise.conf and remove the entry named last_exercise, then set the add on to the desired time and date.

I will think on this request, the problem I have is that I need to also have a way to determine the last exercise time so I can know when a bi-weekly exercise is due. To meet your request I need to find a way to reset this value without disrupting the bi-weekly operation. I feel sure I can find a way to do this, I just need to think about the trade offs.

alphaa211, 2019-03-06:
What this would allow a user to do is “bury” or “hide” the OEM bi-weekly exercise within the add-on enhanced exercise cycle effectively eliminating the standard exercise.  The controller ignores an exercise cycle if the cycle start time occurs while the unit is running for any reason.  

I’d like to set bi-weekly 30 minute enhanced exercise only without having to have a separate monthly, weekly, or bi-weekly standard exercise.  

Right now I have a weekly standard exercise set for Tuesday’s at 1635 and the add on set for bi-weekly on Tuesday at 1630 for 30 minutes.  The bi-weekly add on cycle effectively replaces the standard exercise when it runs.

Again, the project is great and your responsiveness to enhancements and requests is absolutely top notch.  Thank you very much.

jgyates, 2019-03-06:
This helps me understand what you are looking for (override the existing exercise). I will look into this. Another issue is that the add on is timed from the system clock on the pi, not the clock on the generator so with the current implementation you would need to start at least one minute before the controller exercise time. You can sync the time between the controller and the pi, but since the controller does not export the seconds you could be off as much as 59 seconds.

jgyates, 2019-03-07:
@alphaa211 

I checked in an update that may help some (V1.12.23). This version has the option to use the generator time instead of the system time. Regarding the start time of biweekly, if I change the behavior to start on the next day of the week, then in the situation where the program is restarted it will not adhere to the two week schedule. Part of the problem is the architecture of the program expects all parameter updates (time of exercise, etc) when the program is started, unlike the controller firmware which is never restarted. This is only a problem with the biweekly when trying to override the controller exercise cycle. The best I can come up with is to do this if you want use biweekly and override the controller is to:

1) edit the file /etc/genercise.conf and remove the line with the "last_exercise" entry
2) set your controller to a two week exercise (i.e change the value so it will start on the next specified day of the week instead of waiting 2 weeks)
3) Then set the value of the add on and save (save will restart) and the biweekly will tart on the next specified day of the week)
4) I would set the "Use Generator Time" option and set the add on for 1 min before the controller to be on the safe side.

Once you do this they should remain in sync. Or if disable the add on for two weeks then you could set it and and they would remain in sync.

Let me know if you have any questions or problems.

closed 2019-03-07 by jgyates

liltux, 2019-04-03:
@jgyates, does the enhanced exercise work for Nexus units as well?  I see it is listed as an Evolution option.

jgyates, 2019-04-03:
@liltux 

Unfortunately no. The enhanced exercise add on uses the remote start, off, and start / transfer features. The Nexus does not appear to support these remote commands. If anything other than an Evolution is detected the add on will not show up in the add on section.