jgyates/genmon #243: email settings not quite as flexible as they could be

MolallaComm, 2019-03-06:
I'd like to have my genmon emails go to a distribution list.  Distribution lists don't have passwords, so the currrent GUI won't quite work in this scenario.  I think you have all the plumbing right, but maybe consider changing the settings GUI to be like below, which would allow for distribution lists and maybe be less ambiguous than what is there now:

To Address:
From Address:
SMTP Server:
SMTP Server Port:
Use SSL Encryption: On/Off
Disable TLS Encryption:  On/Off
Require Authentication:  On/Off
	Auth UserName:  If On
	Auth Password:  If On

zekyl314, 2019-03-06:
I send my alerts to a dist group today with no issues.  Have you looked in the notification tab on the GUI?  The settings tab, I had to input my SMTP user and password of course (Since I use an external SMTP server) but for the notifications, I have a group that just gets the outages for example, and then I get everything as a separate notification entry.

[image]

madisdev, 2019-03-06:
The address you are sending to has nothing todo with the username/password. A distribution list is just an email address. 

The username/password is for the account sending the email.

MolallaComm, 2019-03-06:
Many thanks!  I looked at the settings, and thought that was all you got (first time user – had not noticed the Notification setup).  FWIW, the option to not specify an auth user might still be useful.  In a corporate environment, many times the mail server is setup to trust other hosts on the LAN – and that would preclude you from having to create an account just for genmon or using somebody else’s credentials, which breaks your alerts when that somebody leaves the company etc., and get deleted or disabled in active directory or LDAP.

[genmon]<https://user-images.githubusercontent.com/29819262/53915366-c25b9a80-402d-11e9-8446-fb21b8eabeb1.jpg>

zekyl314, 2019-03-06:
There is a way to not use a username and password for a local smtp server.
I don’t know the setting but think it’s in a readme or doc on this project.

-Mike

https://user-images.githubusercontent.com/29819262/53915366-c25b9a80-402d-11e9-8446-fb21b8eabeb1.jpg>

jgyates, 2019-03-06:
@MolallaComm 

If you do not specify a password then the mail module will not attempt a login but still attempt to send the mail, so this should do what you want. I remember adding this behavior early on for folks with their own mail server. The UI will give you a warning nudge that you do not have a password but you can still save the settings. Let me know if you are unable to get this to work. You should also be able to test with no password. The way the code works now the Test Email button will require a password, but I will change that on the next update.

jgyates, 2019-03-07:
I checked in V1.12.23 which should allow the email test to work with a blank password. A blank password will not do a login before sending mail. Feel free to comment in this thread if you have any issues or questions. I am opting not to change the gui and I am concerned that it may cause other people confusion since most email servers require a password.

closed 2019-03-07 by jgyates