jgyates/genmon #244: [Question/Enhancement] Log Brownouts/Flickers & Power Spikes

diverbill, 2019-03-09:
I'm not sure of your polling or the feed from the transfer switch for utility power monitoring, if feasible could it log power loss or spikes even if the generator has not counted down to start? Another option is allow it to have a sample rate that is saved for X period of time. I have noticed my power company runs more on the hot side (248 - 252)

IE Normally here if the power flickers twice you know the power will be out for a bit, tonight we had the power flicker early on around 45 minutes prior to the actual power outage. 

Thanks for a great solution.

jgyates, 2019-03-09:
@diverbill 

The outage log monitors the utility voltage. If the voltage drops below the threshold voltage it will log an outage. Once in an outage, if the utility voltage rises above the pickup voltage the outage is considered over. This will behavior will log outages that may be too short for the generator to kick in (the controller has a timeout that is about 5 seconds by default, but you can change this in the controller settings).  The software does not monitor for overvoltage on the utility line.

The software has a list of modbus registers that it reads in a polling loop.  Once the loop is complete it will check for an outage or a state change in the controller. Using serial this gives about a 1 -2 second granularity on checking for outages.

For what it is worth, my utility voltage run about the same as yours. I think the tolerances (per ANSI spec) is +/- 5%, which 252V should be at that limit.

jgyates, 2019-03-09:
Check your outage log to see if the flicker was logged, also you can setup email notifications on an outage event.

closed 2019-03-11 by jgyates