jgyates/genmon #245: H-100 and Monitoring of associated HTS Transfer Switch

unsword01, 2019-03-12:
I just ran across this project and very excited to have found it. We have a 180kW Generac with an H-100 panel and an HTS Transfer Switch. We were able to get Genmon running pretty easily once we determined the correct Modbus address is 0x64 (100), however we note that the Utility data is not reported in Genmon - but when we use GenLink connected to the same RS232 port on the H-100 panel Utility load is reported, so we know the data is available. Screenshot from Genlink attached.

[image]

Reading through issue #98 it was noted in there that the H-100 does not report Utility data but that statement does not seem to apply when a HTS Transfer switch is associated with the Genset.

Would very much like to assist in having Genmon report voltage and current data from the HTS Transfer Switch which is communicating with the H-100 over a separate Modbus. Please advise how we can participate in this development.

Also - when "Smart Transfer Switch" is ON the UI no longer provides the ability to remote start the Genset or to force a transfer. With a connected HTS these should be available.

- Generator Model: 
    Model : Generac H-100 Industrial Generator
    Name Plate Info : 6.8L NA Gas Default 480V H Panel 
    Controller : H-100 Controller, PM-DCP Release 3.6D , HW00
    Controller Software Version : PM-DCP Release 3.6D , HW00, $Date: 2010/06/03 21:26:11 $

- Genmon Version: V1.12.25

jgyates, 2019-03-12:
@unsword01 

The graphic you included in your post is helpful. In addition to the graphic you posted, if you could submit your logs from the about page, this would be the first step that would allow me to add some of the transfer switch data into genmon. To submit your log and register data you must first have the software setup with outbound email working and the software communicating with the controller. If you have done that just go to the about page and click the button labeled "Submit Logs". This will email your logs and registers to me. From there I can put your register data in my simulator and I should be able to make the initial update. Once I post an update I would need to validate that the info is displayed correctly and I can make any needed changes from there.

Currently the "Smart Switch" option removes functionality for some controllers like utility voltage for Evolution Controllers and Utility vs Generator transfer switch state for H-100 and G Panel. I anticipate having a new option for HTS/GTS Transfer Switches as these appear to be the ones that pass status over the modbus link (correct me if I am wrong here).

Let me know if you have any questions.

liltux, 2019-03-12:
@unsword01 This looks promising.  So far I do not believe @jgyates has had the opportunity to recieve a good register dump from a unit with HTS switch for mapping and functionality features.  The H-100 controller will only report switch status if one of these two conditions exist:
1. You have a HTS or MTS transfer switch installed which communicates RS485 on another port,
2. Or, you have Switch position feedback wired between the switch and the generator. (switch position only)

I have a question for you, with the "SmartTransferSwitch" set to off does your generator remote start via Genmon?  I have been unsuccessful on our H-100 Generator.

unsword01, 2019-03-12:
@jgyates I uploaded two sets of data for you - the first around 9:22 AM (Eastern) was with the generator stopped. It ran on exerciser (not initiated by Genmon) at 9:30AM so I sent you another set of data around 9:40AM when the unit was running on exerciser but off-load.

jgyates, 2019-03-12:
@unsword01 

I checked in version 1.12.26. This version includes initial support for HTS/MTS/STS type transfer switches. Once you update to the latest from the about page, then enable the "HTS/MTS/STS Transfer Switch" on the Settings page, save your settings. Once the software has restarted after saving you will need to refresh your browser as I have added a few more gauges for HTS type switches.

Please validate the values displayed on the Status page under "Line State" are correct. Also, please submit your registers again so I can validate this myself. Let me know if you see any issues or have any questions.

thanks
Jason

unsword01, 2019-03-12:
@jgyates - Outstanding !!

I took a quick look at 1.12.26 and amazed you were able to add this funcationality so quickly. 

At first glance all the pertinent data from the HTS Switch appears to be present and correct. I’ll take some time later to validate the data in more detail.

Question - should the ‘Smart Transfer Switch’ setting be on or off ?

Thanks !

jgyates, 2019-03-12:
There are two settings related to transfer switches:

Smart Transfer Switch and HTS/MTS/STS Transfer Switch and they are mutually exclusive so it depends on your switch and how it is wired into your controller.

Smart Transfer Switch, if enabled, will disable Utility Voltage reporting for Evolution and Nexus Controllers. If disabled for H and G Panel controllers it will not report the controller's status for the transfer switch (Generator or Utility). For G and H Panel you will see a "Transfer Switch State" reading under Status->Line State. This status reflects bits in output register 6 on H Panel and Output Register 5 on G Panel. If Smart Transfer Switch is off this will not show up for H and G panels. From my understanding these bits may not valid for some types of transfer switches so I made it and option you could disable.

The  HTS/MTS/STS Transfer Switch option only shows up for H and G Panel controllers. If enabled this will display the transfer switch registers in the controller.

do the gauges display properly. I added gauges for utility voltage, frequency and power.

One other caveat, the current implementation  will read the current active switch. It appears that the controller can support multiple switches and I don't know how to select other switches at the moment.

When you have time, send me another output of your logs/ registers.

unsword01, 2019-03-13:
@jgyates - I've attached a couple of screenshots of Genmon connected to our system and submitted the register data back to you (that's a really nice feature..)

I note that Target Utility Frequency is shown as 0.60Hz instead of 60Hz, aside from that all the Line State parameters appear to be correct.

[image]
[image]

unsword01, 2019-03-13:
@jgyates On the topic of multiple HTS Switches yes that is possible. We have a system in LA which has two on it. Not sure if that system was the H-100 serial port cabled back into the facility to allow us to run Genmon there but would like to be able to. If that's possible I'll certainly get the register data from that. 

I assume they are running in some form of master/slave mode but haven't seen any details of that configuration in the HTS manual (other than changing the Modbus address)

liltux, 2019-03-13:
The HTS switches are slave and the H-Panel is the Master.

[image]

You can have a maximum of 4 HTS/MTS switchs attached to one H-Panel.  Assume the same is for G-Panel as well.  There are only 4 modbus address that can be configure using dip switches on the switches.

RS-485 (Port 0) (Generators)
The RS-485 port (Port 0) is used to communicate with HTS/MTS switches, RAP and RRP panels. The default
setting of Port 0 of the G-Panel, from the factory, is: Modbus Slave, 9600 Baud, No Parity, 1 Stop Bit. This
is because the G-Panels are used primarily in MPS systems and are set up to communicate with a System
Controller (PM-SC). As a stand-alone generator working with HTS/MTS or RAP/RRP devices, this port needs
to be configured as the Master, correct baud rate, No Parity, 2 Stop Bits. The key when working with any
Generac device which communicates via RS-485 is No Parity, 2 Stop Bits. On all HTS and MTS switches
and 21-LT RAP/RRP, the Parity and Stop Bits are fixed and cannot be changed. Each is a Slave Device that
will need its own slave address. Slave addresses are assigned at each device through dip-switches. The
Baudrates can typically be change to increase the communication speed. The 21-LT RAP/RRP is a passive
monitoring device so does not need a slave address.

jgyates, 2019-03-13:
OK. Let me know want to add support for multiple switches. Might require more investigation..

I just checked in a minor update that will fix the target frequency display. I did not change the version number but just clicking update should get the fix to you.

jgyates, 2019-03-13:
@unsword01 ,

Your registers show that the switch software version is "263" is that correct or should it have a decimal in it?

unsword01, 2019-03-13:
@jgyates,

I will check with Genlink in the AM on the SW version and/or physically inspect the HTS

Do you plan to add the remote start/stop/transfer controls for the H-100 with HTS use-case ?

jgyates, 2019-03-13:
@unsword01 

I would like to add them. I just need to determine what commands to write over modbus. One way to accomplish this is to capture the serial data that is sent/received from genlink. This would require a computer (PC, Mac, etc) with a serial port connected to the transmit line from the PC running genlink that would capture everything transmitted to the controller. If you are interested in helping out with  this let me know and we can talk more about the detail of how to accomplish this.

liltux, 2019-03-13:
@jgyates This would be great!  The remote START and STOP should be the same for all H-Panel.  The Transfer will be unique to the HTS/MTS.

unsword01, 2019-03-13:
@jgyates - Absolutely willing to assist with this. Can do it with another PC as a sniffer or possibly use this
https://www.eltima.com/products/serial-port-monitor/
which would avoid having to play around with hacking together a serial cable with just TX and GND

The Start/Stop we can test anytime. The transfer may be something to do over a weekend. Unless we can do it with Genlink not actually talking to the H100.

unsword01, 2019-03-13:
No joy being able to determine the actual software version for the HTS via Genlink, I also popped the cover off the controller module in the HTS hoping to find a sticker with the as-shipped version number but nothing there.

However - I did identify a report function in Genlink which produced a very detailed dump of practically everything apart from the HTS version number and I'm sure it will be of value in identifying some of the other registers and how scaling is derived for some of the analog values.

Report file attached
[Genlink_report.txt](https://github.com/jgyates/genmon/files/2962118/Genlink_report.txt)

jgyates, 2019-03-13:
Thanks, this is good info. 

BTW, the serial settings are 9600 baud, No parity, 8 data bits, one stop bit (9600,N,8,1)

unsword01, 2019-03-18:
Installed the Serial Port Monitor app noted above, then started up Genlink and initiated a genset run, then about a minute later initiated a genset stop.

The app saves in its own .spm format but I was able to export to a .csv file which I've zipped and attached. Can also provide the .spm file if desired.

[H100_Start_Stop.zip](https://github.com/jgyates/genmon/files/2979232/H100_Start_Stop.zip)

One observation, after disconnecting and reconnecting Genmon the log is not updating to indicate the activity (which I believe should be logged by the H-100) related to the run which we initiated while Genlink was connected to the H-100.

I also just sent you latest logs for reference

unsword01, 2019-03-18:
I updated to 1.12.27 and after the restart the logs updated - so here are the entries for this mornings run which will help you correlate the serial port session capture:

03/18/19 09:50:10 CRANKING 
03/18/19 09:50:11 GEN RUNNING
03/18/19 09:50:13 READY FOR LOAD 
03/18/19 09:54:08 ENGINE RPM 
03/18/19 09:54:18 GEN READY TO RUN 
03/18/19 09:54:18 GEN STOPPED

jgyates, 2019-03-18:
@unsword01 

This is very helpful! I checked in a an update (V1.12.28) that should support start and stop for H and G Panel systems. I also made a small change that might make start and transfer work (not this is only visible in the web interface on H100 and GPanel if HST transfer switch is enabled in the settings).

Let me know how the testing goes.

Also, when it is convenient for your it would be helpful to get a serial dump on a start and transfer and a parallel configuration if possible. No rush as I would guess that these are production systems and a transfer test would need to be scheduled.

In short, the change was that I was writing out a 0001 to start and your log shows that it should be a 0080. I made this change and the same change in the register that controls the transfer switch. There is a parallel enable register that is similar but this is not exposed in the web interface and is still writing a 0001 instead of 0080 as I am guessing this is a mask that controls each switch so it would probably be better to wait for a serial dump on this one.

@liltux , the way the logic should work is that any H or G Panel will have the start and stop options but only the HTS switch systems will have the Start and Transfer option. Does this should correct?

liltux, 2019-03-18:
@jgyates that would be correct.  The HTS or MTS switches are the only ones without outside modifications.  basically you would be sending a command to the transfer switch via the H-Panel through the RS485 connection.  
@unsword01 , Thanks!  I have been trying to find time to perform this same check but just hasnt worked out for me.

unsword01, 2019-03-18:
@jgyates - Glad the data was helpful. I've done the update and see the stop/start/transfer buttons now. I'll test that in the AM when back on-site.

With regards doing the transfer test you are spot-on, we need to schedule that test as it's a production system.

Thanks again for your work on this and the very fast implementation of the updates once you had some data to work with.

unsword01, 2019-03-19:
@jgyates - I was able to test the remote start/stop this morning. A few observations:

1. I was using my iPad on WiFi so I could observe the unit while running the tests, when the WiFi signal as not too good the Web UI got very slow and laggy. My first attempt to start the unit did not work (I did get the 'OK' dialog box).  I moved to a position indoors where I got better WiFi and was able to observe through a window and that attempt to start the unit was successful.

The UI seems to be very 'chatty' and I have observed that accessing it over VPN from home that things do become slow due to latency - the gauges take a few seconds for the pointers to get to their correct positions and it is generally slow if you change pages using the menu. I don't know if there's anything which can be done to improve this, such as using compression ? - if not then maybe a mobile/remote friendly 'status lite' page which just shows the data shown by the gauges in a simple table format with background colors to match the gauge scales?

2. Genmon sent an alert immediately upon the unit startup - but the actual information in the body of the message was incomplete and it seems like waiting a couple more poll cycles would provide a more meaningful message. Here's what I got:

Time: 08:15:21 AM
Date: Tue 19-Mar-2019
 
Status :
 
Engine :
Engine Status : Running from serial
Generator Status : Stopped, Preheating
Switch State : Auto
Output Power : 0.00 kW
Output Power Factor : 1.00
RPM : 1783
Frequency : 62.00 Hz
Throttle Position : 515 Stp
Coolant Temp : 129 F
Coolant Level : 681 Stp
Oil Pressure : 1 psi
Oil Temp : 0 F
Fuel Level : 93
Oxygen Sensor : 0
Current Phase A : 0 A
Current Phase B : 0 A
Current Phase C : 0 A
Average Current : 0 A
Voltage A-B : 379 V
Voltage B-C : 378 V
Voltage C-A : 394 V
Average Voltage : 410 V
Air Fuel Duty Cycle : 1000.00
 
Alarms :
Number of Active Alarms : 0
Number of Acknowledged Alarms : 0
 
Battery :
Battery Voltage : 25.77 V
Battery Charger Current : 0.00 A
 
Line State :
Transfer Switch State : Utility
Target Utility Voltage : 480 V
Target Utility Frequency : 60 Hz
Utility Frequency : 59.98 Hz
Utility Voltage A-B : 473 V
Utility Voltage B-C : 467 V
Utility Voltage C-A : 472 V
Average Utility Voltage : 471 V
Utility Current Phase A : 82 A
Utility Current Phase B : 79 A
Utility Current Phase C : 80 A
Average Utility Current : 80 A
Utility Power Factor : 1.00
Utility Power : 65 kW
Switch Reported Generator Average Voltage : 0 V
Switch Reported Generator Average Frequency : 0.00 Hz
Backup Battery Volts : 4.22 V
Switch Software Version : 263
Switch Selected : 0
 
Time :
Monitor Time : Tuesday March 19, 2019 08:15:19
Generator Time : Tuesday March 19, 2019 07:15:44

I'd suggest the message start with (or the subject is) "Genset started by <reason> at <Date / Time>" to convey the actual time it was started, then allow a short delay for the H-100 data to populate correctly and use that as what's reported in the message body.

Similarly, the genmon message send when the unit was stopped still indicated data from when it was running, even though it showed as 'Stopped' 

Time: 08:18:26 AM
Date: Tue 19-Mar-2019
 
Status :
 
Engine :
Engine Status : Stopped, Key SW Auto
Generator Status : Running,cooling down
Switch State : Auto
Output Power : 0.00 kW
Output Power Factor : 1.00
RPM : 1803
Frequency : 60.20 Hz
Throttle Position : 472 Stp
Coolant Temp : 127 F
Coolant Level : 691 Stp
Oil Pressure : 62 psi
Oil Temp : 0 F
Fuel Level : 93
Oxygen Sensor : 0
Current Phase A : 0 A
Current Phase B : 0 A
Current Phase C : 0 A
Average Current : 0 A
Voltage A-B : 481 V
Voltage B-C : 479 V
Voltage C-A : 481 V
Average Voltage : 480 V
Air Fuel Duty Cycle : 1000.00
 
Alarms :
Number of Active Alarms : 0
Number of Acknowledged Alarms : 0
 
Battery :
Battery Voltage : 28.47 V
Battery Charger Current : 0.00 A
 
Line State :
Transfer Switch State : Utility
Target Utility Voltage : 480 V
Target Utility Frequency : 60 Hz
Utility Frequency : 59.98 Hz
Utility Voltage A-B : 473 V
Utility Voltage B-C : 467 V
Utility Voltage C-A : 472 V
Average Utility Voltage : 471 V
Utility Current Phase A : 80 A
Utility Current Phase B : 78 A
Utility Current Phase C : 78 A
Average Utility Current : 79 A
Utility Power Factor : 1.00
Utility Power : 64 kW
Switch Reported Generator Average Voltage : 481 V
Switch Reported Generator Average Frequency : 59.98 Hz
Backup Battery Volts : 4.20 V
Switch Software Version : 263
Switch Selected : 0
 
Time :
Monitor Time : Tuesday March 19, 2019 08:18:25
Generator Time : Tuesday March 19, 2019 07:18:51

3. After startup the log shows Cranking/Running/Ready for load, but after shutdown the log doesn't show the Gen Stopped event.

unsword01, 2019-03-19:
A little while later the unit ran on its weekly exercise and that message did not reflect true state either:

Time: 09:30:42 AM
Date: Tue 19-Mar-2019
 
Status :
 
Engine :
Engine Status : Running exercise
Generator Status : Stopped, Preheating
Switch State : Auto
Output Power : 0.00 kW
Output Power Factor : 1.00
RPM : 1800
Frequency : 60.00 Hz
Throttle Position : 509 Stp
Coolant Temp : 131 F
Coolant Level : 690 Stp
Oil Pressure : 46 psi
Oil Temp : 0 F
Fuel Level : 93
Oxygen Sensor : 0
Current Phase A : 0 A
Current Phase B : 0 A
Current Phase C : 0 A
Average Current : 0 A
Voltage A-B : 473 V
Voltage B-C : 468 V
Voltage C-A : 474 V
Average Voltage : 472 V
Air Fuel Duty Cycle : 1000.00
 
Alarms :
Number of Active Alarms : 0
Number of Acknowledged Alarms : 0
 
Battery :
Battery Voltage : 28.38 V
Battery Charger Current : 0.00 A
 
Line State :
Transfer Switch State : Utility
Target Utility Voltage : 480 V
Target Utility Frequency : 60 Hz
Utility Frequency : 60.00 Hz
Utility Voltage A-B : 474 V
Utility Voltage B-C : 468 V
Utility Voltage C-A : 472 V
Average Utility Voltage : 471 V
Utility Current Phase A : 78 A
Utility Current Phase B : 74 A
Utility Current Phase C : 74 A
Average Utility Current : 75 A
Utility Power Factor : 1.00
Utility Power : 61 kW
Switch Reported Generator Average Voltage : 0 V
Switch Reported Generator Average Frequency : 0.00 Hz
Backup Battery Volts : 4.29 V
Switch Software Version : 263
Switch Selected : 0
 
Time :
Monitor Time : Tuesday March 19, 2019 09:30:40
Generator Time : Tuesday March 19, 2019 08:31:06

liltux, 2019-03-19:
@unsword01, those are correct states.  The messages are triggered off of at state change.  So when the generator is told to start the first step (diesel engine) is preheat, cranking, running, ready for load.  This are all valid statuses.

unsword01, 2019-03-19:
@liltux  - I can see that from the data - but if you don't know how to interpret the data then it's rather confusing and doesn't convey that the unit is running in normal state. If I was to send this to our IT folks for example I could see some consternation arising..

Maybe a second message sent 30 secs after the state change would provide data with more context.

liltux, 2019-03-19:
Ok, I wasn't sure you were aware.  Yes it does get messy with the hpanel messages.  I have played around with MQTT AND NodeRed to put the data into a better message format for the generator at our office.  Might check that out.  Some guys on here have used that to tie into smart home systems, it would be similar.

jgyates, 2019-03-19:
@unsword01, the low bandwidth is a good idea. I will put that on the list.

I will add a message at the top of status emails like I do on Evolution/Nexus:
    
    NOTE: This message is a notice that the state of the generator has changed. The system is not in alarm.

In normal optionation, the software will read the contents of the modbus registers in a loop. At every iteration of the loop the the contents of the "Engine Status" field is checked (as well as the alarm status), if this Engine Status changes the software will update the log registers and send an email is notifying the status has changed.

Status emails send all the data in from the status page (except the gauge and graph)

unsword01, 2019-03-19:
@jgyates - Thanks for considering the 'Low bandwidth' feature. One other small change which would be very helpful is the ability to put a custom field in the status e-mail messages which could be used for info on any action to take and/or a URL which the recipient of the message can click to take them to the real-time status page of Genmon.

jgyates, 2019-03-19:
I checked in V1.12.29. This adds additional language to the status emails for H-100. You can now disable engine state change email notifications via the notification page (engine state changes are in the "info" category). I also added the ability to add a user URL to status emails. This is setup on the advanced page.

The low bandwidth option will take a bit more time as that will require some javascript updates.

unsword01, 2019-03-20:
@jgyates - We ran another genset start/stop test today via Genmon and verified the inclusion of the URL (as set in Advanced Settings) in the alert emails at unit start/stop. Thanks for that addition it will be help helpful.

The issue which still remains is that the Run log does not appear to have captured the Stop event.
My run log shows this:

03/20/19 08:48:23 CRANKING 
03/20/19 08:48:24 GEN RUNNING
03/20/19 08:48:26 READY FOR LOAD 

The unit was only run for 5 minutes but as of 1PM the GEN STOPPED message is not in the log. If I was to restart Genmon then it would populate. Seems like the log entries are not being updated until the *next* state change trigger and as such the logs are not reflecting all the activities correctly.

I'd note that my Genset time and Genmon host times are not synchronized so not sure if this is a contributory factor, since I understand the Run Log is maintained in the H-100 and being read by Genmon.

jgyates, 2019-03-20:
@unsword01 

I checked in V1.12.30. This *should* fix the problem you are seeing with the run log. My guess is that there is a delay between the Engine Status updating and the run log being updated in the controller. I changed the logic so the alarm / run log will be updated in real time instead of just when the engine status changes. Let me know if this corrects the problem or if you have any other issues.

liltux, 2019-03-21:
@jgyates awesome!

unsword01, 2019-03-21:
@jgyates  - Unfortunately that change seemed to break the run log.

We did a test run at 3:15PM today (3/21) and the tail of the run log now shows:

03/19/19 08:31:03 GEN RUNNING
03/19/19 08:31:03 IN WARM UP 
03/19/19 08:31:02 CRANKING 
03/19/19 07:22:00 GEN STOPPED

I've just submitted logs - thx

liltux, 2019-03-21:
@jgyates FYI I have not been able to re-establish my runlog.  Since this has happened on someone else, I am wondering if something is toggling this feature off?

jgyates, 2019-03-21:
@unsword01 

the logs you submitted show this as the last entries in the run log:

 Run Log : 
               03/21/19 14:23:36 GEN STOPPED
               03/21/19 14:23:36 GEN READY TO RUN 
               03/21/19 14:23:24 ENGINE RPM 
               03/21/19 14:16:10 READY FOR LOAD 
               03/21/19 14:16:09 GEN RUNNING
               03/21/19 14:16:08 CRANKING 
               03/20/19 08:54:29 GEN STOPPED
               03/20/19 08:54:29 GEN READY TO RUN 
               03/20/19 08:54:17 ENGINE RPM 
               03/20/19 08:48:26 READY FOR LOAD 
               03/20/19 08:48:24 GEN RUNNING
               03/20/19 08:48:23 CRANKING 
               03/19/19 08:54:34 GEN STOPPED
               03/19/19 08:54:34 GEN READY TO RUN 
               03/19/19 08:54:23 ENGINE RPM 
               03/19/19 08:31:04 READY FOR LOAD 
               03/19/19 08:31:03 GEN RUNNING
               03/19/19 08:31:03 IN WARM UP 
               03/19/19 08:31:02 CRANKING 
               03/19/19 07:22:00 GEN STOPPED

Can you pull up your web interface and see if your log is showing correctly now? I am trying to see if this is a timing issue and if so how much of a delay is present. I am looking over the code and nothing is jumping out at me yet. No errors in your log files

unsword01, 2019-03-21:
@jgyates - Ah I see now - the sort order is now most recent at the top - I was looking at the bottom.

 Run Log : 
        03/21/19 14:23:36 GEN STOPPED
        03/21/19 14:23:36 GEN READY TO RUN 
        03/21/19 14:23:24 ENGINE RPM 
        03/21/19 14:16:10 READY FOR LOAD 
        03/21/19 14:16:09 GEN RUNNING
        03/21/19 14:16:08 CRANKING 
        03/20/19 08:54:29 GEN STOPPED
        03/20/19 08:54:29 GEN READY TO RUN 
        03/20/19 08:54:17 ENGINE RPM 
        03/20/19 08:48:26 READY FOR LOAD 
        03/20/19 08:48:24 GEN RUNNING
        03/20/19 08:48:23 CRANKING 
        03/19/19 08:54:34 GEN STOPPED
        03/19/19 08:54:34 GEN READY TO RUN 
        03/19/19 08:54:23 ENGINE RPM 
        03/19/19 08:31:04 READY FOR LOAD 
        03/19/19 08:31:03 GEN RUNNING
        03/19/19 08:31:03 IN WARM UP 
        03/19/19 08:31:02 CRANKING 
        03/19/19 07:22:00 GEN STOPPED

All the data is there ! - the problem was between the chair and the keyboard...

jgyates, 2019-03-21:
@unsword01  no worries, thanks for the feedback.

@liltux 

Regarding you logs, I suppose anything is possible, however genmon will not issue any modbus writes unless directed to do so by the user (remote start, set time, etc). The only exception to this is setting the generator time and this only occurs if you have "Sync Time" enable (will update the controller time once an hour) or "Sync Daylight Savings Time" which will update the controller time twice per year when daylight savings time changes. Does your run log show up if you are connected to GenLink? If not I wonder if there is something in Genlink that you can do to reset the log?

liltux, 2019-03-21:
@jgyates well I am glad my problem is my problem 😉!  I thought he might have been seeing the same as me.  Yes Genlink reports the same, and I have tried reloading firmware and still doesn't populate the field.  I have been wanting to work on some more but haven't had time, been fixing everyone (in my neck of the woods) else's generator problems!  Have a great weekend you guys!

jgyates, 2019-03-22:
I am closing this issue however feel free to post in this thread if there are issues / questions related.

closed 2019-03-22 by jgyates

jgyates, 2019-03-29:
@unsword01 

I checked in an update (V1.12.31) that reads some additional info from the H100 controller. Once you (or anyone with a H100) validates these new reads work if someone could send me their log files. Then I can parse the data from the new reads that I have added. Once I do this there should be more info in the web interface. This is an effort to add more info in the GenLink report @unsword01  sent me earlier.

unsword01, 2019-03-30:
I did the update and have uploaded logs and registers for you. Monday AM I can do a run test and provide logs while the unit is running.

Thx for continuing to enhance this !

jgyates, 2019-03-30:
I got your logs. Thanks!

I checked in version 1.12.32 which adds some data for the governor, regulator and engine on the Maintenance page. Send me your logs when you can.

unsword01, 2019-03-30:
@jgyates  - Logs just submitted for version 1.12.32 and I confirm seeing additional data on the maintenance page.

As noted, I'll do another upload on Monday while I run the unit

Thx

unsword01, 2019-04-01:
@jgyates - Logs just submitted with unit running (off load)

darkstar2002, 2019-04-17:
I tried this with our h-panel and an eaton ac3000 switch.  I did not get any info on the utility.

jgyates, 2019-04-17:
From my understanding, the transfer switch has to communicate with the H-100 controller as a modbus slave (controller is master, switch is slave) on a separate communication port. Genmon will still act as a master to the controller but on a separate port. The switch has to be HTS, STS o MTS compatible. If you enabled this feature on the settings page and your switch should be compatible, and it is not working you can send me your logs and I can see if any details may be gleaned.

jgyates, 2019-05-23:
@unsword01 

the low bandwidth option is included in V1.13.12. 

    http://ipaddress:8000/low

cc18dd, 2024-11-28:
@unsword01 Are you still active on GitHub?

cc18dd, 2024-11-29:
> Yes, but not actively using Genmon any longer (facility was shutdown)

Can you email me at ***@***.***? I have a question

jgyates, 2024-11-29:
@cc18dd 

your email was censored by github, possibly due to the fact that you responded to this thread in email. If you reply directly from the github interface for this thread it will likely work. Also, you can come back after @unsword01 has responded and edit the comment to remove the email address if you like.

@jgyates

unsword01, 2024-11-30:
I've responded via email

CRANKBOOSTER, 2025-02-16:
Hello,
I am new to all this. I have a 25KW SG0025 showing a SD Engine RPM SN.
Where do I download the GENLINK?
How do I interface with my Laptop?

Thanks.

jgyates, 2025-02-16:
@CRANKBOOSTER 
You would need to get GENLINK from Generac. Genlink is a product you would have to purchase from them. It is my understanding it is not available as a download. 

This forum is for using the genmon software. The genmon software is free, but I don't know enough about your model number to say if it would work on your generator. If you want you use genmon, I would need to know the  controller type in your generator to know if it would work for you.

CRANKBOOSTER, 2025-02-16:
See attached Pictures, Let me know if this helps

[image]
[image]
[image]

> [@CRANKBOOSTER](https://github.com/CRANKBOOSTER) You would need to get GENLINK from Generac. Genlink is a product you would have to purchase from them. It is my understanding it is not available as [...]

jgyates, 2025-02-16:
That does help. It looks like you have an H-Panel controller (H-100). This project does support monitoring the H Panel controller. Info on setting this up is in the project wiki

https://github.com/jgyates/genmon/wiki

The wiki contains info on setting up and installing the software. The H-100 specific steps are here:

https://github.com/jgyates/genmon/wiki/Appendix-G-Generac-H-100,-G-Panel-and-PowerZone-Controllers

CRANKBOOSTER, 2025-02-16:
will i be able to clear errors with the GENMON?

Thanks

jgyates, 2025-02-16:
You can issue the acknowledge alarm command.

It is my understanding to clear most errors you just press the OFF button then the ENTER button on the front panel.

This may be useful:

https://gensetservices.com/wp-content/themes/gensetservices/PartsPDFs/0F3750.pdf

cc18dd, 2025-02-17:
If it is a circuit test that fails it won't let you clear until you fix the circuit. Acknowledging just stops the alarm sound. Clearing get rid of the code if it's not a circuit test code

CRANKBOOSTER, 2025-02-17:
I was trying to get the connection to stop the power output without shutting down the engine, the engine shut down but now with the SD Engine RPM SN code.
 It is not clearing with the enter button, it wont start with manual or Auto mode just showing the error.