jgyates/genmon #247: Unable to get email to connect

Loglivers, 2019-03-12:
I have set up a gmail account to receive genmon messages.   I continually get the following error message - 
An error occurred:Error initiliazing SMTP library: [Errno -3] temporary failure in name resolution.   In my gmail account I have less secure apps turned on.   In genmon I am using port 587, ssl is on and disable tls is off.

jgyates, 2019-03-12:
If you log into your pi via ssh can you successfully ping the gmail server with this command?

   ping smtp.gmail.com

Type Ctrl+C to stop the program.

If you can not ping the address smtp.gmail.com then you DNS (name resolution service) is not setup on your pi correctly.

You can add a DSN server to the file /etc/resolve.conf if this is the case by editing the file:

    sudo nano /etc/resolve.conf

then add the following line:

       nameserver 8.8.8.8

This is google's public DNS server.

liltux, 2019-03-13:
@Loglivers 
I need you to login to your PI and open up the following file using sudo:

sudo nano /etc/resolvconf.conf

About a few lines down it describes using a personal domain name server with the following line:

#name_servers=127.0.0.1

you need that line to read:

name_servers=8.8.8.8

When changed, press ctrl+x and you will be prompted to save, hit Y, then it will ask to confirm file name and press ENTER.  This should save the file.  Then run:  SUDO REBOOT.

I am using dnsmasq to operate the access point dns and dhcp during the AP setup and this may overwrite the resolv.conf file as it needs.  But, if you edit resolvconf.conf this will be a more persistent change.  I tested on my unit and it works.  Also you static IP configuration is not allowing your DHCP in your router to populate the routes on the PI.  I hope this makes sense.  Let me know if you have more issues.

jgyates, 2019-03-13:
@liltux 

Thanks for the additional details. Sorry I forgot to add that part.

@Loglivers, let us know if you get this resolved.

Loglivers, 2019-03-13:
Hi

 

I complete this and verified that it saved which it did.   I am now getting the error:  an error occurred closing.   You mention my static IP configuration.   Is there something that I need to change there?

jgyates, 2019-03-13:
It is a bit unclear what is giving you the error. Can you elaborate? Are you getting an error when you try to ping, when you try to save the file /etc/resolve.conf or in some other situation?

Are you able to perform the ping test on smtp.gmail.com?

liltux, 2019-03-13:
@Loglivers no your static ip should be fine.  Setting a static IP at the pi requires you to specify everything dealling with routers and namerservers.  If you can ping www.google.com:
ping -c4 www.google.com
And if that is successful, then the network settings and configuration is good.  We now need to look at you email settings more.  Can you give more on your error? 

@jgyates just too quick...

Loglivers, 2019-03-13:
I am unable to ping.   I get a temporary failure in name resolution error.   The nameserver is saved at 8.8.8.8

Loglivers, 2019-03-13:
This is what I get

liltux, 2019-03-13:
Can you verify the following using your pc that is posting to this message board:
current ip address
subnet mask
default gateway
If you are using Windows, open up a terminal by clicking on the windows button and type "cmd" in the run box and press enter.  this should open up a terminal window.  then type "ipconfig /all"  some where down there should be a wifi or wireless network settings that will list a gateway, router, dns, and current ip address.

Loglivers, 2019-03-13:
This is my ipconfig

liltux, 2019-03-13:
Your attachements are not coming though to Github.  Try logging in to Github and commenting directly to the thread.

jgyates, 2019-03-22:
@Loglivers 

Are you still having problems with your email?

Loglivers, 2019-03-22:
No, I cleared all of the network setting and re-set them.   Everything is working now.   Thanks for the follow up.

jgyates, 2019-03-22:
No problem. I am glad it is working for you.

closed 2019-03-22 by jgyates