jgyates/genmon #248: Power output always displays 5kW

SethL, 2019-03-15:
I'm not sure if this is a bug or not.  Every time my generator runs, either by exercise, or manually started, the power output shows 5kW.  The transfer switch is still on street power so I would think there wouldn't be a load on the gen.  Last time I did an enhanced exercise where the transfer switch moved to gen power, the same thing happened.  It showed 5kW but when I put a load bigger than 5kW on it, it showed 12kW.  Cut the load off and back down to 5kW.  In short it seems like it always shows 5kW minimum no matter if there is a load on it.

The screenshot attached is from today when it was exercising but the house was still on street power.

I submitted logs just now.

[image]

jgyates, 2019-03-15:
@SethL 

The best explanation that I have is that the sensors readings that are used are based on the electromagnetic output of the generator. The software uses the sensor value to make an calculation that is close to the current that would be output and then calculates a value for the power output. The power value is used to calculate fuel usage. 

The problem is that I don't know when the system is under load for Air Cooled units, and even if I did, if I forced the current calculation to zero when not under load, then the fuel calculation would be off since it would not account for exercise cycles (or anytime the generator was running not under load).

Evolution models have a couple of methods for measuring current, depending on the model. Some use current transformers and some use Hall Effect Sensors. Both use magnetic fields. The software will read modbus registers to obtain numbers the raw sensor values and then use the method described here to make a final calculation:

https://github.com/jgyates/genmon/wiki/Appendix-4-Known-Issues

If your numbers are way off when under load then you may need to adjust the values of current divider or current offset mentioned in the above link.

Based on what you are describing it does not look like any thing is out of the ordinary. While no current is flowing to your house, you are still using fuel and the the CT sensor or Hall Effect Sensor is still providing a reading. 

As a point of reference, my 48kw Liquid Cooled unit uses a Hall Effect Sensor, and my output power is proportionally in line with what you are seeing from your generator. Your Air Cooled unit likely uses CT sensors.

SethL, 2019-03-15:
That makes sense.  I just wanted to make sure that you didn’t have a bug or anything.  What I am getting is anything over 5kW will be accurate which is perfectly fine.

 

Thanks!

 

https://github.com/jgyates/genmon/wiki/Appendix-4-Known-Issues

closed 2019-03-18 by jgyates