jgyates/genmon #249: SSL or port 80

rnatalie, 2019-03-19:
Thanks for a excellent product.   Once I realized I was editing the wrong genmon.conf and I got the Modbus address set to 64, it worked pretty nicely.

One problem I can't yet get around (and this is likely on the rasp os side) is that turning on SSL or even trying to set the port to regular http 80 won't work.    It appears the genmon program is running as root, but there may be some additional firewalling/permissions that I am unaware of. 

Any ideas?

### Your Environment

- Generator Model: Generac 80KVA w H100
- Generator Registers: Submit via the About page
- Genmon Version: V1.12.28

liltux, 2019-03-19:
@rnatalie There is a work around to run the flask server on port 80.  Can't remember exactly what, I had to goggle it.  But looking at the github for flask, it is not recommended to run flask on port 80.  SSL take a look at #230 might help.

jgyates, 2019-03-19:
@rnatalie 

Is your outbound email working? The reason I ask is that I did not receive your registers or log files. Outbound email must be configured and working for the registers or log to be submitted. The file /var/log/genserv.log is the log file associated with the web server. Are you editing the conf file directly or are you making the changes from the web interface? If you are editing it directly, what setting are you changing? There are multiple entries that control the web interface so I want to be sure that you are editing the right one. The file /var/log/genserv.py will have an entry that shows the configured port number so you can validate your settings are correct by looking at this file. It will also show if there are any errors related to the SSL libraries that were installed during the install process.

Jason

jgyates, 2019-03-19:
I just received your log files and registers. I don't see any errors in your genserv.log file. It looks like your settings are for HTTP and port 8080.

If you want to use HTTP and a port other than 8080, just change the "Port of WebServer" value on the settings page. If you want to use HTTP then enable "Optional - Webserver Security Settings" on the settings page and enable "Use Self-signed Certificate". If HTTS is enabled the port will be the HTTPS default (443) so your URL will just be https://youripaddress without the port at the end. If you want to use a non-standard HTTS port you can change that on the advanced page. You can get to the advanced page by clicking the icon in the top right of the web page twice.

Let me know if this helps or if you have other questions.

jgyates, 2019-03-19:
One other thing that jumped out at me in your log files,  make sure you set your nominal kw on the settings page, save the setting and refresh your browser. This value (nominal kw) helps set the bounds for the graphs and gauges on the web interface.

rnatalie, 2019-03-19:
Don't know what happened to my previous post, but here's the resolution just to help others:

Reinstalling pyopenssl as described in the link above fixed the https issue.

There was nothing wrong with port 80, it's just that I didn't realize that turning on ssl disables regular http.

I did fix my KW capacity.

By the way, there's a video kicking around of someone installing some other monitor that shows how to tap in to the +12 and GND on the H-100 so you don't have to go hanging things off the battery.    The rightmost terminal in the barrier strip inside the H-100 labeled "0" is ground.   The leftmost, labeled 13, is +12.   You can loosen the little sliding cover on the back of the H100 and fish your power wire through it.

closed 2019-03-19 by rnatalie