jgyates/genmon #251: Did showing incorrect amps / watts during power outage

krisasmith, 2019-04-15:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Show correct info

### Actual Behavior

Shows 50 amps when it's actually 7.4.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

[image]

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

lmamakos, 2019-04-15:
Do you have a 22KW generator?  I had a similar issue and the conclusion after some discussion in another thread was that the 22KW models didn't have the current transformers installed to measure the actual current consumption.   You might review Issue #156 for some some background that might be helpful.

krisasmith, 2019-04-15:
Yes I do have the 22kw model and it appears issue #156 does accurately describe what I am seeing. Given we know it's broken/working as designed, is there an option in the config to disable the power output and output current?

jgyates, 2019-04-15:
Hi @krisasmith 

See the setting named "Disable Power / Current Display" on the settings page. This will disable all features that use the calculated current.

closed 2019-04-15 by krisasmith