jgyates/genmon #253: Pi3 strange response

jcherney, 2019-04-18:
If this isnt' appropriate in this forum, please remove.

Trying to change the Wifi SSID and password on my Pi3. Used sudo raspi-config and made the change there. Saved, rebooted, and .... nothing. No change took place. I tried this 3 times. Same issue. Any ideas?

liltux, 2019-04-18:
Is the old network still available?

jgyates, 2019-04-18:
you can also edit wpa_supplicant.conf directly.

https://raspberrypihq.com/how-to-connect-your-raspberry-pi-to-wifi/

jcherney, 2019-04-18:
Ah ha! This must be it! 

Yes, the old network is still working. Strange that it doesn't get replaced... but ??

If I were to shut off the old network, make the change in raspi-config, wait, and restart the old network, would it remove the old info?

I have a backup image of the whole SD card. In fact, the first time I did this, I thought I crapped it out and used the backup. Turns out, it takes a few minutes after the reboot to connect...

> Is the old network still available?
> Raspi-Config only adds entries to wpa_supplicant.conf.
> So if the old network is available it will try to connect to it first before trying the new network.
> I would back up the original copy of wpa_supplicant. By running:
> sudo mv /etc/wpa_supplicant/wpa_supplicant.conf /etc/wpa_supplicant/wpa_supplicant.conf.bak
> 
> Then run Raspi-Config again. Verify you have an acceptable wpa_supplicant.conf before rebooting.

jcherney, 2019-04-18:
I saw that, and tried it. Not sure if I did it correctly, messed it up, or ....  But when it didn't connect immediately, I went to the backup.  Thanks...

> you can also edit wpa_supplicant.conf directly.
> 
> https://raspberrypihq.com/how-to-connect-your-raspberry-pi-to-wifi/

closed 2019-04-20 by jgyates