jgyates/genmon #254: Text Messages from Genmon to Android Cell

RoyRomeo, 2019-04-20:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Text Messages from Genmon to android phone

### Actual Behavior

Text Messages cut short

### Steps to Reproduce (including precondition)

NOTE: This message is a notice that the state of the gen

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: 0062440
- Generator Registers: Submit via the About page
- Genmon Version: V1.12.33

jgyates, 2019-04-20:
@RoyRomeo 

There are three different ways to send SMS messages from genmon. Can you be more specific in your description as to how you have your SMS setup and how much of the message is getting cut off? 

Genmon can send SMS messages via Twilio, via the PiHAT LTE card, and by using your cell provider email to SMS service. Depending on your service provider, the shortened SMS may be due to a limit of the provider when using email to SMS.  SMS has a limit of 160 characters, but many carriers support concatenation of messages which allows longer than 160 characters.

Also, when using email to SMS the message sent is longer than 160 characters, but if you use twilio or the LTE hat the message will be less than 160 characters.

RoyRomeo, 2019-04-20:
I have (phonenumber)@messaging.sprintpcs.com.The text message reads "NOTE:  This message is a notice that the state of the gen" (everything else is cut off)The email message reads "NOTE:  This message is a notice that the state of the generator has changed.  The system is not in alarm.  It  then lists "Status" of the Engine, Line, and Last Log Entries.

jgyates, 2019-04-20:
Since you are using email to text this is likely on your service provider end. From the genmon software perspective, the software is simply sending an email. The conversation to SMS takes place at the service provider. 

I would try copying and pasting the contents of the email and resend it as an email from your PC to your (phonenumber)@messaging.sprintpcs.com email address. You should receive this message as a SMS. If the SMS message is cut off when you send it from your PC then the problem is from your service provider.

RoyRomeo, 2019-04-20:
Thanks for the info.  I did what you suggested and the message is cut off, so the problem is with my service provider.

closed 2019-04-20 by jgyates

liltux, 2019-04-20:
Just wanted to chime in.  The email to txt or sms gateways are limited in characters.  The email alert sends more characters than what is allowed in a single message, i.e. a whole webpage of data.  This goes for all service provider gateways.  So once your message reaches the character limit the rest is dropped.

jgyates, 2019-04-20:
@RoyRomeo 

I checked in V1.13.00. This version adds a new addon that will send shorter messages via email. This can be used with a Email to SMS email carrier provided email address. The message will not be cut off by your carrier, but it will send an abbreviated message. Messages will be send when the state of the generator changes or if the state if the utility line changes. Note that industrial generators only support utility status if you are using a transfer switch that is MTS, STS, HTS compatible. Evolution and Nexus generator controllers support the utility line monitoring.

USAFPride, 2019-04-21:
Just saw this and I very much appreciate the addon.  The simpler way is to use MMS messaging vs. Text.  Using ATT for example:
text = xxx@txt.att.net
MMS = xxx@mms.att.net

swvalenti194, 2019-04-21:
@USAFPride thanks for that much better than txt option and MMS now comes under one text from my number not two from random numbers!