jgyates/genmon #255: Ability to change "quiet mode on"/quiet mode off" from genmon

RoyRomeo, 2019-04-22:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior 

Ability to change "quiet mode on"/quiet mode off" from genmon

### 

Unable to change "quiet mode on"/"quiet mode off" from Genmon maintenance settings.

### Steps to Reproduce (including precondition)

"quiet mode on"/quiet mode off" can only be set via generator controller.

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generac 0062440
- Generator Registers: Submit via the About page
- Genmon Version: V1.13.00

jgyates, 2019-04-23:
@RoyRomeo 

Is your outgoing email configured and working? I have not received your registers. You can test your outbound email on the settings page.

Also, once you get your outbound email working it would be helpful to submit your registers, then change quiet mode from the generator , then submit them again. This would allow me to see if the values in the registers that correspond to quiet mode are changing for the Synergy controller.

Thanks
Jason

jgyates, 2019-04-23:
@RoyRomeo 

I received your registers. Here is what appears to be happening:

The software can read the exercise and quiet mode from register 6. In evolution and nexus controllers this has been used as a read only register (writing has not effect). To set the exercise time and quiet mode the software will write registers 002c, 002d and 002e for the exercise time and date, and register 002f for quiet mode. Writing to your generator controller (Synergy) register 002f does not appear to do anything, however register 0006 (the read only register) is updated when you change the quiet mode on the controller. This is behavior similar to the Nexus controller. Setting quiet mode does not work on Nexus controllers either.

jgyates, 2019-04-23:
Ok. My mistake. I was looking over your registers and it appears you do not have a Synergy Controller, you have a regular Evolution controller. My apologies, I got my support threads crossed. Considering you have an Evolution Air cooled unit then I would suggest doing a power cycle of your controller. If that does not work then you could try to see if an update to your controller firmware works, however this would need to be done by a Generac dealer as the firmware is not publicly available.

RoyRomeo, 2019-04-23:
Can you tell me how to power cycle the controller?

liltux, 2019-04-23:
@RoyRomeo the easiest way is to locate a small 2 wire connector which is below the controller.  It is the only two wire bundle in a high temperature sleeve and there was is a thermistor inside the sleeve as well.  Disconnect that connector, which removes voltage for the battery charger.  Then remove the 7.5 amp fuse on top of the controller.  At this point the controller will power down.  Then re apply in reverse order.  You will have the startup wizard when it first starts asking for language, fuel type, etc.

jgyates, 2019-04-23:
@liltux 

Since this is a question that has been asked a few times I created an wiki page for power cycling the controller.

https://github.com/jgyates/genmon/wiki/Appendix-9-Power-Cycle-the-Evolution-Controller

Thanks

liltux, 2019-04-23:
@jgyates the same procedure is used for Nexus, Evo1&2, Synergy and Same for Evo Liquid cooled.  Nexus Liquid cooled does not have a simple 2 wire connector and it is easiest to place in off and disconnect both of the connectors in back of controller.
Oh and just in case it would be best interest to place all controllers in Off prior to power cycle and then back to Auto.

jgyates, 2019-04-23:
new link:
https://github.com/jgyates/genmon/wiki/Appendix-9-Power-Cycle-the-Evolution---Nexus-Controller

RoyRomeo, 2019-04-23:
Thanks for the info, and thanks very much for the "Mobile Carrier Email to SMS" add-on, it resolved my text message issue.

jgyates, 2019-04-23:
No problem

jgyates, 2019-04-25:
@RoyRomeo, Let us know if a power cycle fixed the problem.

RoyRomeo, 2019-04-25:
I power cycled the controller, but it did not resolve the issue.

jgyates, 2019-04-25:
I think the latest version of the Air cooled Evolution firmware is at least V1.20. Your registers indicate yours is V1.14. I don't know that this would definitively fix your problem or if it is worth the cost of having a Generac service person do a firmware update but I at least wanted to pass along the info.

RoyRomeo, 2019-04-25:
Thanks.  I do my own generator servicing, and have the 5 year extended warranty that will expire in June.  I contacted the local Generac dealer and they agreed to come out next Monday afternoon April 29 to update the firmware, under warranty.  I will let you know if that resolves the issue.

RoyRomeo, 2019-04-29:
Generator tech arrived today and upgraded firmware to V1.20, but it did not resolve the issue.

liltux, 2019-04-29:
@jgyates, has anyone successfully set quiet test on air cooled unit? I was thinking I stumbled across same issue when testing in the my lab on air cooled unit.  I wasn't concerned with changing quiet test so it never came to mind.  I will have to test next time I am in shop.

jgyates, 2019-04-29:
I believe so, but it is worth testing. Honestly, this feature has been in since the first release so it would be curious to see that this has not been tested yet. It works on on my liquid cooled unit.

jgyates, 2019-04-29:
It may be an issue with air cooled in general. Here is another issue from last year like yours. If I can get one more data point I can disable the writing of quiet mode on EvoAC. Then I will need someone to help out with testing to get this working (if the controller supports this):

https://github.com/jgyates/genmon/issues/152

RoyRomeo, 2019-04-30:
I am not familiar with the software, as I purchased a complete unit from Jared on Ebay.  But if there is anything I can do on my end, please let me know.

jgyates, 2019-04-30:
@RoyRomeo 

I have done a bit of debugging and it appears that this feature is not supported on the Air Cooled firmware. I am removing this option from the software for Evolution Air Cooled and Nexus units. I checked in V1.13.03. This update should show the quiet mode state but not allow it to be changed from the web interface.

Thanks for the feedback and testing and let me know if you have any other questions.

closed 2019-04-30 by jgyates

RoyRomeo, 2019-04-30:
Okay, thanks very much for looking in to this.

UPS99, 2019-05-01:
Generac's discontinued Advanced Wireless Monitor also had the option to change the exercise speed remotely - and while the setting appeared to change at the remote, the Evolution ignored the command and remained in whatever mode it had been set to from the panel.  Possibly the function was accidentally removed in an early firmware update - or never worked?