jgyates/genmon #256: Repeated "Generator Monitor Starting at -mysystemname-" emails after upgrade

UPS99, 2019-04-25:
- Generator Model: 6244, firmware 1.17
- Genmon Version: 1.13.01

After updating via the "Upgrade to Latest Version" tab to v 1.13.01 (from v 1.12.31), Genmon sent 10 identical emails about 10 seconds apart "Generator Monitor Starting at -mysystemname-".  After that, the typical "Generator Notice: -mysystemname-", with status info, was sent once. 

This could be a fluke  - perhaps the PI was being slow?  It is not a major issue, and I only am reporting it in case there was a related change to the code.

jgyates, 2019-04-25:
@UPS99 

this could be because the Flask library is not shutting down properly. There is code in genserver.py that will attempt to restart if a failure is detected during restart. If this code is triggered you will see multiple emails as you describe. A restart of the pi may be in order if you continue to see this problem.

UPS99, 2019-04-25:
The pi shows an uptime of 327 days, so it may be time for a reboot.  I'll do that and see what happens the next time there is an update.

UPS99, 2019-05-01:
I cold rebooted the pi today.  When it came up it sent the 2 expected Genmon alerts.  However when I ran the updater, it sent the same 10 emails as I had reported above.

UPS99, 2019-05-01:
[genmon.log](https://github.com/jgyates/genmon/files/3133805/genmon.log)

here is the genmon.log 
up until the update to 1.13.01, genmon would load once, or occasionally load, shutdown and load again.  However now it does it 10 times.

jgyates, 2019-05-01:
Can you either submit your logs from the about page or post your genserv.log file. this is the log that would contain the relevant info.

UPS99, 2019-05-02:
I just submitted the logs via the About page.

jgyates, 2019-05-02:
I do not see errors from the Flask library, however I do see some errors related to the serial port in /var/log/myserial.log:

      /dev/serial0:Attempting to use a port that is not open

Can you tell me the output of this command (run from the genmon directory)?:

      sudo python ./OtherApps/serialconfig.py -c

jgyates, 2019-05-02:
Also, what operating system and browser are you using? Can you duplicate this with another browser? I am trying to determine if the serial errors are the source of the problem or the result of another problem. Normally what happens is that the update is requested then the software is restarted, if an error occurs on restart then the software attempts to stop and restart again. In your case it looks like the update is happening multiple times. At the moment I don't know what is causing this to happen.
It appears that the update/restart is being happening several times and that should not be happening. If there were an issue with the flask library then the update would happen once and the restart could occur more than once, but that is not the case here.

UPS99, 2019-05-02:
It appears the problem is related to the browser I use - Palemoon, which is built on the Mozilla code used for Firefox, but somewhat lighter (and have used it with Genmon since first set up).  I just tried the latest Genmon update using IE 11 instead and it did not produce the extra restarts.  Both browsers are on Windows 7.  Palemoon is not acting abnormally otherwise (I am writing this on it now).  Maybe somehow it repeats the update acknowledgement?

The output of the serialconfig.py does show BlueTooth is enabled - if I am reading this right, and it should be disabled...

Checking : BT service disabled: Fail
Checking : Serial console command line removed: OK
Checking : Is enable UART in boot config: OK
Checking : Is BT overlay disabled: OK
Checking : Serial console service disabled: OK
Serial port may not work as expected. Not all required settings changes were detected._

jgyates, 2019-05-02:
Try running :

     sudo python ./OtherApps/serialconfig.py -e

I will try to test with this browser. It make sense that it is the browser since the JavaScript appears to be called multiple times. There may be a fix I can find.

UPS99, 2019-05-03:
I ran the command above and it did disable Bluetooth - so the checks now all result in "OK".

jgyates, 2019-05-03:
I have duplicated the issue with palemoon on windows 10 and attempted to debug this, however I don't think this is something I will be able to work around without some major restructuring. What normally happens is that the javascript running in the browser request a software update to the server running on the pi. The server restarts the software once the update is complete and the code in the browser should detect that the server is not responding (this is done by the ajax call generating an error and the javascript detects this error and refreshes the browser window which starts using the update code). What appears to be happening with palemoon is that the ajax call never returns (i.e. is it not detecting the socket error) and then the browser reissues the ajax call. There may be some setting in "about.config" within the browser that could change this behavior, but I do not know what that would be. I took a quick look for ajax related settings but nothing jumped out as something valid to try.

I will try to look into this a bit more, but for now I would just avoid palemoon when updating the software.

UPS99, 2019-05-03:
Something only done occasionally (updates) can be done from another browser.  There is a chance that this resulted from a change in the underlying Mozilla code and might affect Firefox too.  It could be that some change in about:config would fix it.  There might be some info at https://forum.palemoon.org/ - the developer is on there daily - I will see what a search shows up.

jgyates, 2019-05-03:
FYI : I just tested with Firefox. Firefox for Windows shows the same symptom. Firefox for Mac and Linux work as expected.

jgyates, 2019-05-03:
I updated the known issues page with this info:

https://github.com/jgyates/genmon/wiki/Appendix-4-Known-Issues

jgyates, 2019-05-03:
I modified the title a bit so others how have an issue would be drawn to this thread.

UPS99, 2019-05-03:
This must be in a Mozilla code release from earlier this year - I know they have had other new issues too - such as problems with the Google ReCaptcha logins where you have to select all the pictures with a bus in them, or similar.

jgyates, 2019-06-26:
I am closing this issue. Feel free to post here if you find that the issue is resolved by Firefox or Palemoon. I have documented this in the Known Issues section of the wiki.

closed 2019-06-26 by jgyates