jgyates/genmon #257: Genmon won’t upgrade

jcherney, 2019-04-27:
I’m on version 1.13.00 and can’t get any upgrades to take. Tried w 1.13.01 and now 1.13.02.  No difference. 

The downloads work fine. And the restart works fine. But nothing changes. It’s as if I never did anything.  Monitor still shows 1.13.00

Any thoughts?

liltux, 2019-04-27:
@jcherney I have had similar issue before. Check your networking.  Logged into the pi try pinging google.com or 8.8.8.8
Also run the route command and see if the default gateway is correct.

jgyates, 2019-04-27:
After you validate your network is working to outside addresses you can log into your pi via ssh and do this:

Change directories into the genmon directory:

     cd genmon 

Then try manually updating:

    bash ./genmonmaint.sh update

The output of the genmonmaint.sh script should display more details about the update and why it is not working.

jgyates, 2019-04-27:
Also, after you use the genmonmaint.sh script, the script will prompt you to hit Y for yes to continue. You can type "exit" to end your ssh session.

jcherney, 2019-04-28:
Thanks. I’ll try this tonight.

jcherney, 2019-04-29:
Weird: I can ping google.com and bing.com w no problem. I can't ping 8.8.8.8  I also cant ping my company website (although it resolved to an ip address).  Can ping some websites but not others.

Any idea whats happening?

> 
> @jcherney I have had similar issue before. Check your networking. Logged into the pi try pinging google.com or 8.8.8.8
> Also run the route command and see if the default gateway is correct.

jcherney, 2019-04-29:
So, I edited the wpa_supplicant file. Rebooted it. Now I can't connect. I'll put in the backup in the morning and start over.

jgyates, 2019-04-29:
regarding wifi, this configuration depends on the version of linux you are using. For the pi this is typically either rasbpian based on debian 8 or debian 9.

You can view your basic route info with the "route" command and looking at /etc/resolve.conf

Below is a typical output. Note the 192.168.1.1 may be different for you, depending on your local network config.

    pi@genmon:~ $ route
    Kernel IP routing table
    Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
    default         192.168.1.1    0.0.0.0         UG    202    0        0 eth0
    default         192.168.1.1    0.0.0.0         UG    303    0        0 wlan0
    192.168.1.0    0.0.0.0         255.255.255.0   U     202    0        0 eth0
    192.168.1.0    0.0.0.0         255.255.255.0   U     303    0        0 wlan0

    pi@genmon:~ $  cat /etc/resolv.conf 
    # Generated by resolvconf
    nameserver 8.8.8.8
    nameserver 192.168.1.1

jgyates, 2019-04-29:
fyi for raspbian 9 a typical wpa supplicate config would look like this:

    pi@genmon:~ $ cat /etc/wpa_supplicant/wpa_supplicant.conf 
    ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
    update_config=1
    country=US

    network={
	    ssid="wifinetworkname"
    	    psk="wifipassword"
    }

IP addresses are typically configured in /etc/dhcpcd.conf or with /etc/network/interfaces

This page describes the differences. It would be useful to determine if the problem is IP related or WiFI related. The following command will let you know if your wifi is connecting, if it is then you issue may be ip related.

    sudo iwconfig

To look at the IP config you can type:

    sudo ifconfig

jcherney, 2019-04-30:
So, it turned out to be a corrupt dhcpcd.conf file. Thanks to some very much appreciated assistance, it was corrected and now works beautifully .

closed 2019-04-30 by jcherney