jgyates/genmon #258: Text Anomaly

RoyRomeo, 2019-05-01:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Stable text.

### Actual Behavior

The word "Day" disappears and reappears every few seconds.

### Steps to Reproduce (including precondition)

In the exercise function of the maintenance section, if you select the Monthly option, choose a day of the month and time, then save it, there is an anomaly as the word "Day" disappears and reappears every few seconds, in the text statement "Exercise Time: Monthly Day-20 13:30 Quiet Mode On".  If you select Weekly or Biweekly the text is stable.  It doesn't matter if you set the monthly option from the generator controller or GenMon, it still does it.  It's certainly not a big deal, just thought I would bring it to your attention.

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: 0062440 - Air cooled 20KW
- Generator Registers: Submit via the About page
- Genmon Version: V1.13.03

jgyates, 2019-05-01:
I can duplicate this. I will look into this.

jgyates, 2019-05-01:
I checked in V1.13.04 which should fix this. Let me know if you have any issues.

closed 2019-05-01 by jgyates

RoyRomeo, 2019-05-01:
Thanks very much.  Works great!