jgyates/genmon #259: Maintenance UI Missing Start/Transfer

sgtmaj27, 2019-05-03:
- Generator Model: 5870 Nexus
- Genmon Version: 1.13.04

I noticed on this recent update to remove the Quiet Excersize functionon the Nexus controller, , that there is no longer an option to Start and Transfer the Generator.  Is this by design or do I have an error in my installation after the upgrade?

jgyates, 2019-05-03:
When I run in simulation they with with Nexus registers it is are enabled for me. Check your settings page and see if you have "Smart Transfer Switch" enabled. If this is enabled then this is why it is now showing up. Do you have a "Smart Transfer Switch (i.e. your generator runs in two wire mode and is controlled by your transfer switch and your transfer switch monitors the utility line) or do you have a "dumb" transfer switch (your controller tells the transfer switch when power is out)?

If this is not the case then submit your logs vis the about page and I can look and see if anything jumps out.

sgtmaj27, 2019-05-03:
Thank you for your rapid response, Jason.

Sent the logs.  I have the "dumb" controller, not two wire connection and
Smart Transfer Switch is not checked in settings.  Not sure when this
option disappeared.  It did occur recently as I just used the Start and
Transfer function in the last several weeks.

jgyates, 2019-05-03:
I have not received your logs. To send the logs you have to have outgoing mail working.

sgtmaj27, 2019-05-03:
I'll try again.

jgyates, 2019-05-03:
I still have not received your logs. The software sends and email to generatormonitor.software@gmail.com to submit the logs. If you are getting emails from the software when the software starts, generator runs, etc. The you may look at your mail server settings to make sure you don't have any filters blocking unknown outbound email. You can also test outbound email on the Settings page.

sgtmaj27, 2019-05-03:
This is strange.  I received emails that the Genmon restarted when I
updated last night.  Today, I can not send a test message.  Looks like
there is a problem with password/linking Genmon to my google account.  I
will look at this later today and get it fixed.  I will let you know when I
am able to send the logs. Thanks

jgyates, 2019-05-03:
you may look at /var/log/mymail.log. It may have some clues as to why your email is not working.

sgtmaj27, 2019-05-03:
Okay....Logs are sent.  Thank you for the hint irt the mymail.log.  I
received a security warning about my other gmail account and inadvertently
turned off access to less secure apps for this email address instead of the
other one.

jgyates, 2019-05-03:
I received your logs. The logs also contain your register contents which I loaded into my simulator and attempted to duplicate. I can not duplicate. 

What browser and OS are you using. Can you duplicate this with another browser? I hate to ask this but just making sure, you have scrolled all the way to the bottom of the page, correct?

sgtmaj27, 2019-05-03:
I have tried several browsers and devices, Windows, Apple and Android and
all give the same as the attached picture.  Since this just happened, and
you can not duplicate it with my data, I will try a reinstall of 1.13.04
later. I think that some file of mine got corrupted during the last update
and a reinstall will work.  Since I have relied on the update button, which
method should I use to best insure success of this reinstall.

Again, thanks for you support!

jgyates, 2019-05-03:
First make a backup of your settings from the web interface on the about page. If you want a fresh install, I would rename the existing genmon directory. from the pi directory type:

      rename genmon genmon.old 
      
Then update your pi system software:

      sudo apt-get update
      sudo apt-get dist-upgrade

Type Y if prompted to update the software by the second command.

Then follow what is on this page to install a fresh copy of genmon (this will create a new genmon directory):

https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

Not that during the execution of the install script, the script will prompt you if you want to copy the config files. If you type Y then your old files config files in the /etc directory (which you should have a of backup now) will be overwritten. You can re-resetup or copy or individual config files into the etc directory from the backup archive you downloaded in stop one

jgyates, 2019-05-03:
Sorry, that should be:

    mv genmon genmon.old

instead of 

    rename genmon genmon.old

sgtmaj27, 2019-05-03:
Thanks, I will get to this this evening and report back.

sgtmaj27, 2019-05-03:
Okay, I went through the reinstall process, but have not inputted any
configurations yet.  I still am missing the Start and Transfer option
button on the Maintenance page.  I did check the logs and right after
reinstalling there was an error that the Message and Feedbat dat files were
missing from the genmonlib folder.  I just copied both files from the
genmon.old install and the error went away.  Want to figure out why I am
missing the Start and Transfer button before going through the
configuration.

liltux, 2019-05-03:
Does Nexus support the start and transfer?

sgtmaj27, 2019-05-03:
Yes.  I have been using it up until this previous update.  I have used it
monthly to run the generator with a load without interrupting house power.

jgyates, 2019-05-03:
ok. Let me check something on my end. I was thinking (incorrectly) that the Nexus supported Start and stop, but not start / transfer. But it looks like I am mistaken. When I run your registers, I get a start button and a stop button, but no "Start / Transfer" Button. The picture you sent earlier did not come thru. To post a picture you have to use the web interface,. If you responding to email with an attachment, the attachment will get stripped off.

sgtmaj27, 2019-05-03:
Sorry about the picture problem.  I have the Start Stop buttons.....IThe
Start/Transfer is what is missing.

It was time for me to clean up this installation anyway.  It has been in
use for several years.

jgyates, 2019-05-03:
@sgtmaj27 

I just checked in V1.13.06. This should fix your problem. Sorry about the confusion on my end. For some reason I had it in my head that Start / Transfer was not supported on Nexus. I also enabled support for the Enhanced Exercise add on program for Nexus controllers. This will allow you to automate the transfer test. Check out the Add on page for this feature if you wish and let me know if it works for you. I am guessing it will.

The enhanced exercise add on supports full speed start/stop, quiet exercise (reduced RPM test), and start/ transfer test cycles. Let me know if the quiet mode test works if you have a moment to test that out. The controller supports four basic commands (start, stop, start / transfer, and start quiet mode). I don't think I have a data point on the remote command for quiet mode for Nexus. The web UI does not expose quiet test because at the time I had not written the enhanced exercise add on and I did not think there would be much use for it.

Also, anyone using a Nexus Liquid Cooled unit: can you validate that start / transfer works.

sgtmaj27, 2019-05-03:
I will look at these added functions tomorrow and let you know how they.

Thanks so very much for ALL of this!

sgtmaj27, 2019-05-04:
Jason:  I tested the Enhanced exercise function this morning.  I set Excercise Type to transfer, Frequency to weekly.  I then scheduled for 8 this morning and 15 minute.  The generator started, transferred, but never shut off after the scheduled time.  I scheduled 15 minutes and shut it down using the Stop button on the maintenance page after 25 minutes.  I then scheduled the Quiet function......which worked....other than not shutting down automatically.  The unit ran at approx 2450 rpm vs 3600 and was @ 40hz.  Appears to work great except for the shutdown.  The option to extend the exercise time is a definite plus for those of us in the colder regions.

One question....Does your Enhanced Exercise function override the internal exercise program.  IAW how can we have your program now be the sole control for exercise?

jgyates, 2019-05-04:
One think to keep in mind: The the generator runs under load, once the stop command is issued it will set the transfer switch in the off position (powered from the utility line) and the generator engine will continue to run in cool down mode for about 5 minutes. The status should show Cooling Down.

If it continued to run without the cool down phase then send me your logs and I can take a look.

The Enhanced Exercise does not replace the internal exercise. There is no way to disable the internal exercise.

sgtmaj27, 2019-05-04:
Apparently, during the Enhanced Exercise, a stop command was not sent after the scheduled time.  I set it to 15 minutes.  After 25 minutes the genrator was still running under load.  It was not until I issued the stop command from the maintenance page that the generator transfered the load off and cooled down.  sending logs.

jgyates, 2019-05-04:
According to the logs it looks like the enhanced exercise program generates an error when checking the generator time (in /var/log/genexercise.log). Do you have "Use Generator Time" enabled? I added some additional logging on this error. Can you update your software from the about page and test again when you have time, then send me your logs. I tried to duplicate the error see in your logs your registers data but I could not.

what is suppose to happen (and it does in my testing) is if Use Generator Time is enabled then the time will be read from the generator and the time will be converted to a python datetime data format. For some reason you have errors in your log related to the conversation. The code should fall back to the system time if this error occurs. Is your generator time about the same as your pi system time (look at the bottom of the main status page in the web UI). If not you can sync the two times (at least within a minute) on the Maintenance page by clicking "Set Generator Time". The pi system time should be synced with a time server by default (assuming you have setup the time zone using raspi-config).

sgtmaj27, 2019-05-04:
I updated and ran the Load test again, this time for 5 min test.  Sent the logs

jgyates, 2019-05-04:
Ok. I think the error in your log is just an error getting the time when the program first starts. If all of the registers have not been read over serial this could happen. I only see it once, not many times (i.e. every minute could indicate a bigger issue). Let's try this:

Log into the pi and get into the genmon directory, then type:

     python ClientInterface.py

This will give you a prompt like this ":>" .  At this prompt type:

     generator: setremote=starttransfer 

This will start the generator and activate the transfer switch.

Then once the transfer switch is activated, type:

     generator: setremote=stop 

This should stop the generator.

The four commands that you can use are:

     generator: setremote=start 
     generator: setremote=stop 
     generator: setremote=starttransfer 
     generator: setremote=startexercise 

This will test the base functionality without the genexercise add on involved. Genexercise calls these commands. The above method calls directly into genmon.py which issues the modbus commands.

jgyates, 2019-05-04:
Another thing we can try is edit the file /etc/genexercise.conf and add the following line:

     debug=True

Then restart the software. 

     bash ./startgenmon.sh restart 

This will add more info to the log file like when a start or stop is attempted.

sgtmaj27, 2019-05-04:
I will do starttransfer and stop using the ClientInterface.py. Also I will add the line to genexercise.conf.  After I do this do you want logs sent?  I will not be able to do this though until I return home later today.

jgyates, 2019-05-04:
Just let me know how it responds using the commands from ClientInterface.py. If it acts the same then the issue is not with genexercise.py, it is with the controller. If it acts as expected (stop will terminate a run that was started with starttransfer) then we can start looking at genexercise by enabling debug=true, doing another exercise from the web interface and sending logs.

No rush, just when you can get to it...

sgtmaj27, 2019-05-04:
I ran the commands from thr ClientInterface.py.  The unit started, transfered and shut down with the appropriate commands.  I then added debug=True to the genexercise.conf and ran the unit using the web interface.  It started and transferred on time, however I needed to manually shut it down via the web interface.  Logs sent

sgtmaj27, 2019-05-04:
Of course I waited more than the 5 minutes that I had it scheduled to run before shutting it down.

jgyates, 2019-05-04:
I found a bug in genexercise that would not call the stop functions if a warmup cycle was not specified. I fixed this. Thanks for helping me track this down. I appreciate you help in debugging as well. Let me know if this does not fix the problem.

sgtmaj27, 2019-05-04:
Thank you for all your hard work!

sgtmaj27, 2019-05-05:
Last update fixed problem with the Nexus controller.  All addon exercise functions operational.

closed 2019-05-05 by sgtmaj27