jgyates/genmon #261: Liquid Cooled Nexus Data

liltux, 2019-05-07:
Nexus liquid cooled.  Rpi Zero W.  No Alarm Log or Service Log display on Log page, only event/status Log.

@jgyates I just got around to plugging up to a liquid cooled Nexus at the shop.  I found that the remote start button, start/transfer both work on Liquid Cooled Nexus.

Although I did get an unknown register value/change and setup auto feedback.  

I also do not have any alarm or service logs on the Logs Screen, but I do have Run logs.  I have submitted several registers/logs under Rpi zeroW.

I do not have an air cooled nexus to use for data points.

[image]

jgyates, 2019-05-07:
I got your logs. One thing that jumps out is the software currently thinks the underlying controller is an Air Cooled Evolution yet Register 0000 = 0x0006 which indicates a Nexus Air Cooled. The logs also indicate this (Nexus only has two logs, Evolution has three). One of two things could cause this:

1) The /etc/genmon.conf file is using the controller detetion override settings (evolutioncontroller=true, liquidcooled=false, these two should be empty, not true or false by default).
2) Genmon was not restarted after connecting the serial link after having been previously connected to another controller. Early on in the load process the software detects the controller and sets flags accordingly. If you unplug serial from one controller and reconnect it to another the genmon needs to be restarted. To resolve this just reastart genmon.

Let me know if a restart does not correct the problem.

liltux, 2019-05-08:
@jgyates I will looking into the settings prolly hard coded as this was one of my original builds prior to some of the new settings and still using the original genmon conf file.  Thanks

liltux, 2019-05-08:
@jgyates , got it all up to date.  So I have a few values not recorded.
0x0D - Missing Cam Pulse alarm
0x10 - Ignition Fault alarm

jgyates, 2019-05-08:
Thanks, I will add these to the next release.

closed 2019-05-08 by jgyates