jgyates/genmon #262: Check battery alarm severity?

hokie21, 2019-05-10:
I got a check battery alarm. I was surprised to see it is of "critical" severity instead of "warning" severity. In my case it was just because the battery needed to have the water checked based on a timer. Does this alarm serve more than this one purpose that accounts for its "critical" severity?

liltux, 2019-05-10:
@jgyates I believe all of the service alarms trigger the critical comment.  I am not sure how the controller differentiates between generator alarms and service timer alarms, or they may be treated the same?  I had one in the lab trigger the same way awhile back and didn't think much of it.
Check battery is an annual timer.
Level A service is every 200 hrs timer.
Level B service is every 400 hrs timer.

jgyates, 2019-05-10:
The controller treats every alarm the same (i.e. there is not status from the controller to differentiate between critical or warning).

The software will report "CRITICAL" if the generator is in alarm and "WARNING" if the software detects something is wrong with genmon (dead thread, no comms, system still initializing, etc).

The software will monitor the service alarms and report that service is due, but once the alarm expires the controller sets an alarm condition. Many, if not all, of the alarms require the alarm to be reset before the generator will be allowed to start. This may be the reason that all alarms are treated with the same severity.

liltux, 2019-05-10:
Assumed as much

closed 2019-05-11 by jgyates